betaflight/betaflight #7341: Add support for Frsky EU-LBT protocol

alexbirkett, 2019-01-05:
Add support for EU-LBT FrSky protocol

Unnecessary extra hardware is being built into EU versions of quads to comply with EU regulations. Missing support for EU-LBT means the EU versions of quads are more expensive, heavier, draw more current and have more drag. I wrote [a post](https://alexbirkett.github.io/quad/2019/01/05/mobula_7_EU.html) about it.

etracer65, 2019-01-05:
Do you mean this that was submitted 5 hours before you opened the issue?

0crap, 2019-01-05:
@alexbirkett you mean the onboard FC receivers?
Isn't that solved with pull request #7339 ?

alexbirkett, 2019-01-05:
That's a coincidence. #7339 will fix this issue

mikeller, 2019-02-05:
@alexbirkett: Has this been solved for you? If so, can you please close this?

closed 2019-02-12 by stale[bot]