betaflight/betaflight #7346: GPS rescue mode: radio link delay

vannoz, 2019-01-08:
after the failsafe we have the GPS-RESCUE mode activate. We need to set a delay time value (via CLI is better) to have some seconds after the radio link come back. This because when the kwad is in the 'failsafe area', it need some seconds to enter into the radio link hemisphere where the RSSI is above a certain threshold level to avaoid a second failsafe. So with this delay the FC will remain in GPS-RESCUE until the RX receiver signal is below a threshold... unfortunately we cannot modify as per our need the receiver firmware, so we have to play with the GPS-RESCUE setting...
what do you think?
thanks
BR
Andrea

SchonRichtig, 2019-01-08:
Put GPS Rescue on a Switch and activate it after you had an Failsafe.
So it will stay in GPS Rescue even when the link is back.
When you feel safe deaktivate it by the switch

vannoz, 2019-01-08:
You can see here a video where after failsafe, then GPS RESCUE goes on, the
rx take back the signal, then goes again in failsafe.

https://drive.google.com/file/d/1woGtZFJ1vV1Au505_ysNpOi3xI4Gm2hO/view

In this small time you have not time to react and switch to GPS-RESCUE...

SchonRichtig, 2019-01-08:
It looks like it stayd in failsafe Stage 1 at the first RX Loss.
At the second Loss the time was long enough to activate stage 2 GPS-Rescue.
At this point maybe the sanitychecks kicked in because something was wrong with the GPS.
Then it disarmed itself.
Have you a DVR of this situation? Maybe you had to few sats?

While i was testing the older R9 Flex firmware i had enough chances to test the Failsafe behavior :)
When i had a failsafe the quad went to angle mode (set in stage 1), throttle was on hold all, other on auto.
than GPS Rescue was engaged and i flipped the switch to GPS rescue and after about 10 seconds i disabled it.
So it works in my case.

vannoz, 2019-01-08:
No I haven't sorry...
However sometime also 4 sec could not be a long time to react... Especially
if you are waiting for the failsafe... Maybe you think to hit something...
And for sure you have the finger not on the right switch...

vannoz, 2019-01-10:
Especially if you are not waiting for the failsafe

ChilledoutUK, 2019-01-21:
I Have a issue open from 26 May 2018 for this exact problem asking for the inav failsafe_recovery_delay to be added to betaflight.

https://github.com/betaflight/betaflight/issues/5972

I think this is absolutely necessary for this mode to actually work properly as even with it enabled via a switch gps rescue is constantly reinitialized when going in and out of failsafe.

This constant re initialization causes a lot of altitude gain and possibly outside the legal limit.

We need this we cant rely on people knowing that they have to enable gps rescue if they loose video from orientation they may not even know its trying to function on gps rescue.

If people are worried about gps rescue staying stuck on too long it could be added that if you enable gps rescue via a switch and then back to a normal say air mode it interrupts the gps rescue delay period.

vannoz, 2019-01-21:
Let's try to launch a vote on Betaflight FB group.... I don't know other
ways to sensibilize the Authors...

mikeller, 2019-01-21:
@vannoz: Not much point in launching a vote, I don't think anybody disputes the usefulness of this - just open a pull request for the change you want to make.

vannoz, 2019-01-21:
Ok I'll do.. Thanks bro!

ChilledoutUK, 2019-01-25:
Huh I though pull requests were for merging code/code changes that you have written?

vannoz, 2019-01-25:
No no... Any code from my side...

mikeller, 2019-01-26:
@ChilledoutUK: Yes, they are.

TonyBlit, 2019-03-25:
I had an idea to change how to recover control from gps rescue in failsafe mode: once RX comm is restored, just show a message asking the pilot to switch rescue mode on and then off manually. If the pilot does not do this, gps rescue does its thing until the end. I think this is much better than a delay (which will never suit all cases), at the small cost of having to map rescue on a switch (which is very recommended anyway). I prototyped this change months ago but then I got distracted fixing other rescue related issues. If you find it useful I will try to find some time to do the PR.

mikeller, 2019-03-25:
@TonyBlit: I am not a GPS rescue user myself, so I can't comment on the practical aspects. But from a UX point of view this sounds like a good suggestion - it uses an existing resource (the GPS rescue mode switch), and it provides confirmation that the link is actually good again. Not even sure a specific message is needed - just allowing the pilot to disable GPS rescue (be it originally automatically enabled or enabled with the switch) by switching the mode off makes sense.

ChilledoutUK, 2019-03-27:
Yes having gps rescue stay in gps rescue until you turn it on then off again would solve all the intermittent problems perfectly.

closed 2026-09-26 by blckmn