betaflight/betaflight #7347: Dshot problem (?) starting build #1328/#1329?

ottofung, 2019-01-08:
**Describe the bug**
Starting build #1328/1329, CRAZYBEEF3FR suddenly becomes not working. I can arm the quad but motors not responding to any throttle inputs

mikeller, 2019-01-08:
Looks like this is caused by 816da6cbf5105981a00c015f3d71a391349f30f3:

My testing with CRAZYBEEF3FR:

257a09d69319a18be3e8f3c78f46f3db74766503,
```
set motor_pwm_protocol = DSHOT300
set dshot_burst = on
```

=> motors work

816da6cbf5105981a00c015f3d71a391349f30f3
```
set motor_pwm_protocol = DSHOT300
set dshot_burst = on
```

=> motors don't work

816da6cbf5105981a00c015f3d71a391349f30f3
```
set motor_pwm_protocol = DSHOT300
set dshot_burst = off
```

=> motors work

Looks like #7264 broke burst DMA for Dshot at least on F3.

@joelucid: Do you have time to look into this?

@ottofung: `set dshot_burst = off` should get you flying again for now.

ottofung, 2019-01-08:
thx, will give it a try, cheers.

joelucid, 2019-01-08:
I'll look at it very soon, @mikeller

joelucid, 2019-01-08:
I can reproduce it on SPRACINGF3. My NOX F4 works with burst. So seems like a F3 specific regression.

mikeller, 2019-01-08:
Thanks @joelucid. Updated title to avoid duplicate reports.

mikeller, 2019-01-09:
Fixed by #7439.

closed 2019-01-09 by mikeller