betaflight/betaflight #7354: OmnibusF4SD motor resource A03

jeober, 2019-01-09:
Lost a motor operation using resource A03 between #1328 and #1329 and still present in latest.   Confirmed the same behavior on two quads having identical hardware, firmware, and settings.

mikeller, 2019-01-09:
@jeober: Can you please supply the information that you were asked to supply when opening the issue?

jeober, 2019-01-09:
```# version
# Betaflight / OMNIBUSF4SD (OBSD) 4.0.0 Jan  9 2019 / 20:26:00 (a1eac6f0d) MSP API: 1.41

board_name OMNIBUSF4SD
manufacturer_id 

# name
name x216-2206/7

# resources
resource MOTOR 1 A02
resource MOTOR 4 B00

# dmaopt

# mixer

# servo

# servo mix

# feature
feature -RX_PARALLEL_PWM
feature -OSD
feature RX_SERIAL
feature TELEMETRY

# beeper
beeper -GYRO_CALIBRATED
beeper -RX_LOST_LANDING
beeper -DISARMING
beeper -ARMING
beeper -ARMING_GPS_FIX
beeper -BAT_CRIT_LOW
beeper -BAT_LOW
beeper -GPS_STATUS
beeper -ACC_CALIBRATION
beeper -ACC_CALIBRATION_FAIL
beeper -READY_BEEP
beeper -DISARM_REPEAT
beeper -ARMED
beeper -SYSTEM_INIT
beeper -ON_USB
beeper -BLACKBOX_ERASE
beeper -CRASH_FLIP
beeper -CAM_CONNECTION_OPEN
beeper -CAM_CONNECTION_CLOSE

# beacon
beacon RX_LOST
beacon RX_SET

# map

# serial
serial 2 64 115200 57600 0 115200

# led

# color

# mode_color

# aux
aux 0 0 0 975 1025 0 0
aux 1 1 1 1975 2025 0 0
aux 2 13 1 1475 2025 0 0
aux 3 28 1 975 1525 0 0

# adjrange
adjrange 0 0 2 900 1175 12 2 0 0

# rxrange

# vtx

# rxfail
rxfail 5 s 2000
rxfail 6 s 1000

# master
set gyro_lowpass_type = PT1
set gyro_lowpass_hz = 0
set dyn_notch_width_percent = 6
set dyn_notch_q = 135
set dyn_lpf_gyro_min_hz = 175
set dyn_lpf_gyro_max_hz = 575
set acc_calibration = 12,-18,46
set mag_hardware = NONE
set baro_hardware = NONE
set min_check = 1010
set max_check = 1990
set rc_interp_int = 4
set max_aux_channels = 4
set serialrx_provider = FPORT
set serialrx_halfduplex = ON
set blackbox_p_ratio = 64
set dshot_idle_value = 350
set motor_pwm_protocol = DSHOT1200
set failsafe_delay = 10
set failsafe_off_delay = 90
set failsafe_throttle = 1245
set failsafe_throttle_low_delay = 90
set failsafe_procedure = AUTO-LAND
set bat_capacity = 850
set current_meter = NONE
set force_battery_cell_count = 4
set vbat_scale = 108
set yaw_motors_reversed = ON
set small_angle = 30
set pid_process_denom = 1
set frsky_vfas_precision = 1
set report_cell_voltage = ON
set rate_6pos_switch = ON

# profile
profile 0

set dyn_lpf_dterm_min_hz = 120
set dyn_lpf_dterm_max_hz = 245
set dterm_lowpass_type = PT1
set dterm_lowpass_hz = 120
set dterm_lowpass2_hz = 0
set vbat_pid_gain = ON
set anti_gravity_threshold = 175
set anti_gravity_gain = 3000
set iterm_relax_type = GYRO
set pidsum_limit = 1000
set pidsum_limit_yaw = 1000
set throttle_boost = 0
set p_pitch = 32
set i_pitch = 8
set d_pitch = 2
set f_pitch = 150
set p_roll = 32
set i_roll = 8
set d_roll = 2
set f_roll = 150
set p_yaw = 12
set i_yaw = 32
set f_yaw = 150

# rateprofile
rateprofile 3

set thr_mid = 0
set thr_expo = 90
set roll_rc_rate = 170
set pitch_rc_rate = 170
set yaw_rc_rate = 150
set roll_srate = 75
set pitch_srate = 75
set yaw_srate = 71
set tpa_rate = 0

# ```

RX is R-XSR via Fport; using latest Frsky beta firmware
FC is F4 "Nano" from RCXHOBBY.COM model:  RCX07-582
ESC is Spedix GS35; latest BLHeli_32 firmware
OpenTX 2.2 on Taranis; all works including Lua
Tried BF configurators 10.4 and 10.5 without difference.
My diff's from #1328 (all motors run) and #1339 (motor 3 A03 doesn't run) are identical.
I have run this RX and FC combination with various ESCs using regularly updated nightlies for at least a year on multiple quads without issue until nightly 1329.  The Spedix is a newer ESC for me but it runs OK up to 1328.  Motor 3 on A03 which is the default resource does not operate from #1329+, either on the configurator motor tab or from Taranis.

Asizon, 2019-01-10:
@jeober what Flight Controller are you using with this firmware?

Asizon, 2019-01-10:
Wait, you talk about Omnibus Nano V6? There is a lot of information about it in the network. The factory has an OMNIBUSF4SD custom firmware installed. But you must install the OMNIBUSF4FW firm so everything works, including the motor. in the case of the normal OmnibusF4 V6 (not nano), it already has its own target OMNIBUSF4V6

jeober, 2019-01-10:
Dshot600 or DShot1200 doesn't make a difference.

This doesn't run motor A03 for builds past #1328:
resource MOTOR 1 A02
resource MOTOR 2 B01
resource MOTOR 3 A03
resource MOTOR 4 B00

This does run all motors but creates wrong positions for my build:
resource MOTOR 1 A03
resource MOTOR 2 A02
resource MOTOR 3 B00
resource MOTOR 4 B01

I also tried reordering the first resource list but no success.  ???

closed 2019-01-10 by jeober