betaflight/betaflight #7358: Motolab TempestF4 won't write blackbox to sd card

tavdog, 2019-01-10:
I'm on Betaflight / MLTEMPF4 (MLTE) 3.5.4 and I can't get it to write a blackbox log file.  The blackbox page states that the card is ready.  The card can be written to when used in a MatekF405.  The blackbox log num displayed in osd is stuck at #2. Sometimes it will create a 0 byte file, then the log file num shown in osd flight stats increments by 1

The FC label shows TempestF4 V1.

mikeller, 2019-02-09:
Betaflight is quite picky about what SD cards it works with - try different cards from different manufacturers, stick with what you find to be working.

tavdog, 2019-02-10:
I'm believe this is a hardware issue as another user says he's only been able to get 1/4 of these boards to write to the sd card.

closed 2019-02-10 by tavdog