betaflight/betaflight #7359: Strange current/voltage readings

klutvott123, 2019-01-10:
**Describe the bug**
I just got a Matek F722 STD FC, flashed the latest 4.0.0 build 1339 to test it, discovered strange behaviour from current and voltage sensor on my telemetry screen and in the configurator. Current reading stays at 0 and jumps up a little from time to time only to settle back too 0. The voltage reading behaves a little differently. When armed/before arming it seems to work ok, but when i disarm, it jumps up to a voltage higher than the supply voltage and stays there. For example, the power supply i use for bench testing is 12.15V. When disarming, the displayed voltage jumps up to 14.8-15V and stays there until i arm again. Funny thing is that VFAS- follows it as it rises. It doesn't freeze since i can see the value changing up and down a little bit. 

I then started working my way back from build 1339 and found that it starts working again at build 1332. 
I don't know if anything other than current/voltage is affected. I don't think it has anything to do with telemetry since i see this behaviour in the configurator too. 

**To Reproduce**
Flash Matek F722 FC with build 1333 or above. Arm and increase throttle. Observe current/voltage readings. 

**Expected behavior**
I expect to see the actual current/voltage.

**Flight controller configuration**
```
# version
# Betaflight / MATEKF722 (MKF7) 4.0.0 Jan  8 2019 / 14:49:41 (40279f339) MSP API: 1.41

board_name MATEKF722
manufacturer_id 

# name
name Martian 215

# feature
feature -OSD
feature LED_STRIP
feature AIRMODE

# beeper
beeper -GYRO_CALIBRATED
beeper -RX_LOST_LANDING
beeper -DISARMING
beeper -ARMING
beeper -ARMING_GPS_FIX
beeper -BAT_CRIT_LOW
beeper -BAT_LOW
beeper -GPS_STATUS
beeper -ACC_CALIBRATION
beeper -ACC_CALIBRATION_FAIL
beeper -READY_BEEP
beeper -DISARM_REPEAT
beeper -ARMED
beeper -SYSTEM_INIT
beeper -ON_USB

# beacon

# map
map TAER1234

# serial
serial 1 0 115200 57600 0 115200
serial 2 64 115200 57600 0 115200

# led
led 0 4,4::CT:7
led 1 4,5::C:0
led 2 4,6::C:0
led 3 4,7::C:13
led 4 7,7::C:13
led 5 7,6::C:0
led 6 7,5::C:0
led 7 7,4::CT:7

# color
color 1 0,255,15
color 2 0,0,15
color 3 30,0,15
color 4 60,0,15
color 5 90,0,15
color 6 120,0,15
color 7 150,0,15
color 8 180,0,15
color 9 210,0,15
color 10 240,0,15
color 11 270,0,15
color 12 300,0,15
color 13 330,0,15

# mode_color

# aux
aux 0 0 0 1950 2050 0 0
aux 1 13 3 1950 2050 0 0
aux 2 15 3 1450 1550 0 0
aux 3 26 1 1450 1550 0 0
aux 4 36 2 1950 2050 0 0

# rxfail
rxfail 3 h

# master
set gyro_sync_denom = 2
set acc_hardware = NONE
set mag_hardware = NONE
set baro_hardware = NONE
set min_check = 1005
set max_check = 1995
set serialrx_provider = FPORT
set serialrx_inverted = ON
set serialrx_halfduplex = ON
set motor_pwm_protocol = DSHOT600
set bat_capacity = 1550
set vbat_max_cell_voltage = 420
set pid_process_denom = 2

# profile
profile 0

set vbat_pid_gain = ON
set p_pitch = 54
set i_pitch = 45
set d_pitch = 34
set p_roll = 47
set i_roll = 40
set d_roll = 32

# rateprofile
rateprofile 0

set roll_rc_rate = 223
set pitch_rc_rate = 223
set yaw_rc_rate = 210
set roll_expo = 80
set pitch_expo = 80
set yaw_expo = 70
set roll_srate = 0
set pitch_srate = 0
set yaw_srate = 0

```

**Setup / Versions**
Matek F722 STD with Matek FCHUB PDB
Aikon AK32 35A BlHeli32 ESCs
R-XSR receiver FPORT 180827 beta firmware
Taranis QX7 with opentx 2.2.2

McGiverGim, 2019-01-10:
Strange, 1333 was https://github.com/betaflight/betaflight/pull/7306 and I don't see nothing about vbat, only OSD menu changes...

klutvott123, 2019-01-10:
@McGiverGim Yes, it's very strange. My first thoughts was that it's related to the recent dma changes, but those builds are working as expected. I don't see how #7306 could cause it to break unless there's something else that is wrong, and #7306 was just enough to push it over the edge. 😄 

I have tested back and forth between 1332 and 1333 with full chip erases each time. Same thing happens. Works on 1332 and breaks on 1333.

klutvott123, 2019-01-10:
I have done some more testing. First i flashed build 1339 from the configurator. Doesn't work. 
Then i did a local build of the latest master which should be the same as 1339. Works just fine. Repeated this test several times. 
Also tested on a FF Fortini F4. The builds from the configurator works just fine on it. 

@McGiverGim Is it possible that something goes wrong with the jenkins builds? 
Since it looks like this started with build 1333(January 8), can it have something to do with this https://developer.github.com/changes/2018-11-05-github-services-brownout/?

McGiverGim, 2019-01-10:
I don't think so, but let's wait other developers that have more experience than mine in the compiler part

klutvott123, 2019-01-11:
Here's some new info on this. Flashed build 1342 and did some more testing. Current still doesn't work and voltage still gives strange readings. Then i connected to the configurator and the voltage immediately settles on the correct value both on telemetry screen and in the configurator. It gives correct current readings too when controlling the motors through the motors tab. Then i press the disconnect button and it stops working again. Had the usb cable plugged in the whole time so i don't think it's a power issue.

closed 2019-02-17 by stale[bot]