betaflight/betaflight #7360: matek 405 arm switch cannot match beeper switch

justinpmorrow, 2019-01-10:
**Describe the bug**
A clear and concise description of what the bug is.

**To Reproduce**
Steps to reproduce the behavior:

**Expected behavior**
A clear and concise description of what you expected to happen.

**Flight controller configuration**
Create a `diff` and post it here in a code block. Put ``` (three backticks) at the start and end of the `diff` block (instructions  on how to do a diff: https://oscarliang.com/use-diff-not-dump-betaflight/)

**Setup / Versions**
 - Flight controller [what type is it, where was it bought from];
 - Other components [RX, VTX, brand / model for all of them, firmware version where applicable];
 - how are the different components wired up.

**Additional context**
Add any other context about the problem here.

justinpmorrow, 2019-01-10:
Target: MATEKF405
Name/Version: 4.0.0 #1339
Binary: betaflight_4.0.0_MATEKF405-1339-a1eac6f0d.hex
Date: 10-01-2019 06:48

had beeper set to middle of 3 bottom switch and arm set to bottom of same 3 position switch
quad wouldn't arm old showing armswitch warning.
moved beeper to a different switch and now arms without issue
[image]

jflyper, 2019-01-10:
@justinpmorrow Please provide the information as requested by the instructional text (which you left unmodified in the first comment).

etracer65, 2019-01-11:
I've tracked this down to an interaction with DSHOT beacon (using the motors for a beeper) and PR https://github.com/betaflight/betaflight/pull/7264.  Starts with build 1329.

Looking into it.

@justinpmorrow I really need your settings `diff`.

justinpmorrow, 2019-01-12:
@etracer65 correct. Dshot 600 and motors for the beeper

PR #7378 (2019-01-12): Fix race condition between DSHOT beacon and bidirectional telemetry #7372
This builds on @etracers fix, but adds a dshot command queue and limits max rpm notch filter frequency.

Fixes #7360

If DSHOT beacon was enabled when arming, the guard time delay would defer the arming as is normal. However the extra DSHOT command to enable telemetry was using the blocking method so this would temporarily cause a 100% CPU load which then triggered the arming disable flag.

The command to enable DSHOT telemetry needed to be after the guard period. Also needed some additional conditionals to ensure DSHOT  [... cut]
closed 2019-01-12 by joelucid