betaflight/betaflight #7366: Betaflight 4.0 Motor #4 Not Working with DSHOT

brunoleaomaia, 2019-01-10:
I'm using Matek F405-CTR and Betafilght 4.0#1342

I need remap the motors because the FC is 90 degrees yaw. It's working fine with BF 3.5.3.

resource MOTOR 1 C07
resource MOTOR 2 C09
resource MOTOR 3 C06
resource MOTOR 4 C08
save 

After commands in CLI, the motor 4 don't work. Don't receive any signal with DSHOT300/600/1200. If I change to ONESHOT125 motor 4 works after reboot.

Before remap the motors all motors works good with DSHOT1200.

jflyper, 2019-01-11:
Please supply CLI `diff` information to reduce the support cost.

brunoleaomaia, 2019-01-11:
# reset configuration to default settings
defaults nosave

# name
name MAIAFPV

# resources
resource MOTOR 1 C07
resource MOTOR 2 C09
resource MOTOR 3 C06
resource MOTOR 4 C08
resource MOTOR 5 NONE
resource MOTOR 6 NONE
resource LED_STRIP 1 A15
resource SERIAL_TX 2 NONE
resource SERIAL_TX 11 A02
resource CAMERA_CONTROL 1 A08

# mixer

# servo

# servo mix

# feature
feature -SOFTSERIAL

# beeper
beeper -ON_USB

# beacon
beacon RX_LOST
beacon RX_SET

# map

# serial
serial 2 2048 115200 57600 0 115200

# led

# color

# mode_color

# aux
aux 0 0 0 1300 2100 0 0
aux 1 1 2 900 1300 0 0
aux 2 13 1 1700 2100 0 0
aux 3 28 2 1700 2100 0 0

# adjrange

# rxrange

# vtx

# rxfail
rxfail 5 s 1000
rxfail 7 s 1000

# master
set acc_calibration = 9,82,-42
set mag_hardware = NONE
set baro_hardware = NONE
set min_check = 1030
set max_check = 1995
set rssi_channel = 8
set rc_smoothing_type = FILTER
set motor_pwm_protocol = DSHOT1200
set align_board_yaw = 90
set yaw_motors_reversed = ON
set small_angle = 180
set deadband = 5
set yaw_deadband = 5
set pid_process_denom = 1
set osd_warn_esc_fail = OFF
set osd_warn_core_temp = OFF
set osd_rssi_alarm = 40
set osd_cap_alarm = 1300
set osd_tim1 = 1281
set osd_tim2 = 2560
set osd_vbat_pos = 2401
set osd_rssi_pos = 2413
set osd_tim_1_pos = 2454
set osd_tim_2_pos = 33
set osd_flymode_pos = 2081
set osd_throttle_pos = 2443
set osd_vtx_channel_pos = 2103
set osd_crosshairs_pos = 2253
set osd_current_pos = 2433
set osd_mah_drawn_pos = 2423
set osd_craft_name_pos = 2091
set osd_gps_speed_pos = 218
set osd_gps_lon_pos = 82
set osd_gps_lat_pos = 65
set osd_gps_sats_pos = 51
set osd_home_dir_pos = 302
set osd_home_dist_pos = 303
set osd_compass_bar_pos = 266
set osd_altitude_pos = 247
set osd_pid_roll_pos = 423
set osd_pid_pitch_pos = 455
set osd_pid_yaw_pos = 487
set osd_debug_pos = 1
set osd_power_pos = 321
set osd_pidrate_profile_pos = 2085
set osd_warnings_pos = 2313
set osd_avg_cell_voltage_pos = 449
set osd_pit_ang_pos = 257
set osd_rol_ang_pos = 289
set osd_battery_usage_pos = 425
set osd_disarmed_pos = 2187
set osd_nheading_pos = 311
set osd_nvario_pos = 279
set osd_esc_tmp_pos = 82
set osd_esc_rpm_pos = 83
set osd_stat_tim_1 = ON
set osd_stat_max_spd = OFF
set osd_stat_endbatt = ON
set osd_stat_battery = ON
set osd_stat_bbox = OFF
set vtx_channel = 3
set vtx_power = 2
set vtx_low_power_disarm = ON
set vtx_freq = 5780
set vcd_video_system = NTSC

# profile
profile 0

set dterm_notch_cutoff = 0
set feedforward_transition = 15
set iterm_relax = RP
set iterm_relax_type = SETPOINT
set iterm_relax_cutoff = 12
set throttle_boost = 2
set p_pitch = 32
set i_pitch = 65
set d_pitch = 25
set f_pitch = 120
set p_roll = 30
set i_roll = 60
set d_roll = 23
set f_roll = 120
set p_yaw = 40
set i_yaw = 120
set f_yaw = 100

# profile
profile 1

# profile
profile 2

# restore original profile selection
profile 0

# rateprofile
rateprofile 0

set roll_rc_rate = 80
set pitch_rc_rate = 80
set yaw_rc_rate = 80
set tpa_rate = 15
set tpa_breakpoint = 1050

# rateprofile
rateprofile 1

# rateprofile
rateprofile 2

# rateprofile
rateprofile 3

# rateprofile
rateprofile 4

# rateprofile
rateprofile 5

# restore original rateprofile selection
rateprofile 0

# save configuration
save

brunoleaomaia, 2019-01-11:
`# reset configuration to default settings
defaults nosave

# name
name MAIAFPV

# resources
resource MOTOR 1 C07
resource MOTOR 2 C09
resource MOTOR 3 C06
resource MOTOR 4 C08
resource MOTOR 5 NONE
resource MOTOR 6 NONE
resource LED_STRIP 1 A15
resource SERIAL_TX 2 NONE
resource SERIAL_TX 11 A02
resource CAMERA_CONTROL 1 A08

# mixer

# servo

# servo mix

# feature
feature -SOFTSERIAL

# beeper
beeper -ON_USB

# beacon
beacon RX_LOST
beacon RX_SET

# map

# serial
serial 2 2048 115200 57600 0 115200

# led

# color

# mode_color

# aux
aux 0 0 0 1300 2100 0 0
aux 1 1 2 900 1300 0 0
aux 2 13 1 1700 2100 0 0
aux 3 28 2 1700 2100 0 0

# adjrange

# rxrange

# vtx

# rxfail
rxfail 5 s 1000
rxfail 7 s 1000

# master
set acc_calibration = 9,82,-42
set mag_hardware = NONE
set baro_hardware = NONE
set min_check = 1030
set max_check = 1995
set rssi_channel = 8
set rc_smoothing_type = FILTER
set motor_pwm_protocol = DSHOT1200
set align_board_yaw = 90
set yaw_motors_reversed = ON
set small_angle = 180
set deadband = 5
set yaw_deadband = 5
set pid_process_denom = 1
set osd_warn_esc_fail = OFF
set osd_warn_core_temp = OFF
set osd_rssi_alarm = 40
set osd_cap_alarm = 1300
set osd_tim1 = 1281
set osd_tim2 = 2560
set osd_vbat_pos = 2401
set osd_rssi_pos = 2413
set osd_tim_1_pos = 2454
set osd_tim_2_pos = 33
set osd_flymode_pos = 2081
set osd_throttle_pos = 2443
set osd_vtx_channel_pos = 2103
set osd_crosshairs_pos = 2253
set osd_current_pos = 2433
set osd_mah_drawn_pos = 2423
set osd_craft_name_pos = 2091
set osd_gps_speed_pos = 218
set osd_gps_lon_pos = 82
set osd_gps_lat_pos = 65
set osd_gps_sats_pos = 51
set osd_home_dir_pos = 302
set osd_home_dist_pos = 303
set osd_compass_bar_pos = 266
set osd_altitude_pos = 247
set osd_pid_roll_pos = 423
set osd_pid_pitch_pos = 455
set osd_pid_yaw_pos = 487
set osd_debug_pos = 1
set osd_power_pos = 321
set osd_pidrate_profile_pos = 2085
set osd_warnings_pos = 2313
set osd_avg_cell_voltage_pos = 449
set osd_pit_ang_pos = 257
set osd_rol_ang_pos = 289
set osd_battery_usage_pos = 425
set osd_disarmed_pos = 2187
set osd_nheading_pos = 311
set osd_nvario_pos = 279
set osd_esc_tmp_pos = 82
set osd_esc_rpm_pos = 83
set osd_stat_tim_1 = ON
set osd_stat_max_spd = OFF
set osd_stat_endbatt = ON
set osd_stat_battery = ON
set osd_stat_bbox = OFF
set vtx_channel = 3
set vtx_power = 2
set vtx_low_power_disarm = ON
set vtx_freq = 5780
set vcd_video_system = NTSC

# profile
profile 0

set dterm_notch_cutoff = 0
set feedforward_transition = 15
set iterm_relax = RP
set iterm_relax_type = SETPOINT
set iterm_relax_cutoff = 12
set throttle_boost = 2
set p_pitch = 32
set i_pitch = 65
set d_pitch = 25
set f_pitch = 120
set p_roll = 30
set i_roll = 60
set d_roll = 23
set f_roll = 120
set p_yaw = 40
set i_yaw = 120
set f_yaw = 100

# profile
profile 1

# profile
profile 2

# restore original profile selection
profile 0

# rateprofile
rateprofile 0

set roll_rc_rate = 80
set pitch_rc_rate = 80
set yaw_rc_rate = 80
set tpa_rate = 15
set tpa_breakpoint = 1050

# rateprofile
rateprofile 1

# rateprofile
rateprofile 2

# rateprofile
rateprofile 3

# rateprofile
rateprofile 4

# rateprofile
rateprofile 5

# restore original rateprofile selection
rateprofile 0

# save configuration
save`

brunoleaomaia, 2019-01-11:
```
# reset configuration to default settings
defaults nosave

# name
name MAIAFPV

# resources
resource MOTOR 1 C07
resource MOTOR 2 C09
resource MOTOR 3 C06
resource MOTOR 4 C08
resource MOTOR 5 NONE
resource MOTOR 6 NONE
resource LED_STRIP 1 A15
resource SERIAL_TX 2 NONE
resource SERIAL_TX 11 A02
resource CAMERA_CONTROL 1 A08

# mixer

# servo

# servo mix

# feature
feature -SOFTSERIAL

# beeper
beeper -ON_USB

# beacon
beacon RX_LOST
beacon RX_SET

# map

# serial
serial 2 2048 115200 57600 0 115200

# led

# color

# mode_color

# aux
aux 0 0 0 1300 2100 0 0
aux 1 1 2 900 1300 0 0
aux 2 13 1 1700 2100 0 0
aux 3 28 2 1700 2100 0 0

# adjrange

# rxrange

# vtx

# rxfail
rxfail 5 s 1000
rxfail 7 s 1000

# master
set acc_calibration = 9,82,-42
set mag_hardware = NONE
set baro_hardware = NONE
set min_check = 1030
set max_check = 1995
set rssi_channel = 8
set rc_smoothing_type = FILTER
set motor_pwm_protocol = DSHOT1200
set align_board_yaw = 90
set yaw_motors_reversed = ON
set small_angle = 180
set deadband = 5
set yaw_deadband = 5
set pid_process_denom = 1
set osd_warn_esc_fail = OFF
set osd_warn_core_temp = OFF
set osd_rssi_alarm = 40
set osd_cap_alarm = 1300
set osd_tim1 = 1281
set osd_tim2 = 2560
set osd_vbat_pos = 2401
set osd_rssi_pos = 2413
set osd_tim_1_pos = 2454
set osd_tim_2_pos = 33
set osd_flymode_pos = 2081
set osd_throttle_pos = 2443
set osd_vtx_channel_pos = 2103
set osd_crosshairs_pos = 2253
set osd_current_pos = 2433
set osd_mah_drawn_pos = 2423
set osd_craft_name_pos = 2091
set osd_gps_speed_pos = 218
set osd_gps_lon_pos = 82
set osd_gps_lat_pos = 65
set osd_gps_sats_pos = 51
set osd_home_dir_pos = 302
set osd_home_dist_pos = 303
set osd_compass_bar_pos = 266
set osd_altitude_pos = 247
set osd_pid_roll_pos = 423
set osd_pid_pitch_pos = 455
set osd_pid_yaw_pos = 487
set osd_debug_pos = 1
set osd_power_pos = 321
set osd_pidrate_profile_pos = 2085
set osd_warnings_pos = 2313
set osd_avg_cell_voltage_pos = 449
set osd_pit_ang_pos = 257
set osd_rol_ang_pos = 289
set osd_battery_usage_pos = 425
set osd_disarmed_pos = 2187
set osd_nheading_pos = 311
set osd_nvario_pos = 279
set osd_esc_tmp_pos = 82
set osd_esc_rpm_pos = 83
set osd_stat_tim_1 = ON
set osd_stat_max_spd = OFF
set osd_stat_endbatt = ON
set osd_stat_battery = ON
set osd_stat_bbox = OFF
set vtx_channel = 3
set vtx_power = 2
set vtx_low_power_disarm = ON
set vtx_freq = 5780
set vcd_video_system = NTSC

# profile
profile 0

set dterm_notch_cutoff = 0
set feedforward_transition = 15
set iterm_relax = RP
set iterm_relax_type = SETPOINT
set iterm_relax_cutoff = 12
set throttle_boost = 2
set p_pitch = 32
set i_pitch = 65
set d_pitch = 25
set f_pitch = 120
set p_roll = 30
set i_roll = 60
set d_roll = 23
set f_roll = 120
set p_yaw = 40
set i_yaw = 120
set f_yaw = 100

# profile
profile 1

# profile
profile 2

# restore original profile selection
profile 0

# rateprofile
rateprofile 0

set roll_rc_rate = 80
set pitch_rc_rate = 80
set yaw_rc_rate = 80
set tpa_rate = 15
set tpa_breakpoint = 1050

# rateprofile
rateprofile 1

# rateprofile
rateprofile 2

# rateprofile
rateprofile 3

# rateprofile
rateprofile 4

# rateprofile
rateprofile 5

# restore original rateprofile selection
rateprofile 0

# save configuration
save
```

neon-xx, 2019-01-25:
Having the same issue here on an Airbot Omnibus f4 nano running omnibusf4sd target.
Motors work fine until remapped. Issue seems to have started at version 1329, works fine on previous builds.

[eclair.txt](https://github.com/betaflight/betaflight/files/2794821/eclair.txt)

kajsilander78, 2019-02-12:
Issue also on kakute f4 v2

Motors fine with:
resource MOTOR 1 B00
resource MOTOR 2 B01
resource MOTOR 3 A03
resource MOTOR 4 A02

with:

resource MOTOR 1 B01
resource MOTOR 2 A02
resource MOTOR 3 B00
resource MOTOR 4 A03

Motor 3 not working.

kajsilander78, 2019-02-12:
dshot_burst = OFF seems to fix it

goodnerdboy, 2019-03-04:
This issue happens with clracing F7 too, without any remapping. Build 1487.

Changing Dhot_burst's value doesn't change anything.

Issue seemed to appear in build 1469.
Version 1468 and earlier, motor 4 works like a charm with Dshot.

[diffCLracingF7.txt](https://github.com/betaflight/betaflight/files/2928409/diffCLracingF7.txt)

closed 2019-04-10 by stale[bot]