betaflight/betaflight #7376: Motors stay at idle after arming regardless of stick input - RPM_Filter/dshot_bidir=on

smasslich, 2019-01-12:
After arming, the motors continue to idle regardless of stick input.  

Flashed build 1341 to FC.  Set dshot_birdir=on, left rpm_filter settings as default.  Set gyro update and PID loop to 4k/4k.  Disabled dshot beeper.  ESC/motor protocol set to dshot1200.  set motor_poles = 12. Flashed esc's with BLHeli_32 Test code Rev32.61 

blheli_32 settings:  48 khz pwm frequency and auto timing

```diff all

# version
# Betaflight / OMNIBUSF4 (OBF4) 4.0.0 Jan 10 2019 / 15:44:42 (ccea11bfc) MSP API: 1.41

board_name OMNIBUSF4
manufacturer_id AIRB
mcu_id 003a005c3436510937333338
signature 

# reset configuration to default settings
defaults nosave

# name
name Brat1

# resources

# dmaopt

# mixer

# servo

# servo mix

# feature
feature -RX_PARALLEL_PWM
feature RX_SERIAL

# beeper
beeper -GYRO_CALIBRATED
beeper -RX_LOST
beeper -RX_LOST_LANDING
beeper -DISARMING
beeper -ARMING
beeper -ARMING_GPS_FIX
beeper -BAT_CRIT_LOW
beeper -BAT_LOW
beeper -GPS_STATUS
beeper -RX_SET
beeper -ACC_CALIBRATION
beeper -ACC_CALIBRATION_FAIL
beeper -READY_BEEP
beeper -DISARM_REPEAT
beeper -ARMED
beeper -SYSTEM_INIT
beeper -ON_USB
beeper -BLACKBOX_ERASE
beeper -CRASH_FLIP
beeper -CAM_CONNECTION_OPEN
beeper -CAM_CONNECTION_CLOSE

# beacon

# map

# serial
serial 0 64 115200 57600 0 115200
serial 2 8192 115200 57600 0 115200

# led

# color

# mode_color

# aux
aux 0 0 0 1450 2100 0 0
aux 1 13 1 1950 2100 0 0
aux 2 28 0 1950 2100 0 0
aux 3 35 2 1950 2100 0 0

# adjrange

# rxrange

# vtx

# rxfail

# master
set gyro_sync_denom = 2
set dyn_lpf_gyro_max_hz = 650
set acc_hardware = NONE
set mag_hardware = NONE
set baro_hardware = NONE
set min_check = 1025
set max_check = 1975
set rssi_channel = 8
set serialrx_provider = SBUS
set blackbox_p_ratio = 64
set dshot_idle_value = 450
set dshot_bidir = ON
set motor_pwm_protocol = DSHOT1200
set motor_poles = 12
set small_angle = 180
set deadband = 2
set yaw_deadband = 2
set pid_process_denom = 1
set osd_units = IMPERIAL
set osd_warn_batt_not_full = OFF
set osd_warn_batt_warning = OFF
set osd_warn_visual_beeper = OFF
set osd_warn_esc_fail = OFF
set osd_warn_core_temp = OFF
set osd_warn_fail_safe = OFF
set osd_vbat_pos = 2433
set osd_rssi_pos = 2458
set osd_tim_1_pos = 2071
set osd_vtx_channel_pos = 2450
set osd_mah_drawn_pos = 393
set osd_warnings_pos = 2055
set osd_avg_cell_voltage_pos = 2441
set osd_stat_tim_1 = ON
set osd_stat_tim_2 = OFF
set osd_stat_max_spd = OFF
set osd_stat_endbatt = ON
set osd_stat_min_rssi = OFF
set osd_stat_max_curr = OFF
set osd_stat_used_mah = OFF
set debug_mode = DSHOT_TELEMETRY
set vtx_band = 5
set vtx_channel = 5
set vtx_power = 4
set vtx_freq = 5806
set vcd_video_system = NTSC

# profile
profile 0

set p_pitch = 36
set d_pitch = 20
set f_pitch = 100
set p_roll = 34
set i_roll = 70
set d_roll = 19
set f_roll = 100
set p_yaw = 30

# profile
profile 1

# profile
profile 2

# restore original profile selection
profile 0

# rateprofile
rateprofile 0

set roll_rc_rate = 144
set pitch_rc_rate = 144
set yaw_rc_rate = 141
set roll_srate = 72
set pitch_srate = 72
set yaw_srate = 62

# rateprofile
rateprofile 1

# rateprofile
rateprofile 2

# rateprofile
rateprofile 3

# rateprofile
rateprofile 4

# rateprofile
rateprofile 5

# restore original rateprofile selection
rateprofile 0

# save configuration
save`

FC:  HGLRC XJB F440  (OMNIBUSF4 target)
ESC:  40A BL32_40A 3-4S ESC BLHeli_32  (HGLRC_DinoShot_40A_ST_Multi_32_61 esc FW)
Motors:  HYPERLITE 1407-4022

 also have the same issue with the following:

FC:  MATEK F405-OSD (MATEKF405 target)
ESC:  Spedix GS30A blheli_32 (Spedix_GS30A_V1_1_Multi_32_61 FW)
motors:  Hypetrain 2306-2450 (v1)

Additional notes:  
-Quad won't arm, angle arming flag when accelerometer is enabled and small angle set to 180.
-when gyro/PID loop is 2k/2k, motors idle for a sec and then stop when armed on the MATEKF405 target
-when I connect battery, connect usb, get status from cli I see dshot invalid packets
-Quad flies if I use proshot1000, however, 1 motor goes to max and it kind of desyncs at high throttle.  
-Quad flies fine with full throttle range when dshot_bidir=OFF.  
-Tried builds 1341, 1343, and 1333 - same issue
[logs-status output.zip](https://github.com/betaflight/betaflight/files/2751782/logs-status.output.zip)

dronejunkie, 2019-01-12:
I had the same issue on build #1338 and 1336. Fc target is omnibusf4sd.   Motors stay idle regardless of stick input.  Bbl log shown none zero P and d error but P and d were   zero all the time.  This only happen with bidir on.  I was running dshot1200 and 4k4k.

dronejunkie, 2019-01-12:
[
[image]
](url)

dronejunkie, 2019-01-12:
I have tried etracer dshot arming fix same issue but once I enable acc quad was respond

ITSFPV, 2019-01-12:
I had the exact same issue with the following;
FC: Kakute F4v2 AIO
ESC: 35a Tekko32 Individual ESCs
Motors: Cobra 2207-2400kv (14 poles)

On onther quad, slightly different issue but related. I would get 1 motor not arming, or spinning up in BLHeli Suite Motor test.
FC: Kakute F4v2
ESC: 35a Tekko32 4-1 ESC
Motors:  T-Motor F80 Kv2200 (14 poles)

joelucid, 2019-01-12:
Would appreciate if someone with this issue could test if it still exists with this fix: https://github.com/betaflight/betaflight/pull/7378

ITSFPV, 2019-01-12:
so had to  disable Dshot beacon, disable GPS, set angle to 180 and max overclock CPU (240) in order to get it to arm reliably.

smasslich, 2019-01-12:
> @smasslich you should have 14 poles not 12.. there are 12 pole motors out there but I’m 99% sure your not using em.

The  HYPERLITE 1407-4022 do have 12 poles according to the manufacturer.

smasslich, 2019-01-12:
flashed the matekf405 scheduler fix and can fly now with dshot.  Log below.  Thanks!

[Switch-RPM_Scheduler_fix-Dshot1200-LOG1.zip](https://github.com/betaflight/betaflight/files/2752411/Switch-RPM_Scheduler_fix-Dshot1200-LOG1.zip)

smasslich, 2019-01-12:
The HGLRC FC (omnibusf4) also worked with the schedule fix.  Log below, disregard the propwash, etc.  not tuned.
[Brat1-Scheduler_fix-Dshot1200-LOG1.zip](https://github.com/betaflight/betaflight/files/2752414/Brat1-Scheduler_fix-Dshot1200-LOG1.zip)

smasslich, 2019-01-14:
Fixed with:  https://github.com/betaflight/betaflight/pull/7378

closed 2019-01-14 by smasslich