betaflight/betaflight #7379: Bidirectional DSHOT and RPM Filter not working with music on esc's

Ipoc78, 2019-01-12:
**Describe the bug**
I was unable to get the status in the CLI to show 0 RPM 
# status
Dshot reads: 11
Dshot invalid pkts: 1
Dshot irq micros: 3
Dshot RPM Motor 0: 4095
Dshot RPM Motor 1: 4095
Dshot RPM Motor 2: 4095
Dshot RPM Motor 3: 4095

***After removing the music from esc's ***
Dshot reads: 232997
Dshot invalid pkts: 0
Dshot irq micros: 3
Dshot RPM Motor 0: 0
Dshot RPM Motor 1: 0
Dshot RPM Motor 2: 0   
Dshot RPM Motor 3: 0

**Flight controller configuration**
JBF4  (Clracingf4 target)
set 4k/4k
Dshot-1200
Bolt32

# diff all

# version
# Betaflight / CLRACINGF4 (CLR4) 4.0.0 Jan 12 2019 / 15:22:57 (51a437c26) MSP API: 1.41

board_name CLRACINGF4
manufacturer_id
mcu_id 004200483235511030333937
signature

# reset configuration to default settings
defaults nosave

# name
name dolphin-1

# feature
feature TELEMETRY
feature ESC_SENSOR

# serial
serial 0 64 115200 57600 0 115200
serial 2 32 115200 57600 0 115200
serial 3 1024 115200 57600 0 115200
serial 5 2048 115200 57600 0 115200

# aux
aux 0 0 0 900 1225 0 0
aux 1 2 1 1400 1650 0 0
aux 2 13 1 900 1075 0 0
aux 3 26 2 1725 2100 0 0
aux 4 28 2 1325 2100 0 0

# master
set gyro_sync_denom = 2
set acc_calibration = -67,34,163
set min_check = 1010
set max_check = 1990
set rssi_channel = 8
set serialrx_provider = SBUS
set blackbox_p_ratio = 64
set blackbox_device = SPIFLASH
set dshot_bidir = ON
set motor_pwm_protocol = DSHOT1200
set bat_capacity = 1300
set current_meter = ESC
set beeper_frequency = 0
set beeper_dshot_beacon_tone = 2
set yaw_motors_reversed = ON
set small_angle = 180
set pid_process_denom = 1
set tlm_halfduplex = OFF
set osd_units = IMPERIAL
set osd_warn_core_temp = OFF
set osd_rssi_alarm = 30
set osd_cap_alarm = 0
set osd_alt_alarm = 0
set osd_vbat_pos = 0
set osd_rssi_pos = 2275
set osd_tim_1_pos = 54
set osd_tim_2_pos = 2456
set osd_flymode_pos = 2445
set osd_throttle_pos = 2413
set osd_vtx_channel_pos = 377
set osd_ah_pos = 200
set osd_current_pos = 2432
set osd_mah_drawn_pos = 2081
set osd_craft_name_pos = 2059
set osd_gps_speed_pos = 218
set osd_gps_lon_pos = 82
set osd_gps_lat_pos = 65
set osd_gps_sats_pos = 51
set osd_home_dir_pos = 302
set osd_home_dist_pos = 303
set osd_compass_bar_pos = 266
set osd_altitude_pos = 247
set osd_pid_roll_pos = 423
set osd_pid_pitch_pos = 455
set osd_pid_yaw_pos = 487
set osd_debug_pos = 1
set osd_power_pos = 321
set osd_pidrate_profile_pos = 345
set osd_warnings_pos = 2090
set osd_avg_cell_voltage_pos = 2241
set osd_pit_ang_pos = 257
set osd_rol_ang_pos = 289
set osd_battery_usage_pos = 392
set osd_disarmed_pos = 2422
set osd_nheading_pos = 311
set osd_esc_tmp_pos = 82
set osd_esc_rpm_pos = 2400
set osd_stat_max_spd = OFF
set osd_stat_min_batt = OFF
set osd_stat_bbox = OFF
set osd_stat_max_fft = ON
set debug_mode = RPM_FILTER
set vtx_band = 3
set vtx_power = 3
set vtx_freq = 5705
set vcd_video_system = NTSC

# profile
profile 0

set iterm_rotation = OFF
set iterm_relax_type = GYRO
set throttle_boost = 8

# restore original profile selection
profile 0

# rateprofile
rateprofile 0

set thr_mid = 0
set thr_expo = 1
set roll_rc_rate = 80
set pitch_rc_rate = 80
set yaw_rc_rate = 80
set roll_srate = 84
set pitch_srate = 84
set yaw_srate = 80

and by default 
# get dshot
dshot_idle_value = 550
Allowed range: 0 - 2000

dshot_burst = OFF
Allowed values: OFF, ON

dshot_bidir = ON
Allowed values: OFF, ON
Default value: OFF

fujin, 2019-02-12:
@Ipoc78 please log an issue upstream so @sskaug can track it: https://github.com/bitdump/BLHeli/issues/new

closed 2019-02-12 by fujin

Ipoc78, 2019-02-19:
Copy that.