betaflight/betaflight #7386: BARO Missing for Target MKF7

darkmatter2222, 2019-01-13:
**Is your feature request related to a problem? Please describe.**
Target MKF7 (Matek F722-STD) Does have a barometer onboard however BARO not visible in Configurator (10.4.0). 

**Describe the solution you'd like**
Enable BARO for Target MKF7.

jflyper, 2019-01-13:
@darkmatter2222 

Which version of Betaflight software are you dealing with?

This is with all defaults.
[image]

fujin, 2019-01-13:
Not able to reproduce.
1) Flash latest BF build to MK72
2) 'Baro' box (top right) is enabled by default
  [image]
3) Sensors tab 'barometer' field records values
  [image]
Please post your `status` and `diff all`.

closed 2019-01-13 by jflyper

darkmatter2222, 2019-01-13:
My mistake, I wasn't clear enough.
BF 3.5.4
I am not seeing the Baro on the modules tab.
[image]
[image]
[image]
[image]

darkmatter2222, 2019-01-13:
Modes* tab. Baro is missing

jflyper, 2019-01-14:
@darkmatter2222 

Ah. Sorry, BARO mode was dropped.