betaflight/betaflight #7394: Launch Controll didn't display in OSD

MrOmbre, 2019-01-14:
I tested this in Built 1339-1345 and by active Launch Controll Mode nothing is displayed in OSD.
Warnings are active.

etracer65, 2019-01-14:
As requested when you opened the issue, provide a `diff` of your settings and details about your setup.

>**Describe the bug**
A clear and concise description of what the bug is.
>
>**To Reproduce**
>Steps to reproduce the behavior:
>
>**Expected behavior**
>A clear and concise description of what you expected to happen.
>
>**Flight controller configuration**
>Create a `diff` and post it here in a code block. Put ``` (three backticks) at the start and end of the `diff` block (instructions  on how to do a diff: https://oscarliang.com/use-diff-not-dump-betaflight/)
>
>**Setup / Versions**
> - Flight controller [what type is it, where was it bought from];
> - Other components [RX, VTX, brand / model for all of them, firmware version where applicable];
 > - how are the different components wired up.
>
>**Additional context**
>Add any other context about the problem here.

etracer65, 2019-01-17:
Tested latest builds and the launch control indicator properly displays in the OSD warnings element as designed. Closing due to lack of response to requests for more information.

closed 2019-01-17 by etracer65