betaflight/betaflight #7401: Wiki Page for blinking/Error Codes

exocode, 2019-01-15:
**Is your feature request related to a problem? Please describe.**
In order to solve errors on controllers running Betaflight, there should be a list, where the (mostly red) blinking LEDs on the controller mean.

So I have the case, that my FlightController (SP Racing F3 VI) start blinking red (5 times very fast, then four times slow). If someone else has this problem, so logically I searched the betaflight Wiki for status codes, error codes, flashing Led,...) But I didn't find anything. 

**Describe the solution you'd like**
Someone with the fitting knowledge could write it.

**Describe alternatives you've considered**
Searched google didn't find a comprehensive list.

etracer65, 2019-01-15:
It's already in the wiki.

https://github.com/betaflight/betaflight/wiki/Frequently-Asked-Questions#why-does-my-flight-controller-blinkbeep-lots-of-times-when-powering-up-

exocode, 2019-01-15:
shame on me. didn't watched the faqs. sorry for making noise!

closed 2019-01-15 by exocode