betaflight/betaflight #7407: builds 1355-1360 corrupt or black out my osd

butterSnake, 2019-01-16:
**Describe the bug**
A clear and concise description of what the bug is.

**To Reproduce**
Steps to reproduce the behavior:

**Expected behavior**
A clear and concise description of what you expected to happen.

**Flight controller configuration**
Create a `diff` and post it here in a code block. Put ``` (three backticks) at the start and end of the `diff` block (instructions  on how to do a diff: https://oscarliang.com/use-diff-not-dump-betaflight/)

**Setup / Versions**
 - Flight controller [what type is it, where was it bought from];
 - Other components [RX, VTX, brand / model for all of them, firmware version where applicable];
 - how are the different components wired up.

**Additional context**
Add any other context about the problem here.

fujin, 2019-01-17:
@butterSnake & @iCr4sh have tracked this down to `video_type=PAL` setting in diff, although @butterSnake has PAL cameras.

w/ video type `AUTO`, osd works (?)

I have advised how to get logs while in auto mode:
```
set video_type = AUTO
set debug_mode = MAX7456_SIGNAL
set blackbox_mode = ALWAYS
```

butterSnake, 2019-01-18:
@fujin should i close this since we got to the bottom of it?

closed 2019-02-05 by butterSnake