betaflight/betaflight #7408: SD Card Issues since msc_use_persistence_layer

DavidBoone, 2019-01-16:
**Describe the bug**
`sd_info` reveals the following error: `Filesystem: Fatal - no FAT MBR partitions`

**To Reproduce**
Install betaflight from commit d4f164b3ccf1d23d8d4eef9a32ca63aa4f9a08fd or later.  Usually (though not 100%, periodically after a power cycle it reports OK) the Blackbox tab in the configurator shows an error and `sd_info` command in CLI shows above error.

**Expected behavior**
`sd_info` shows `Filesystem: Ready`

**Flight controller configuration**
```board_name CLRACINGF4
manufacturer_id 

# name
name Martioone

# resources

# dmaopt

# mixer

# servo

# servo mix

# feature
feature TELEMETRY
feature AIRMODE
feature ESC_SENSOR

# beeper

# beacon
beacon RX_SET

# map
map TAER1234

# serial
serial 0 2048 115200 57600 0 115200
serial 3 64 115200 57600 0 115200
serial 5 1024 115200 57600 0 115200

# led

# color

# mode_color

# aux
aux 0 0 0 1700 2100 0 0
aux 1 1 1 1300 1700 0 0
aux 2 13 2 1700 2100 0 0
aux 3 35 1 1700 2100 0 0

# adjrange

# rxrange

# vtx

# rxfail

# master
set gyro_sync_denom = 2
set gyro_lowpass_hz = 200
set min_check = 1010
set max_check = 1990
set serialrx_provider = CRSF
set blackbox_p_ratio = 128
set dshot_bidir = ON
set motor_pwm_protocol = DSHOT600
set current_meter = ESC
set beeper_frequency = 0
set yaw_motors_reversed = ON
set small_angle = 180
set pid_process_denom = 1
set tlm_halfduplex = OFF
set osd_warn_no_gps_rescue = OFF
set osd_warn_core_temp = OFF
set osd_warn_rc_smoothing = OFF
set osd_warn_fail_safe = OFF
set osd_warn_launch_control = OFF
set osd_vbat_pos = 54
set osd_rssi_pos = 2083
set osd_tim_1_pos = 54
set osd_tim_2_pos = 2134
set osd_flymode_pos = 408
set osd_throttle_pos = 2423
set osd_current_pos = 2401
set osd_mah_drawn_pos = 2434
set osd_craft_name_pos = 2445
set osd_warnings_pos = 329
set osd_avg_cell_voltage_pos = 2102
set osd_disarmed_pos = 2346
set osd_stat_max_fft = ON
set debug_mode = GYRO_SCALED
set vtx_band = 5
set vtx_channel = 2
set vtx_power = 2
set vtx_freq = 5695
set vcd_video_system = NTSC

# profile
profile 0

set dterm_lowpass_hz = 0
set dterm_lowpass2_hz = 0

# rateprofile
rateprofile 0

set roll_rc_rate = 120
set pitch_rc_rate = 120
set yaw_rc_rate = 130
set roll_srate = 75
set pitch_srate = 75
set yaw_srate = 75
```

**Setup / Versions**
 - CLRacing F4S
 - Tested with 8GB and 32GB SD cards installed and formatted

**Additional context**
I'm building from source.
 - e1d7948b0a66fec70b4c584e4fd63e644c2cc565 works fine
 - d4f164b3ccf1d23d8d4eef9a32ca63aa4f9a08fd has the problem

Additionally I had success using the latest master (8962e2ccd3e00d16329ad039e09f561290121e33) with d4f164b3ccf1d23d8d4eef9a32ca63aa4f9a08fd reverted.

mikeller, 2019-01-16:
@DavidBoone: What did you use to format the SD cards?

mikeller, 2019-01-16:
@DavidBoone: Can you try again using this utility: https://www.sdcard.org/downloads/formatter_4/eula_mac/

(In general it's a good idea to format SD cards with it, it uses a special geometry that is optimised for writing streamed data onto an SD card.)

DavidBoone, 2019-01-16:
Latest results:

```Flash (full chip erase), do *not* apply diff, straight from firmware upgrade to Connect/CLI/sd_info

1349: worked 4x in a row
1355: 1 failure, then worked 3x in a row, then 3x failures
1349: worked 5x in a row
```

DavidBoone, 2019-01-16:
I'm now suspecting 492d1f323eab653c94048b9ffc7169fd732e1720 (Merge pull request #6809 from jflyper/bfdev-spi-transaction).  I really need to get some sleep before morning, I will attempt further testing after the day-job.

mikeller, 2019-01-16:
Good night!

Well possible that the change to the SPI driver implementation is behind this. Unfortunately, the SPI emulation used by a lot of current SD cards seems to be flaky, and susceptible to minor changes in things like timing, making them work very intermittently, and for some models not at all.

etracer65, 2019-01-16:
I can't replicate this. Tried OMNIBUSF4SD, AIRBOTF4SD and MATEKF722. All recognize the SD card every time and report no errors. Also tested `msc` mode (on MacOS) and worked perfectly in each case.

Also #7390 is unlikely to be the problem as it is only involved in regards to how the MCU reboots - either normally, into boot loader, or into mass storage mode (msc). It's can't have any impact on the card not working or `sd_info` reporting errors when already booted into the firmware.

Note there are other PR's included in build 1357 that may be the cause. For example #7383.

So here are 2 firmwares that are granular to the commits to try. This first one only has #7390.
[betaflight_4.0.0_CLRACINGF4_PR7390.hex.zip](https://github.com/betaflight/betaflight/files/2764772/betaflight_4.0.0_CLRACINGF4_PR7390.hex.zip)

The second has #7390 and #7383 
[betaflight_4.0.0_CLRACINGF4_PR7390_PR7383.hex.zip](https://github.com/betaflight/betaflight/files/2764773/betaflight_4.0.0_CLRACINGF4_PR7390_PR7383.hex.zip)

See how these behave. If it's still intermittent with both then I'm inclined to blame the card or possibly something target specific (unlikely).

jflyper, 2019-01-16:
@DavidBoone When the issue appeared, would you try re-inserting the card few times and see if it makes any change?

DavidBoone, 2019-01-16:
@etracer65 I will try those two firmwares you linked and report back.

@jflyper will do, see if re-insertion of card has any influence on the issue.

Will be ~8 hours before I am able to test and report back.  Thanks

DavidBoone, 2019-01-17:
@jflyper: Both your hexes had the same result, 100% success rate recognizing the SD card.  Process was, unplug/replug USB, check status on Blackbox page of configurator, repeat 10x.

betaflight_4.0.0_CLRACINGF4_PR7390.hex: 10/10 successful
betaflight_4.0.0_CLRACINGF4_PR7390_PR7383.hex: 10/10 successful

From source, I built two images, master w/ #6809 merged, and the commit immediately before:
492d1f323 "merged-pr6809": 2/10 successful (BAD)
7f58ecc77 "before-pr6809": 10/10 successful (GOOD)
492d1f323 "merged-pr6809": 2/10 successful (BAD)
7f58ecc77 "before-pr6809": 5/5 successful (GOOD)

@jflyper SD card re-insertion did not affect above results.

mikeller, 2019-01-17:
Thanks for testing and reporting back @DavidBoone!

jflyper, 2019-01-18:
I'm looking into this.

PR #7439 (2019-01-22): [SDCARD] Quick fix: Discard excess/bogus data in DR at prologue of spiTransferByte
Fixes #7408 

This non-intrusive fix provides a quick and pragmatic fix for SD card not recognized after shared SPI code merge.

It appears that how SD card driver under shared SPI code interact with the card leaves one extra data byte in receive data register (`(SPI_TypeDef *)instance->DR`) which causes immediate return of `sdcard_waitForIdle` and `sdcard_waitForNonIdleByte` to malfunction.

A root cause of this phenomena is yet unknown, but clearing the extra data by dummy reading the `DR` at the beginning o f [... cut]
closed 2019-01-22 by mikeller