betaflight/betaflight #7411: Corrupt vtx_freq config.

AndersHoglund, 2019-01-16:
**Describe the bug**
Jenkins build 1360 and later. When having set default config values, either by `defaults `command or by full chip erase. you get this when doing the `diff `command:

> set vtx_freq = -22025 ###CORRUPTED CONFIG###

Also when setting some explicit value,  it is accepted but when saving and rebooting it also becomes corrupted:

```
# set vtx_freq=0
vtx_freq set to 0
# get vtx_freq
vtx_freq = 0
Allowed range: 0 - 5999
Default value: 5740

# set vtx_freq=5806
vtx_freq set to 5806
# get vtx_freq
vtx_freq = 5806
Allowed range: 0 - 5999
Default value: 5740
save

# get vtx_freq
vtx_freq = -22025 ###CORRUPTED CONFIG###
Allowed range: 0 - 5999
Default value: 5740

```

 
**To Reproduce**
Flash Jenkins build 1360 or later with full chip erase selected.
Do a CLI diff command.
Build 1358 is OK. 

**Expected behavior**
Non-corrupt config.

**Flight controller configuration**
Default configuration.

**Setup / Versions**
Fc: SpeedyBeeF4
Rx: Spektrum 4649T
Vtx: Tramp

```
# version
# Betaflight / SPEEDYBEEF4 (SBF4) 4.0.0 Jan 15 2019 / 23:36:06 (eec46d4f8) MSP API: 1.41

```

Prime suspect: PR #7285

jflyper, 2019-01-16:
Ah... This is the type of problem with migrating from the current vtx system to user defined table scheme...

The legacy (original) vtx system assumed the standard band-channel-frequency, and it is referenced in multiple ways; (a) vtx common and vtx io system was referencing the table directly, and (b) vtx device drivers had a pointer to the table in their respective `vtxDevice_s` structures.

When I did the #7285, I thought having the per device frequency table (among other device specific parameters) is a very sensible approach, as it will allow multiple different device with different frequency table to co-exist in a single system. In this scheme, band-channel-frequency can only be looked up for a device normally after the device is initialized.

However, semantics of the `vtx_xxx` CLI variables, that is, to give persistent storage for band-channel-frequency does not allow the lookup before the device is initialized. `vtxInit` called from `init` does this lookup to sync frequency from band/channel before any vtx device is initialized.

Let me give this situation a thought. Any comment/question is welcome.

jflyper, 2019-01-16:
I think the only sensible solution to the chicken-and-egg problem is to call `vtxInit` from respective device initialization, immediately after `vtxDevice_s` is filled with device specific pariameters.

With this change:

1. The `vtx_xxx` CLI variables can be synced using parameters specific to whatever the device configured (at run time) and initialized.

2.  If no device is initialized, the `vtx_xxx` variables will remain untouched.

Let me create a PR with this change.

phobos-, 2019-01-16:
I can confirm the same behaviour with crazybeef3fs target compiled against recent master.

jflyper, 2019-01-16:
Yes. It's common to all and every target with vtx features enabled. There's no more report necessary. I know exactly what's going on.

TheIsotopes, 2019-01-18:
@jflyper can confirm this but for me it is even worse 
my new matek f7 mini has not booted since build `#1360` ... last build without probs is `#1358`

hope #7415 will fix this and merged asap

jflyper, 2019-01-18:
@TheIsotopes What's Matek F722-mini?

TheIsotopes, 2019-01-18:
this one [FLIGHT CONTROLLER F722-MINI](http://www.mateksys.com/?portfolio=f722-mini#tab-id-3)
same like Matek F722-SE but with onboard flash instead sdcard and mounting 20 x 20mm

TheIsotopes, 2019-01-19:
tested `build #1370` and now all works perfect ... thx 👍