betaflight/betaflight #7413: SPRACINGF7DUAL wedging on boot with #7395

etracer65, 2019-01-16:
**Describe the bug**
PR @7395 makes SPRACINGF7DUAL wedge on boot

**To Reproduce**
Flash build 1361 or later

**Additional context**
Problem not replicated with other flight controllers (such as MATEKF722, OMNIBUSF4SD).

Verified that the previous commit does not have the issue.

etracer65, 2019-01-16:
Basically what I found so far is that SPRACINGF7DUAL wedges on boot starting with the commit for GPS Rescue disabled warning #7395. So I've been beating on it and it's not making any sense. Can't find anything wrong and nonsensical things are happening. Commenting out random things or putting in some debugging (even in unrelated areas) will sometimes make the problem go away. So on a lark I just tried undef'ing some features and boom, no more problem.  Basically if I disable even something small like `USE_SMART_FEEDFORWARD` (just something small) then it boots normally. So I'm thinking that we're overflowing something, but the post compile report doesn't raise any alarm bells. The other odd thing is that a couple of times when I was testing and booting failed I left it for about 30 seconds or so. I then had the device popup in STM FS mode (I have a USB arbitrator running that shows when devices connect). So not sure how that happened. Also wondering if maybe we're stepping over some RAM boundary region possibly corrupting the persistent block? But not sure if it really tried to boot in mass storage or just some half-borked USB startup.

etracer65, 2019-01-16:
For reference, here's an compile report from a version that wedges (master):
```
Memory region         Used Size  Region Size  %age Used
        ITCM_RAM:       14344 B        16 KB     87.55%
      ITCM_FLASH:          0 GB        16 KB      0.00%
ITCM_FLASH_CONFIG:          0 GB        16 KB      0.00%
     ITCM_FLASH1:          0 GB       480 KB      0.00%
      AXIM_FLASH:       14698 B        16 KB     89.71%
AXIM_FLASH_CONFIG:          0 GB        16 KB      0.00%
     AXIM_FLASH1:      362272 B       480 KB     73.70%
        DTCM_RAM:       17948 B        64 KB     27.39%
           SRAM1:       50076 B       176 KB     27.79%
           SRAM2:          0 GB        16 KB      0.00%
       MEMORY_B1:          0 GB         0 GB       nan%
   text	   data	    bss	    dec	    hex	filename
 364470	  12500	  55520	 432490	  6996a	./obj/main/betaflight_SPRACINGF7DUAL.elf
```
And this is with something small removed (`#undef USE_SMART_FEEDFORWARD`) and has a normal boot:
```
Memory region         Used Size  Region Size  %age Used
        ITCM_RAM:       14288 B        16 KB     87.21%
      ITCM_FLASH:          0 GB        16 KB      0.00%
ITCM_FLASH_CONFIG:          0 GB        16 KB      0.00%
     ITCM_FLASH1:          0 GB       480 KB      0.00%
      AXIM_FLASH:       14700 B        16 KB     89.72%
AXIM_FLASH_CONFIG:          0 GB        16 KB      0.00%
     AXIM_FLASH1:      362048 B       480 KB     73.66%
        DTCM_RAM:       17944 B        64 KB     27.38%
           SRAM1:       50080 B       176 KB     27.79%
           SRAM2:          0 GB        16 KB      0.00%
       MEMORY_B1:          0 GB         0 GB       nan%
   text	   data	    bss	    dec	    hex	filename
 364248	  12500	  55520	 432268	  6988c	./obj/main/betaflight_SPRACINGF7DUAL.elf
```

etracer65, 2019-01-17:
Ok, I've nailed it down to a problem with ITCM_RAM. The region is defined to be 16K, but as soon as we exceed 14K of use then we have boot wedges. Exactly 2K (2048) bytes appear to be reserved or otherwise used (or maybe offsets are wrong) - don't know enough about the hardware to say.

Confirmed that this is not specific to SPRACINGF722DUAL. Testing with other F722 boards reveals the same problem. It just looks like the SPRACINGF722DUAL got to the limit first.

#7395 is not directly a cause and there are no problems with it. It just happened to be the commit that pushed ITCM_RAM usage over 14K for this target.

@jflyper @DieHertz Any thoughts?

hydra, 2019-01-18:
How about writing a bit of code that verifies all 16KB of ITCM ram can be read/written and verified?  Do it just before the .tcm_code block is copied to ITCM ram in init.c.

hydra, 2019-01-18:
also, what happens if you offset the content of the ITCM ram, does the 14kb of code still work when it's offset?

Set the memory address of ITCM ram to values other than 0x00000000 and decrease the length of it accordingly, see `stm32_flash_f722.ld`. e.g.

```
ITCM_RAM (rx)           : ORIGIN = 0x00000800, LENGTH = 14K
```

hydra, 2019-01-20:
that's odd since the linker script, values it generates and code should all match up. it should work.  if it works with less than 12k then it certainly indicates a boundary issue.

jflyper, 2019-01-22:
```
ITCM_RAM (rx)           : ORIGIN = 0x00000800, LENGTH = 14K
```

This can be done with MATEKF722, and boots normally.

```

jflyper, 2019-01-22:
@etracer65 What is the CPU doing when it failed to boot?

mikeller, 2019-02-28:
@etracer65: Is this still a problem?

etracer65, 2019-02-28:
@mikeller I can no longer replicate. Which is concerning since nothing was actually done to address. When I originally reported the problem I was able to show that even one byte over 14K would cause the boot wedging - not just for SPRACINGF7DUAL, but for other F722's as well. Still not sure if the 14K part (on a 16K address space) is relevant or just exposing some other problem. But concerning in that it went away without any changes that I'm aware of that should have fixed it.

mikeller, 2019-02-28:
@etracer65: Agreed that this is concerning. Are you going to do more analysis?

etracer65, 2019-02-28:
I don't have the tools (or skill) to do this kind of low level hardware debugging. But at least let me see if I can bisect where it starts to work again. I just confirmed that the build I noted above (1361) still does have the problem. So I'll try and figure out where it was "fixed" and maybe that'll shed some light on the issue.

etracer65, 2019-02-28:
Well it just gets stranger! Pulling the 1361 build from the Jenkins server for SPRACINGF722DUAL shows the boot wedge (as does 1362, 1363, etc.). But building locally from the same commit as 1361 (commit ec71965dd83c8133fe9e1d9e17df48b9a5896e3f) doesn't have the problem! And I know that at the time I was able to replicate it locally!

Checking the latest builds on the Jenkins server shows that they're okay. So I have no idea what's going on.

hydra, 2019-03-02:
check the build logs for 1361 from jenkins, double check your compiler version matches.

etracer65, 2019-03-02:
@hydra The strange(r) thing is I know for a fact that at the time I could replicate the problem locally. And I haven't updated the compiler. Can you try build 1361 on SPRACINGF7DUAL to confirm the boot wedging just as an extra data point?

https://ci.betaflight.tech/job/Betaflight/1361/artifact/obj/

MJ666, 2019-03-03:
Not sure if this is related? This is what I can experience with AFNGF7 boards since quite a while already. The board never boots properly after a new flash of the firmware as soon the eeprom needs to be reinitialized. Sometimes it will come up with an unknown USB device after a while or really nothing will happen. If I repower the board it will initialize properly. It takes a bit longer if it is the first time after flashing. I can see this behavior with the native target and also with the STM32F7X2 target. All later reboots even with configuration changes will work as expected with no issues. Most of the time I need to reconnect with the configurator manually after the reboot since it will not reconnect automatically. In some cases the configurator is even not connecting and a full configurator restart is needed to be able to connect. No additional action for the FC is needed in this case. Not a big deal since you can workaround this easily but this may indicate some bigger problem. 

I have two AFNGF7 based micro brushed copters which fly without any issues. But my only  AFNGF7 brushless copter (135mm) has dramatic yaw twitches and I never got it to fly properly. All of this FC's use the ICM20602 Gyro. For the twitching one i already used another new Gyro chip with the same results. Motor 1 and 4 get burning hot within seconds. Running motors individualy in the configurator they all run smooth and i can not see real vibrations in the gyro traces. I have a very similar build with an F4 based FC and MPU6000 which works with no issues. But this is may be something for an different BF issue.

awolf78, 2019-03-06:
We have had a similar issue. It would just hang or crash randomly. Commenting out code made it hang/crash in different spots. Turned out it was the initialization of the HAL structs, example:
`UART_HandleTypeDef m_uartHandle;`

changed to

`UART_HandleTypeDef m_uartHandle = {};`

Doing that for all HAL init structs solved the issue. Not saying that's your issue, but maybe it helps.

hydra, 2019-03-06:
yes, probably not a bad idea as you dont want any hal init structures eith garbage.  and since there is support for more than one mcu we can't guarantee that we initialise all members to zero between mcu types.

klutvott123, 2019-03-08:
This issue probably started long before build 1361. I can only reproduce it by flashing my FC. Then my MATEKF722 disconnects, and doesn't connect again. Sometimes it is "discovered" by windows but shows up with another COM number which can't be connected to. And sometimes it takes a while before it connects again, but it doesn't work and windows just complains.

Some observations i have made.
It seems like it happens when flashing to a different firmware version. If i try flashing the same version one more time, it will successfully boot and I'm able to connect after flashing. So i have been flashing back and forth between build 1445 and 1450 and can consistently reproduce this behaviour. 

One thing that i have noticed is that the caches are enabled in ``SystemInit()``, which runs sometime before we enter the main function. It is recommended by ST to not enable the caches before entering ``main()`` as this can cause hard faults. If that happens and we enter the ISR for the hard fault, we probably end up in a ``while(1){ }``.

To test this I removed the enabling of caches from ``SystemInit()`` and placed them in ``init()`` right after copying code and data to ITCM/DTCM RAM takes place. ``init()`` is called at the top of ``main()``. I did this for 1445 and 1450. Then i flashed back and forth between them and it seems to be working. After flashing is done it reboots, is discovered by windows and I'm able to connect. I have just tested this briefly so I can't say that I'm 100% certain that this is the issue, but it seems like a probable cause.

mikeller, 2019-03-09:
Good find, @klutvott123. This runs contrary to what #7626 wants to do, so a careful evaluation is needed before this can be merged.

hydra, 2019-03-09:
@kluttvott123 please can you try #7626?

From my perspective, copying code to the right locations and initialising memory should be done as early as possible so that code and data can be used from any memory location.  I'm fine with enabling caches after this has been done, probably not a good idea to enable caches before this has happened unless you add code to invalidate cached regions which seems sub-optimal in terms of code size/complexity.

DieHertz, 2019-03-09:
Enabling I and D caches using stm-supplied functions invalidates them automatically.
Disabling also clears them.

hydra, 2019-03-09:
yup, just stating a preference when to enable caches.  i.e. after memory regions are initialised.

DieHertz, 2019-03-09:
Yes, it's a nice in the right direction.
I have a branch which gets rid of all assembly for F4/F7 and consolidates the so-called scatter loading process, which includes DTCM and ITCM RAMs.

klutvott123, 2019-03-09:
@mikeller I'm not sure I know what you mean. #7626 would also put the copying of code before enabling the caches, but the cache enabling is still done in ``SystemInit()``. #7626 should still work as intended even if enabling the caches is moved to somewhere in ``main()``?

@hydra I have tested #7626 now using the same method i described in my post further up. I'm still able to consistently reproduce the wedging when flashing between versions. Then I moved the cache enabling from ``SystemInit()`` to ``init()`` again and it seems to be working(still with #7626 applied).

But as I said before, I'm not sure if this actually fixes it or if it's just a coincidence.

mikeller, 2019-03-09:
@klutvott123: Yes, you are right, #7626 affects ITCM / DTCM RAM, and not D-cache / I-cache. Can you please open a pull request for your proposed fix, so we can get it tested on other boards to see if it causes problems there?

hydra, 2019-03-09:
@kluttvott123 thanks for testing. good to know that other PR doesn't have any effect on this issue.

please make a PR with your suggested fix.

klutvott123, 2019-03-10:
@hydra , @mikeller Done!

mikeller, 2019-03-26:
@etracer65: Is this fixed now?

etracer65, 2019-03-26:
I haven't see it for a while so I guess. Not sure we ever found a definitive answer though. Closing.

closed 2019-03-26 by etracer65