betaflight/betaflight #7416: OSD related issues - black screen of death / slow OSD task execution

fujin, 2019-01-16:
**Describe the bug**
We are tracking reports of two individual OSD issues:
* skonk has two machines that after flashing CI build 1355+ cause the OSD to generate a blank screen.
* omerco has a machine that is experiencing length (750us avg) OSD task loop time.

**To Reproduce**
* TBD

**Expected behavior**
* OSD to function as per usual.

**Flight controller configuration**
* TBC

**Setup / Versions**
* TBC

**Additional context**
I am gathering additional information and will file other issues or track them both through here.

loomerco, 2019-01-16:
**Describe the bug:**
on an OMNIBUSF4 (FLAME F4): when connected only by USB (no battery) OSD task is:
Task list rate/hz  max/us  avg/us maxload avgload     total/ms
 OSD)            59     928     734       5.9%       4.8%            254

once the battery is connected it normalizes:
 OSD)            59     112      37       1.1%         0.7%         3882
**To Reproduce**
Steps to reproduce the behavior:
1. connect USB
2. go to CLI type tasks.

**Flight controller configuration**
 ``` # version
# Betaflight / OMNIBUSF4 (OBF4) 4.0.0 Jan 15 2019 / 15:02:26 (718b29d67) MSP API: 1.41

board_name OMNIBUSF4
manufacturer_id AIRB

# name
name Crusader

# resources
resource MOTOR 1 A03
resource MOTOR 2 A02
resource MOTOR 3 B00
resource MOTOR 4 B01
resource MOTOR 5 NONE
resource MOTOR 6 NONE
resource PPM 1 NONE
resource PWM 1 NONE
resource PWM 2 NONE
resource PWM 3 NONE
resource PWM 4 NONE
resource PWM 5 NONE
resource PWM 6 NONE
resource SONAR_TRIGGER 1 NONE
resource SONAR_ECHO 1 NONE
resource LED_STRIP 1 NONE
resource SERIAL_TX 1 NONE
resource SERIAL_TX 6 NONE
resource SERIAL_RX 6 NONE
resource INVERTER 1 NONE
resource LED 1 NONE
resource ADC_CURR 1 NONE

# dmaopt

# mixer

# servo

# servo mix

# feature
feature -RX_SPI
feature RX_SERIAL
feature MOTOR_STOP

# beeper
beeper -ARMING_GPS_FIX
beeper -GPS_STATUS
beeper -ARMED
beeper -ON_USB

# beacon
beacon RX_LOST

# map

# serial
serial 0 64 115200 57600 0 115200
serial 2 1024 115200 57600 0 115200

# led

# color

# mode_color

# aux
aux 0 13 1 1700 2100 0 0
aux 1 26 0 1300 1700 0 0
aux 2 28 0 1300 1700 0 0
aux 3 35 1 1650 2100 0 0

# adjrange
adjrange 0 0 2 900 1300 6 4 0 0
adjrange 1 0 2 1300 1700 7 4 0 0
adjrange 2 0 2 1700 2100 8 4 0 0

# rxrange

# vtx

# rxfail

# master
set align_gyro = CW0FLIP
set gyro_sync_denom = 2
set gyro_lowpass_type = PT1
set gyro_lowpass_hz = 0
set yaw_spin_threshold = 1200
set dyn_lpf_gyro_max_hz = 0
set align_acc = CW0FLIP
set acc_hardware = NONE
set mag_hardware = NONE
set baro_hardware = NONE
set min_check = 1005
set max_aux_channels = 2
set serialrx_provider = SPEK2048
set airmode_start_throttle_percent = 10
set blackbox_p_ratio = 64
set blackbox_record_acc = OFF
set min_throttle = 1060
set dshot_idle_value = 600
set dshot_bidir = ON
set motor_pwm_protocol = DSHOT1200
set failsafe_delay = 10
set failsafe_throttle = 1150
set failsafe_throttle_low_delay = 50
set failsafe_procedure = AUTO-LAND
set vbat_min_cell_voltage = 350
set current_meter = NONE
set vbat_detect_cell_voltage = 30
set vbat_scale = 111
set beeper_dshot_beacon_tone = 3
set yaw_motors_reversed = ON
set small_angle = 180
set auto_disarm_delay = 0
set pid_process_denom = 1
set runaway_takeoff_deactivate_throttle_percent = 10
set osd_vbat_pos = 2508
set osd_current_pos = 384
set osd_warnings_pos = 2473
set osd_esc_tmp_pos = 417
set osd_esc_rpm_pos = 439
set debug_mode = D_CUT
set scheduler_optimize_rate = ON

# profile
profile 0

set dterm_lowpass_type = PT1
set dterm_lowpass_hz = 120
set dterm_lowpass2_hz = 0
set p_pitch = 55
set i_pitch = 65
set d_pitch = 45
set f_pitch = 160
set p_roll = 55
set d_roll = 45
set f_roll = 160

# rateprofile
rateprofile 0

set roll_expo = 10
set pitch_expo = 10
set yaw_expo = 10
set roll_srate = 77
set pitch_srate = 77
set yaw_srate = 77
set tpa_rate = 40
set tpa_mode = PD ``` 

**Setup / Versions**
 - OMNIBUSF4 (FLAME F4). bought from banggood.
 - Other components: Spectrum SAT, regular Vtx (non serial). straight simple wiring via the onboard OSD.

dadoribeiro, 2019-01-16:
My FC is a DALRCF722:

Task with only USB:
Task list             rate/hz  max/us  avg/us maxload avgload     total/ms
00 - (         SYSTEM)      9       1       0    0.5%    0.0%         0
01 - (         SYSTEM)    995       2       0    0.6%    0.0%        25
02 - (       GYRO/PID)   7839      45      34   35.7%   27.1%     16963
03 - (            ACC)    995      12      10    1.6%    1.4%       609
04 - (       ATTITUDE)     99      12      10    0.6%    0.5%        65
05 - (             RX)     32      47      44    0.6%    0.6%        97
06 - (         SERIAL)    100     623       1    6.7%    0.5%        18
08 - (BATTERY_VOLTAGE)     49       3       2    0.5%    0.5%         6
09 - (BATTERY_CURRENT)     49       3       0    0.5%    0.0%         3
10 - ( BATTERY_ALERTS)      5       3       2    0.5%    0.5%         0
11 - (         BEEPER)     99       6       1    0.5%    0.5%         8
19 - (            OSD)     59     927     753    5.9%    4.9%      2502
21 - (            CMS)     60       2       1    0.5%    0.5%         3
22 - (        VTXCTRL)      5       2       1    0.5%    0.5%         0
23 - (        CAMCTRL)      5       1       0    0.5%    0.0%         0
25 - (    ADCINTERNAL)      0       3       3    0.5%    0.5%         0
26 - (       PINIOBOX)     19       2       1    0.5%    0.5%         0
RX Check Function                   3       1                         2
Total (excluding SERIAL)                        50.0%   38.0%

Tasks with USB + LipoTask list             rate/hz  max/us  avg/us maxload avgload     total/ms
00 - (         SYSTEM)      9       1       0    0.5%    0.0%         0
01 - (         SYSTEM)   1000       5       1    1.0%    0.6%        16
02 - (       GYRO/PID)   7955      50      33   40.2%   26.7%     11678
03 - (            ACC)    999      18       9    2.2%    1.3%       396
04 - (       ATTITUDE)     99      13       9    0.6%    0.5%        43
05 - (             RX)     32      58      44    0.6%    0.6%        66
06 - (         SERIAL)     99      76       2    1.2%    0.5%         8
08 - (BATTERY_VOLTAGE)     49       3       2    0.5%    0.5%         4
09 - (BATTERY_CURRENT)     50       4       2    0.5%    0.5%         3
10 - ( BATTERY_ALERTS)      5       3       2    0.5%    0.5%         0
11 - (         BEEPER)     99       5       1    0.5%    0.5%         4
19 - (            OSD)     59     928      35    5.9%    0.7%      1167
21 - (            CMS)     59       2       1    0.5%    0.5%         2
22 - (        VTXCTRL)      4      48      10    0.5%    0.5%         3
23 - (        CAMCTRL)      4       1       0    0.5%    0.0%         0
25 - (    ADCINTERNAL)      0       4       2    0.5%    0.5%         0
26 - (       PINIOBOX)     20       2       1    0.5%    0.5%         0
RX Check Function                   2       1                         1
Total (excluding SERIAL)                        55.5%   34.4%

My current diff all:
# version
# Betaflight / DALRCF722DUAL (DLF7) 4.0.0 Jan 16 2019 / 13:14:54 (e0fc9300e) MSP API: 1.41

board_name DALRCF722DUAL
manufacturer_id 
mcu_id 0030002e3137510234393337
signature 

# reset configuration to default settings
defaults nosave

# name
name Rooster 2

# resources

# dmaopt

# mixer

# servo

# servo mix

# feature

# beeper

# beacon

# map
map TAER1234

# serial
serial 4 2048 115200 57600 0 115200

# led

# color

# mode_color

# aux
aux 0 0 0 1700 2100 0 0
aux 1 1 1 900 1300 0 0
aux 2 13 2 1700 2100 0 0
aux 3 28 1 1700 2100 0 0
aux 4 35 3 1700 2100 0 0
aux 5 36 3 1300 2100 0 0

# adjrange

# rxrange

# vtx

# rxfail

# master
set dyn_lpf_gyro_max_hz = 500
set acc_calibration = -34,-44,-9
set mag_hardware = NONE
set baro_hardware = NONE
set rssi_channel = 16
set fpv_mix_degrees = 30
set blackbox_p_ratio = 64
set min_throttle = 1050
set dshot_idle_value = 750
set motor_pwm_protocol = DSHOT1200
set failsafe_delay = 8
set vbat_warning_cell_voltage = 340
set vbat_scale = 165
set ibata_scale = 147
set small_angle = 180
set pid_process_denom = 1
set osd_warn_core_temp = OFF
set osd_rssi_alarm = 70
set osd_cap_alarm = 1100
set osd_vbat_pos = 2423
set osd_rssi_pos = 2413
set osd_tim_1_pos = 54
set osd_tim_2_pos = 2401
set osd_flymode_pos = 2445
set osd_throttle_pos = 2101
set osd_vtx_channel_pos = 377
set osd_crosshairs_pos = 2253
set osd_ah_pos = 200
set osd_current_pos = 0
set osd_mah_drawn_pos = 2082
set osd_craft_name_pos = 362
set osd_gps_speed_pos = 218
set osd_gps_lon_pos = 82
set osd_gps_lat_pos = 65
set osd_gps_sats_pos = 51
set osd_home_dir_pos = 302
set osd_home_dist_pos = 303
set osd_compass_bar_pos = 266
set osd_altitude_pos = 247
set osd_pid_roll_pos = 0
set osd_pid_pitch_pos = 0
set osd_pid_yaw_pos = 0
set osd_debug_pos = 1
set osd_power_pos = 321
set osd_pidrate_profile_pos = 345
set osd_warnings_pos = 2313
set osd_avg_cell_voltage_pos = 76
set osd_pit_ang_pos = 257
set osd_rol_ang_pos = 289
set osd_battery_usage_pos = 392
set osd_disarmed_pos = 138
set osd_nheading_pos = 3
set osd_stat_max_spd = OFF
set osd_stat_used_mah = OFF
set osd_stat_bbox = OFF
set osd_stat_bb_no = OFF
set osd_stat_max_fft = ON
set debug_mode = D_CUT
set vtx_band = 5
set vtx_power = 4
set vtx_low_power_disarm = ON
set vtx_freq = 0
set vcd_video_system = NTSC

# profile
profile 0

set dterm_lowpass2_type = PT1
set anti_gravity_gain = 7500
set feedforward_transition = 63
set iterm_relax_type = GYRO
set p_pitch = 49
set d_pitch = 50
set f_pitch = 65
set i_roll = 73
set d_roll = 47
set f_roll = 65
set dterm_cut_gain = 16

# profile
profile 1

# profile
profile 2

# restore original profile selection
profile 0

# rateprofile
rateprofile 0

set roll_rc_rate = 130
set pitch_rc_rate = 130
set yaw_rc_rate = 130
set roll_expo = 35
set pitch_expo = 35
set yaw_expo = 28
set roll_srate = 66
set pitch_srate = 66
set yaw_srate = 64

butterSnake, 2019-01-16:
Mine is here 

https://github.com/betaflight/betaflight/issues/7407

vkolotov, 2019-01-16:
Not sure if it is related really, please don't take it into account as I'm not sure if it is FW related, but I have flashed #1353 and I had two consequent blackouts after that, never had that before. 

However, it might be due to some hardware failures, I suspect VTX board was too close to FC board so the channel button on VTX was pushed accidentally in flight and caused a channel switch. I've added more spacers after that, had a few flights with no issues so far.

etracer65, 2019-01-16:
Can't replicate the task average issue on OMNIBUSF4SD even using the diffs above. Task time is around 32us regardless of battery connected or not.

etracer65, 2019-01-16:
For those that can replicate, try going back to some older builds to see if it's still happening. If so then we need to try an narrow down which build starts having the problem.

McGiverGim, 2019-01-17:
I have checked against MATEK F722SE and the max is high, but the average is correct, under USB:
```
# version
# Betaflight / MATEKF722SE (MF7S) 4.0.0 Jan 14 2019 / 13:41:11 (234186f49) MSP API: 1.41
# tasks
Task list             rate/hz  max/us  avg/us maxload avgload     total/ms
00 - (         SYSTEM)      9       1       0    0.5%    0.0%         0
01 - (         SYSTEM)   1000    1746       2  175.1%    0.7%        11
02 - (       GYRO/PID)   8000      51      35   41.3%   28.5%       722
03 - (            ACC)   1000      12       9    1.7%    1.4%        33
04 - (       ATTITUDE)    498      13       9    1.1%    0.9%        17
05 - (             RX)     32      64      46    0.7%    0.6%         5
06 - (         SERIAL)     99      91       2    1.4%    0.5%         2
08 - (BATTERY_VOLTAGE)     49       4       2    0.5%    0.5%         0
09 - (BATTERY_CURRENT)     50       2       1    0.5%    0.5%         0
10 - ( BATTERY_ALERTS)      4       2       1    0.5%    0.5%         0
11 - (         BEEPER)     99       3       2    0.5%    0.5%         0
12 - (            GPS)     99      42      16    0.9%    0.6%         4
13 - (        COMPASS)      9     134      95    0.6%    0.5%         5
14 - (           BARO)     43      98      65    0.9%    0.7%        10
15 - (       ALTITUDE)     40       7       5    0.5%    0.5%         0
17 - (      TELEMETRY)    250      36       0    1.4%    0.0%        14
19 - (            OSD)     59     677      61    4.4%    0.8%         6
20 - (     ESC_SENSOR)     99       2       1    0.5%    0.5%         0
21 - (            CMS)     60       2       1    0.5%    0.5%         0
22 - (        VTXCTRL)      4      24       4    0.5%    0.5%         0
23 - (        CAMCTRL)      4       1       0    0.5%    0.0%         0
24 - (       RCDEVICE)     20      11       3    0.5%    0.5%         0
25 - (    ADCINTERNAL)      0       4       0    0.5%    0.0%         0
26 - (       PINIOBOX)     19       3       2    0.5%    0.5%         0
RX Check Function                   3       1                         0
Total (excluding SERIAL)                       234.6%   39.7%
```

etracer65, 2019-01-17:
Related to the high average OSD task time when on USB: This is just a hardware design issue. It appears that the problematic boards are not supplying power to the max7456 chip when on USB (either partial or completely). The long task times are due to the communication with the max7456 timing out. On the DALRCF722DUAL I confirmed that the max7456 isn't being detected when powered only by USB, but other boards that don't experience this problem like the Matek722-STD/SE, real OMNIBUSF4SD, etc. are in fact detecting the max7456 and the OSD does actually work when only powered by USB. I have a test setup with one of these with rx and 5v vtx powered from the board and I test OSD stuff all the time when only connected to USB.

So don't think there's anything we can do about this and it's not really a problem - just a limitation of the particular board design.

vkolotov, 2019-01-17:
Did a few flights since I added more spacers between FC and VTX, no blackouts since then. Please ignore my comment above re blackouts.

butterSnake, 2019-01-22:
ok so i'm running 4.0 build 1365 on a couple of quads and i have some osd issues still different to the last ones.. so the osd drops in and out and flickers..  it can be brought on by using save and reboot from the stick menu.. it also happens some times on power up.. power cycling a few times fixes it

TheFlaash, 2019-02-24:
Hi i do also have a Problem with my osd. Its not working on 4.0 daily updated builds. it only flickers sometimes. Board is Kakute F4 V2. I tried Ntsc/Auto.
# status
MCU F40X Clock=168MHz (PLLP-HSE), Vref=3.30V, Core temp=50degC
Stack size: 2048, Stack address: 0x1000fff0
Config size: 3222, Max available config: 16384
GYRO=ICM20689
System Uptime: 12 seconds, Current Time: 0000-01-01T00:00:00.000+00:00
CPU:8%, cycle time: 128, GYRO rate: 7812, RX rate: 33, System rate: 9
Voltage: 3 * 0.01V (0S battery - NOT PRESENT)
I2C Errors: 0
Arming disable flags: RXLOSS CLI MSP

# version
# Betaflight / KAKUTEF4V2 (KTV2) 4.0.0 Feb 17 2019 / 13:49:09 (966971c75) MSP API: 1.41

# dump all

# version
# Betaflight / KAKUTEF4V2 (KTV2) 4.0.0 Feb 17 2019 / 13:49:09 (966971c75) MSP API: 1.41

board_name KAKUTEF4V2
manufacturer_id 
mcu_id 004f002a4848500120373743
signature 

# name
name FLASH

# display_name
display_name -

# resources
resource BEEPER 1 C09
resource MOTOR 1 B00
resource MOTOR 2 B01
resource MOTOR 3 A03
resource MOTOR 4 A02
resource MOTOR 5 NONE
resource MOTOR 6 NONE
resource MOTOR 7 NONE
resource MOTOR 8 NONE
resource SERVO 1 NONE
resource SERVO 2 NONE
resource SERVO 3 NONE
resource SERVO 4 NONE
resource SERVO 5 NONE
resource SERVO 6 NONE
resource SERVO 7 NONE
resource SERVO 8 NONE
resource PPM 1 C07
resource PWM 1 NONE
resource PWM 2 NONE
resource PWM 3 NONE
resource PWM 4 NONE
resource PWM 5 NONE
resource PWM 6 NONE
resource PWM 7 NONE
resource PWM 8 NONE
resource LED_STRIP 1 C08
resource SERIAL_TX 1 A09
resource SERIAL_TX 2 NONE
resource SERIAL_TX 3 B10
resource SERIAL_TX 4 A00
resource SERIAL_TX 5 NONE
resource SERIAL_TX 6 C06
resource SERIAL_TX 7 NONE
resource SERIAL_TX 8 NONE
resource SERIAL_TX 9 NONE
resource SERIAL_TX 10 NONE
resource SERIAL_TX 11 NONE
[... 1161 lines cut ...]
rateprofile 5

set thr_mid = 50
set thr_expo = 0
set rates_type = BETAFLIGHT
set roll_rc_rate = 100
set pitch_rc_rate = 100
set yaw_rc_rate = 100
set roll_expo = 0
set pitch_expo = 0
set yaw_expo = 0
set roll_srate = 70
set pitch_srate = 70
set yaw_srate = 70
set tpa_rate = 50
set tpa_breakpoint = 1500
set tpa_mode = D
set throttle_limit_type = OFF
set throttle_limit_percent = 100
set roll_rate_limit = 1998
set pitch_rate_limit = 1998
set yaw_rate_limit = 1998

# restore original rateprofile selection
rateprofile 0

closed 2019-04-02 by stale[bot]