betaflight/betaflight #7418: LED Strip Battery Function Not Working

wjk1129, 2019-01-16:
We are trying to get the Battery voltage to change the color shown on the LED strip. I am told that this feature worked in version 3.2. We are using version 3.4

Have two separate batteries one at low voltage and one fully charged. Create 3 numbered LEDs in the lower right corner of the grid. Select the 3 squares and select the battery function.

We are getting solid green LEDs for both the low voltage battery and full battery. We are able to change the color of LEDs using the color function and the arm state function works.

Any difference as battery goes from High to Low is expected.

jflyper, 2019-01-17:
@wjk1129 Is this related to a similar attempt described in RCG's RMDO thread?

jflyper, 2019-01-17:
@wjk1129 Did you read the RCG thread? We need you to provide us with how you are powering your board.

wjk1129, 2019-01-17:
I just responded in the RCGroup thread that we are powering from a Power Distribution Board. I think 6v. I do not think that the DoDo accepts 14.8-4s power input anywhere on the board yet it is reading 14.8 roughly. Thankyou for helping.

wjk1129, 2019-01-23:
The good news is that it is working now on 4S power. Starts off Green, as voltage goes down, goes to yellow and then red. Almost completely depleted it goes to flashing green. (3% battery left)

Is that expected behavior?  Any options.  I have played with the voltage levels a little in the BetaFlight software.  Thank you everyone.

jflyper, 2019-01-24:
I think the flashing green at CRITICAL is a bug.

closed 2019-01-24 by wjk1129

jflyper, 2019-01-25:
@wjk1129 The color problem will be fixed by #7460.