betaflight/betaflight #7434: Compile error for USE_LED_STRIP and F3

fgiudice98, 2019-01-19:
Compile error for F3 fc with added USE_LED_STRIP (tested on OMNIBUS and FURYF3)
I get this error after the pull of the latest 4 merges.

```./src/main/interface/settings.c:1072:86: error: 'TABLE_RGB_GRB' undeclared here (not in a function); did you mean 'TABLE_DEBUG'?
     { "ledstrip_grb_rgb",           VAR_UINT8  | MASTER_VALUE | MODE_LOOKUP, .config.lookup = { TABLE_RGB_GRB }, PG_LED_STRIP_CONFIG, offsetof(ledStripConfig_t, ledstrip_grb_rgb) },
                                                                                      ^~~~~~~~~~~~~
                                                                                      TABLE_DEBUG
```
Removing the #define USE_LED_STRIP solves it

fgiudice98, 2019-01-19:
Ok, I figured out:
Compiling with USE_LED_STRIP only give the error, thus stopping all;
Compiling with USE_LED_STRIP and USE_LED_STRIP_STATUS_MODE results in a successful build;
Compiling using only USE_LED_STRIP_STATUS_MODE results in a successful build but smaller than the one with USE_LED_STRIP
Maybe this should be fixed in the Led strip status code or in the common_post.h file

etracer65, 2019-01-19:
@fgiudice98 Thanks for the report. Should be fixed in #7436

fgiudice98, 2019-01-19:
@etracer65 You're welcome and thanks for the fix!

PR #7436 (2019-01-20): Fix LEDSTRIP defines to prevent compilation errors
Fixes #7434 

Compile would fail if `USE_LED_STRIP` was defined but `USE_LED_STRIP_STATUS_MODE` wasn't.

Introduced in #7303
closed 2019-01-20 by mikeller