betaflight/betaflight #7437: BF4.0.0 - Cycle time varies enormously

TheIsotopes, 2019-01-20:
**Describe the bug**
Cycle time varies enormously from 125 to 240 ... but not on default settings 

**To Reproduce**
since `build #1371` and newer and LED Strip enabled

**Flight controller configuration**
```
# Betaflight / MATEKF405 (MKF4) 4.0.0 Jan 19 2019 / 12:31:46 (628fdb8ad) MSP API: 1.41

# name
name Matek

# feature
feature -SOFTSERIAL
feature -TELEMETRY
feature -OSD
feature LED_STRIP

# map
map TAER1234

# led
led 0 9,2::AW:0
led 1 9,3::AW:0
led 2 9,4::AW:0
led 3 9,5::AW:0
led 4 6,5::AW:0
led 5 6,4::AW:0
led 6 6,3::AW:0
led 7 6,2::AW:0

# mode_color
mode_color 6 0 3
mode_color 6 1 6

# aux
aux 0 0 0 1600 2000 0 0
aux 1 13 1 1000 1200 0 0
aux 2 26 0 1600 1800 0 0
aux 3 31 0 1200 1400 0 0
aux 4 36 1 1800 2000 0 0

# master
set align_gyro = CW180
set align_acc = CW180
set acc_hardware = NONE
set mag_hardware = NONE
set baro_hardware = NONE
set serialrx_provider = SUMD
set blackbox_device = SPIFLASH
set motor_pwm_protocol = DSHOT600
set pid_process_denom = 1
```

etracer65, 2019-01-20:
Yes, there appears to be a problem with excessive CPU utilization for the ledstrip code. We'll look into it.

etracer65, 2019-01-21:
@TheIsotopes #7442 should fix this problem. If possible can you please test to confirm?

[betaflight_4.0.0_MATEKF405.hex.zip](https://github.com/betaflight/betaflight/files/2777217/betaflight_4.0.0_MATEKF405.hex.zip)

TheIsotopes, 2019-01-21:
@etracer65 tested, fixed and confirmed ... thx 👍

PR #7442 (2019-01-22): Refactor ledstrip profiles to fix excessive cpu load
Fixes #7437 

Previous logic bypassed optimizations and updated the ws2811 on every task execution - leading to an average of 92us task execution time. Restored the previous optimizations and additionally optimized the new "RACE" and "BEACON" modes so that they only update the led's when there's an actual state change. Task time for these modes is now in the 2-4us range for F405. For the normal "STATUS" profile the performance has been restored to previous optimized levels and typical configurations will generally have task times in the 8-14 [... cut]
closed 2019-01-22 by mikeller