betaflight/betaflight #7440: Build 1356 breaks Onboard SD card on Matek F405

0crap, 2019-01-20:
Build 1356 gives "Fatal error Reboot to retry" on my onboard SD card for Blackbox. (Red icon.)
# version
# Betaflight / MATEKF405 (MKF4) 4.0.0 Jan 15 2019 / 09:05:16 (521db5473) MSP API: 1.41

Build 1355 is fine and SD card works correctly. (Green icon.)
I flashed multiple times, with consisten results, between 1355 and 1356 with full chip erase.
All default config, No settings changed.

0crap, 2019-01-20:
Note, this is when powered from USB.
No LiPo connected.
I'll check if LiPo connected makes any difference.

0crap, 2019-01-20:
I created a custom build from the current master and only reverted the two changes done in build 1356.
src/main/sensors/current.c
src/main/sensors/voltage.h
That doesn't bring the SD Card back to life. :-(

mikeller, 2019-01-20:
Duplicate of #7408.

closed 2019-01-20 by mikeller

0crap, 2019-01-20:
@jflyper @mikeller Tested jflyper change in bus_spi_stdperiph.c that fixes my problem as well.
Thx!

BlackBird11, 2020-07-28:
> @jflyper @mikeller Tested jflyper change in bus_spi_stdperiph.c that fixes my problem as well.
> Thx!

how exactly do i change that? having the same problem on 4.2.1 on clracing f4s - thank you very much in advance!