betaflight/betaflight #7452: Noxe F4 not connecting after enabling ledstrip

fgiudice98, 2019-01-22:
Flashing the last build (1379) and enabling LED_STRIP result in fc not connecting to pc
For fixing I have to reflash and do a full chip erase
The same will happen if I flash 3.5 and make 8/8 (not tested in 4.0[1379])
I think it's about the cpu usage (should not have been corrected in 1378?)

jflyper, 2019-01-22:
Does not reproduce with genuine Airbot NOX flashed with ci build #1379 (c39ea81b6).
You should start writing to your distributor.

[image]

[image]

[image]

fgiudice98, 2019-01-24:
I did some testing. The LED_STRIP feature can be enabled, the problem is when PPM receiver and LED_STRIP are turned on at the same time. When I do this the board locks at the betaflight logo then, as I said, it must be flashed with full chip erase.

mikeller, 2019-01-25:
From `src/main/target/NOX/target.c`:

```
DEF_TIM(TIM2,  CH3,  PB10, TIM_USE_PPM,   0,     0), // T2C3
[...]
DEF_TIM(TIM2,  CH1,  PA0,  TIM_USE_LED,   0, 0), // T2C1(1,5,3)
```

Looks like PPM and LED_STRIP are using the same timer, which won't work. This seems to be a shortcoming of this target's design. Might be possible to find alternate timers for one of the two.

jflyper, 2019-01-25:
> Might be possible to find alternate timers for one of the two.

- PPM (PB10) can be TIM2
- LED strip (PA0) can be TIM2 or TIM5
- Motor 1 (PA1) can be TIM2 or TIM5

Since there are only two timers for three functions, one of the functions must be sacrificed.

mikeller, 2019-01-25:
Who designs this sh*t?

jflyper, 2019-01-25:
Who knows 😏 

One thing I know is that Jack Ma is wrong.

> Jack Ma : "Fakes: Better Quality and Better Price Than the Real Names"

fgiudice98, 2019-01-25:
You think it's possible to change the led_strip timer?

fgiudice98, 2019-01-25:
If it's possible to change the led_strip (or the ppm) definition to use another timer (for using led_strip and ppm at the same time)

mikeller, 2019-01-25:
@fgiudice98: Not if you want to use motor 1 as well - if you can do without motor 1 then that is possible.

fgiudice98, 2019-01-26:
@mikeller Just checked the voltage, 4.67V
I will try something and let you know

fgiudice98, 2019-01-26:
@mikeller Added a diode, now the voltage is 4.08V, same behavior

fgiudice98, 2019-01-26:
@mikeller I think I found the problem for the led strip. I think I falshed 20 times my fc with different builds. The led strip work, except when I enable Dshot (with Proshot the situation is even worse)
(by the way, the leds works with and without the diode, so the voltage isn't a problem)

joelucid, 2019-01-26:
This is a dma conflict. We changed to other timers to avoid the n channel issue. That causes a dma conflict with dma.

fgiudice98, 2019-01-26:
Fixed removing USE_DSHOT_TELEMETRY from the target
Thanks @joelucid

closed 2019-01-26 by fgiudice98