betaflight/betaflight #7453: Gyro EXTI induced looptime jitter

SteveCEvans, 2019-01-22:
Based on 62da4d09e687b14d3cb3c0153499d73003505adb

Using a Revo Nano to experiment, I used tethered logging to dump out the loop times and saw a lot of variation.

[image]

I realised that gyro interrupts were being serviced but that this did no useful work, so I undefined USE_GYRO_EXTI for this target and found the jitter to have greatly reduced.

[image]

Am I correct in thinking that jitter in the PID task timing will impact the tuning of the PIDs and filters? Should we be aiming to reduce timing jitter? When I did this on OpenPilot the result was much tighter handling.

Given the comments about not accepting new targets due to changes in 4.0 I decided to submit a bug for this as undef'ing USE_GYRO_EXTI for all targets which use it (100 in total!) might not be the way to go.

The observations above arise from investigating using DMA for gyro access. Using a Revo Nano and hooking up a logic analyser using PINIO to get accurate timings, I noted that using the current polled access took 11.67us to read a sample from the gyro, but using DMA took 14.88us of which 6.87 was waiting for the completion interrupt. There is thus some saving to be had by using DMAs, but as only 7 bytes are being read, this is probably not worth the effort.

etracer65, 2019-01-23:
The EXTI data ready interrupt currently doesn't drive the gyro loop in Betaflight. It's purely scheduler based to provide support for targets that don't have the EXTI pin connected (many clones) or not configured properly.

We are investigating ways to improve the gyro scheduling.

SteveCEvans, 2019-01-23:
Hi Bruce,

Yes, I realised it wasn’t doing anything useful. I enabled 32 kHz gyro updates and saw ~75% CPU load. Disabling the interrupt dropped this to ~50% load.

I’ve explored such issues in the past, see https://percepio.com/2015/09/29/open-pilot-next-gen-tunes-isrs-with-tracealyzer/

On OPNG I had the gyro interrupt trigger the read of the gyro data using DMA, and then the DMA completion interrupt set a binary semaphore which triggered the gyro processing task. This used FreeRTOS which used a preemptive scheduler, but the Betaflight scheduler seems rather simpler.

Also as previously mentioned I note that as we only read the gyro data, not the accel and mag data (7 vs 25 bytes IIRC), so polling rather than using the DMA engine may make more sense. That said having the gyro interrupt cascade a sequence of tightly coupled events as I’ve previously done cuts the time from the gyro detecting motion to the motor drive reacting, and virtually eliminates jitter in the control loop.

I’m keen to get involved in your discussions on this topic.

Steve

jflyper, 2019-01-23:
@SteveCEvans 

> That said having the gyro interrupt cascade a sequence of tightly coupled events

This is in consideration, in conjunction with, for example, #7325 for example. Right now, we are doing some preliminary studies on gyro chip behaviors.

For example, we are observing MPU6000 not interrupting periodically. At 8KHz sampling, it interrupts 125us a part for 6 times then 7th interrupt is 171us, and back to 125us after 79us at 8th (That is, intervals comes in set of 125, 125, 125, 125, 125, 125, 171, 79).

The observation you made might have triggered another need for an investigation; why the hell ignored  interrupts cause the loop time jitter, and why the insertion of "just return" isr stops this.

Can you look into it?

SteveCEvans, 2019-01-23:
On the Revo Nano, using an MPU9250 the interrupt stream is rock steady @8kHz, but the reads from the gyro are not.

[image]

I'll take a look at the Revo which has an MPU6000 for comparison.

Steve

etracer65, 2019-01-23:
MPU9250 is the same part as MPU6500 (+mag). It's a later design than MPU6000.

etracer65, 2019-01-23:
I just tested on a variety of F4's and F7's with both 32 and 8K gyros. No difference at all by adding an immediate `return` to the `mpuIntExtiHandler()`.

SteveCEvans, 2019-01-23:
On a Revo Nano if I enable the 32kHz gyro/2kHz PID loop operation the processor hits 100% usage and gyro reads start getting missed.

[image]

I've only just started looking at this and understanding the scheduler, so I'll have a think about what I'm actually looking at here and try to make sense first of what's coming from the hardware, and then how we're processing it.

jflyper, 2019-01-23:
@SteveCEvans
I wanted you to look into why ignored interrupt causes loop time jitter which you have brought up, not interrupt interval of various gyro chips.

SteveCEvans, 2019-01-23:
I should have clarified in the first post that this log was obtained with debug=CYCLETIME so what we're looking at is jitter in the motor timing, actually the call to subTaskMotorUpdate(). The top trace shows the achieved motor update interval in us. It should be 500 as I'm running the PID loop at 2kHz.

Using four PINIO lines I'm monitoring the timing of the mpuGyroReadSPI() & mpuIntExtiHandler() routines which represent the Gyro Update Frequency and PID loop frequency stability respectively. I simply drive the pin high on entering the routine, and low on exit.  I'm then also recording the durations of each scheduler call to a taskFunc. Finally I'm checking for timing violation where the scheduler calls a taskFunc which takes more than 120us to execute. This is a simplistic view that doesn't check for a failure to hit a desired loop time, but it does allow a rogue task to be detected.

```
        // Execute task
#if defined(USE_TASK_STATISTICS)
        if (calculateTaskStatistics) {
            const timeUs_t currentTimeBeforeTaskCall = micros();
            pinioSet(3, 1); // SCEDEBUG
            selectedTask->taskFunc(currentTimeBeforeTaskCall);
            pinioSet(3, 0); // SCEDEBUG
            const timeUs_t taskExecutionTime = micros() - currentTimeBeforeTaskCall;
            selectedTask->movingSumExecutionTime += taskExecutionTime - selectedTask->movingSumExecutionTime / MOVING_SUM_COUNT;
            selectedTask->totalExecutionTime += taskExecutionTime;   // time consumed by scheduler + task
            selectedTask->maxExecutionTime = MAX(selectedTask->maxExecutionTime, taskExecutionTime);
            selectedTask->movingAverageCycleTime += 0.05f * (period - selectedTask->movingAverageCycleTime);

            if (taskExecutionTime > 120) {
                pinioSet(2, 1); // SCEDEBUG
                pinioSet(2, 0); // SCEDEBUG
            }
        } else
#endif

```

It can be seen that timing violations are occurring about every 10ms.
[image]

Zooming in somewhat it can be seen that these occurances result in significant deviations in the loop times. 629us vs the target 500us for the PID loop and 244us vs the target of 125us for the gyro loop.

[image]

As the tethered logging is sampling at fixed intervals (I was using 100ms) which is a multiple of the 10ms noted above, then it was pure luck as to whether the 100ms samples fell exactly on the 10ms interval or not. This drifting in and out of phase can be seen the long tethered sample shown below.

[image]

Enabling the EXTI interrupt or not actually has no effect. My observation that there was a redundant interrupt being processed was not directly related to the jitter although it would be consuming some of the available processing time potentially stretching taskFunc execution time.

Setting a breakpoint to allow inspection of the selectedTask when the timing violation is noted shows the task to be one of the following:

- BARO
- SERIAL

The SERIAL task does indeed update every 10ms, but the BARO only every 50ms.

Removing the USB connection stops the timing violations for the SERIAL task.

Removing the USE_BARO define temporarily prevents all violations at 120us.

Reducing the check to 114us similarly shows no violations, but at 113us there are occasional triggers with the GYRO/PID task, but this looks acceptable.

Given that connecting USB messes things up, moving to the subTaskMotorUpdate() routine and recording the peak excess over the targetPidLooptime I saw a peak figure of 193, although it takes a while to reach that. I guess that's what we're ultimately aiming to reduce.

I'll look into the Baro issue and report back.

SteveCEvans, 2019-01-23:
See 
https://github.com/betaflight/betaflight/pull/7455

jflyper, 2019-01-24:
> Enabling the EXTI interrupt or not actually has no effect. My observation that there was a redundant interrupt being processed was not directly related to the jitter although it would be consuming some of the available processing time potentially stretching taskFunc execution time.

Are you saying this issue report is bogus?

SteveCEvans, 2019-01-24:
Yes, but in investigating I discovered a problem with the baro update task and have fixed that.

jflyper, 2019-01-24:
@SteveCEvans 

I don't understand. Is there any explanation on what caused defining `USE_GYRO_EXTI` reduced the loop time jitter?

SteveCEvans, 2019-01-24:
Yes, I believe it was caused by the periodic sampling period of the tethered logging lining up, or not, with the baro task execution. I just got repeated results by chance. As shown above, a longer tethered log shows that it samples loop time extension in bursts.

SteveCEvans, 2019-01-27:
See https://github.com/betaflight/betaflight/pull/7462

closed 2019-01-27 by SteveCEvans