betaflight/betaflight #7454: Gyro Issues BF4.0 (#1269, #1291, #1323)

TehllamaFPV, 2019-01-22:
I'm getting a fairly repeatable bug with some of my builds once migrated to 4.0.  Three quads so far, tried both on all three BF4.0 setups.  Only one of them has blackbox, which did let me elucidate some of the issue.

This is happening at idle throttle with some stick nudges, and anytime I nudge the throttle up enough to engage airmode, it instantly freaks out and causes anti-taz to kick in.

In the blackbox data, what I'm seeing is that the P/I terms seem to be getting forced to zero at some point, which is then induces a runaway oscillation.  Reducing both P and I gains significantly on an axis will delay onset of this problem, or move the exact same issue to another axis.

I was successfully able to chase this problem from roll, to pitch, to yaw, back to roll until I was running only 5/10 for P/I values.  I could take the extreme solution and intput '1', but I was hoping to actually go fly, and proceeded to toss 3.5.2 on there and all three quads proceeded to fly without issues (I concluded it wasn't a setup issues unique to my setups - especially since it was a 3", 5", and 7" quad all experiencing the same things).  So far, my fix has been to flash 3.5.2 and set the same things, and those things fly without issue.

I'll edit and attach BBL screencaps showing this - really odd, especially since I could induce this before air mode engaged.  If this is a launch mode issue, I'm thoroughly confused, because the first maneuver I attempted idling on the ground was just a basic pitch forward, and that led to an immediate anti-taz.

etracer65, 2019-01-23:
Screencaps of logs are less important than at least telling us which flight flight controller and target(s) you're using along with your settings `diff` as requested in the issue instructions. There are no other reports of these kinds of problems so it's likely something in your configuration.  Beside that you start with a full-chip erase and it's recommended you use the defaults for 4.0 with the exception of your basic setup (radio, motor protocol, etc.) and your preferred rates.

Also check your gyro orientation and make sure the movement on the Setup tab matches reality (accelerometer must be enabled). Some targets had a gyro orientation change in 4.0.

TehllamaFPV, 2019-01-23:
`diff all

# version
# Betaflight / MATEKF411 (MK41) 4.0.0 Dec 29 2018 / 09:55:42 (770e883d8) MSP API: 1.41

board_name MATEKF411
manufacturer_id 
mcu_id 006700433035510538343836
signature 

# reset configuration to default settings
defaults nosave

# name
name tehllama

# resources
resource MOTOR 1 B05
resource MOTOR 2 B04
resource MOTOR 3 B07
resource MOTOR 4 B06
resource LED_STRIP 1 NONE
resource SERIAL_TX 11 A08

# feature
feature -SOFTSERIAL

# beacon
beacon RX_SET

# serial
serial 1 8192 115200 57600 0 115200

# aux
aux 0 0 0 1700 2100 0 0
aux 1 13 1 1700 2100 0 0
aux 2 28 0 1700 2100 0 0
aux 3 35 3 1300 1700 0 0
aux 4 36 2 1700 2100 0 0
aux 5 39 3 1700 2100 0 0

# master
set dyn_notch_q = 2
set dyn_lpf_gyro_max_hz = 650
set acc_calibration = 111,-50,-261
set baro_hardware = NONE
set rssi_channel = 16
set fpv_mix_degrees = 45
set max_aux_channels = 12
set motor_pwm_protocol = DSHOT600
set failsafe_delay = 5
set failsafe_switch_mode = STAGE2
set align_board_yaw = 180
set vbat_max_cell_voltage = 440
set current_meter = VIRTUAL
set ibatv_scale = 230
set beeper_dshot_beacon_tone = 4
set yaw_motors_reversed = ON
set small_angle = 180
set pid_process_denom = 1

# profile
profile 0

set d_pitch = 16
set f_pitch = 90
set d_roll = 15
set f_roll = 90
set p_yaw = 5
set i_yaw = 120
set f_yaw = 120

# restore original profile selection
profile 0

# rateprofile
rateprofile 0

set thr_expo = 15
set roll_rc_rate = 82
set pitch_rc_rate = 82
set yaw_rc_rate = 55
set roll_srate = 67
set pitch_srate = 67
set yaw_srate = 56
set roll_rate_limit = 999
set pitch_rate_limit = 999
set yaw_rate_limit = 999

# save configuration
save
# `

etracer65, 2019-01-23:
Again, what flight controller? A number of different boards use the MATEKF411 target so that doesn't identify the hardware.

You're going to need to carefully test your configuration to confirm it's correct. You have a lot going on. You have the board rotation set to 180, the motors remapped, and the motor direction reversed - all opportunities to have something wrong and cause the behavior you're describing.

Start by verifying the motion on the Setup tab exactly matches the motion of the quad. Move it on all axes to be sure.

Next, using the Motors tab (with no props!) spin each motor and confirm the position and rotation direction match the graphic at the top of the page.

You have some issue with orientation, motor order/direction, or props on wrong. These are the things that cause the behavior you've described.

TehllamaFPV, 2019-01-23:
I realize it sounds cocky to say that I know that isn't the issue, but I know that isn't the issue.
I flashed 3.5.2, pasted the diffall into CLI, and made no other changes.  All three fly without issue.

With 4.0 enabling accelerometer, board orientation was correct (all four), motor numbering was correct (had to manually check this anyway - stupid cheap ESCs), as was direction.

jflyper, 2019-01-23:
In that case, we need a bb log.

etracer65, 2019-01-23:
Everybody always says they checked and "everything is correct"...

`set dyn_notch_q = 2` ????  The default is 120! You really should go back to defaults for all PID and filter settings.

jflyper, 2019-01-23:
Heh, we don't need a bb log from `dyn_notch_q = 2`.

TehllamaFPV, 2019-01-23:
Ha, that was a dumb typo.  It'll take me a while to re-run that setup, the intent was to run a Q of 200 and dyn_lpf_max_hz of 650Hz (this is a 3").  In plasma tree, the noise band is pretty narrow for this one, but also pretty high amplitude when it flies.

The dynamic aspects missing in 3.5.2 would be a candidate for causing something like this, but on the stock Q value, I'm out of ideas (and practically if I get the downtime I'll run the math and see if even that bad a setup could produce the sort of phase delay at really low frequency roll and yaw oscillations).

I was still seeing the same phenomenon on stock, but that setup didn't have blackbox - I guess I'll decomission my OBF4SD bird and run it stock everything to reproduce the problem.

Practically, I won't see an improvement for any of these unless I'm moving from the stock filter values, but IFF I can reproduce this issue on stock filters across stock and low PIDs, then that's going to have value for improving BF... for my part I'm still happy running 95% of my fleet on BF3.2

SimonToast, 2019-01-25:
I have same issue on Matek F405OSD but will post diff/logs etc when I'm. Shall I add here or new issue ticket?

etracer65, 2019-01-25:
@SimonToast No new issue is required. This is not a problem but by design. See here https://github.com/betaflight/betaflight/issues/6761

SimonToast, 2019-01-25:
Oh wow that sticks out like a sore thumb thanks @etracer65 I should have checked the thing before posting but was at work (build in the evenings, fly at lunch). A quick look at the heading window in the configurator shows the probable cause, it's clearly rotated 90 degrees. "set align_board_yaw = 270" worked for me as per the guy in that post.

Thank you also and all the devs for your hard work in Betaflight, you're really helping to keep this hobby exciting on the firmware front!

ijustwannafly, 2019-01-25:
@etracer65 

So matek f4 boards (I use the f405 ctr) with one gyro chip, when bf4.0 is released, will require board alignment after flashing?

Am I reading that correct?  I understand the need for a generic target.  Just taking notes on what to expect for the next major release.

etracer65, 2019-01-25:
@ijustwannafly No. It's just an interim issue as we prepare for new target structure.

TehllamaFPV, 2019-02-25:
Smoked an (extremely cheap) ESC testing - so I'll get another installed and go back to trying to test on vanilla 4.0.

Nytfury has been telling me I need to get all my rigs over, so I should figure out stupid issue I'm having then get that sorted.

closed 2019-04-03 by stale[bot]

TehllamaFPV, 2019-04-10:
Resolved - going back to stock Q and disabling Launch Mode was the ticket.

I'll try testing to verify that forcing launch mode on the same switch as ARM is a root cause, and try to reproduce that issue with my 5"... at which point I'm moving 21 more quads to 4.0.

Thanks as always Bruce!