betaflight/betaflight #7456: PIT mode through rx switch (in modes tab) does not work with AKK vtx's

aurbus, 2019-01-24:
I recently hooked up Smartaudio on my AKK FX2 ultimate Vtx. I went into the configurator and set up the pit mode on a switch function. It does not work, and after having done some research, I found that this problem is not unique to me. There are others who are having trouble with similar 

Steps to reproduce the behavior:

go into the configurator, and using an AKK Vtx, setup smart audio and pit mode on a switch. Now, attempt to put the Vtx into pit mode using the designated switch. 

when you set up smart audio, the actual smart audio functions as desired, but when you flip the switch that you set up to put the Vtx into pit mode, then it does not work (nothing happens). 

here is a diff all of my FC:

# diff all

# version
# Betaflight / BETAFLIGHTF4 (BFF4) 3.5.5 Jan 15 2019 / 09:36:35 (88921670b) MSP API: 1.40

board_name BETAFLIGHTF4
manufacturer_id 
mcu_id 001a00383037470935393636
signature 

# reset configuration to default settings
defaults nosave

# name
name Christian L's qu

# resources

# mixer

# servo

# servo mix

# feature
feature RSSI_ADC
feature AIRMODE

# beeper
beeper -GYRO_CALIBRATED
beeper -DISARMING
beeper -ARMING
beeper -ARMING_GPS_FIX
beeper -GPS_STATUS
beeper -ACC_CALIBRATION
beeper -ACC_CALIBRATION_FAIL
beeper -READY_BEEP
beeper -DISARM_REPEAT
beeper -ARMED
beeper -ON_USB
beeper -BLACKBOX_ERASE
beeper -CRASH FLIP
beeper -CAM_CONNECTION_OPEN
beeper -CAM_CONNECTION_CLOSED

# beacon
beacon RX_LOST
beacon RX_SET

# map
map TAER1234

# serial
serial 1 2048 115200 57600 0 115200
serial 2 32 115200 57600 0 115200

# led

# color

# mode_color

# aux
aux 0 0 0 1700 2100 0 0
aux 1 13 1 1700 2100 0 0
aux 2 35 3 1700 2100 0 0
aux 3 36 2 1425 1575 0 0
aux 4 39 4 1925 2100 0 0

# adjrange

# rxrange

# vtx

# rxfail
rxfail 4 h
rxfail 5 h

# master
set acc_hardware = NONE
set acc_calibration = -63,-62,15
set mag_hardware = NONE
set baro_hardware = NONE
set blackbox_p_ratio = 64
set min_throttle = 1070
set dshot_idle_value = 340
set use_unsynced_pwm = OFF
set motor_pwm_protocol = DSHOT1200
set motor_pwm_rate = 480
set failsafe_delay = 5
set bat_capacity = 1300
set vbat_max_cell_voltage = 42
set vbat_min_cell_voltage = 34
set current_meter = ADC
set battery_meter = ADC
set small_angle = 180
set pid_process_denom = 1
set osd_warn_batt_not_full = OFF
set osd_warn_visual_beeper = OFF
set osd_warn_core_temp = OFF
set osd_vbat_pos = 2403
set osd_rssi_pos = 374
set osd_crosshairs_pos = 2286
set osd_current_pos = 2453
set osd_mah_drawn_pos = 2440
set osd_warnings_pos = 2346
set osd_avg_cell_voltage_pos = 2435
set osd_disarmed_pos = 2219
set osd_esc_tmp_pos = 473
set osd_stat_tim_1 = ON
set osd_stat_tim_2 = OFF
set osd_stat_max_spd = OFF
set osd_stat_endbatt = ON
set osd_stat_battery = ON
set osd_stat_bbox = OFF
set osd_stat_bb_no = OFF
set rcdevice_init_dev_attempts = 4

# profile
profile 0

set dterm_lowpass_hz = 145
set dterm_notch_cutoff = 0
set vbat_pid_gain = ON
set feedforward_transition = 51
set smart_feedforward = ON
set iterm_relax = RPY
set throttle_boost = 9
set p_pitch = 61
set i_pitch = 60
set d_pitch = 24
set f_pitch = 68
set p_roll = 61
set i_roll = 58
set d_roll = 22
set f_roll = 68
set p_yaw = 76
set i_yaw = 54
set d_yaw = 2
set f_yaw = 69
set level_limit = 20
set abs_control_gain = 4

# profile
profile 1

# profile
profile 2

# restore original profile selection
profile 0

# rateprofile
rateprofile 0

set thr_mid = 40
set thr_expo = 30
set roll_rc_rate = 106
set pitch_rc_rate = 106
set yaw_rc_rate = 116
set yaw_expo = 20
set roll_srate = 81
set pitch_srate = 81
set yaw_srate = 82

# rateprofile
rateprofile 1

# rateprofile
rateprofile 2

# rateprofile
rateprofile 3

# rateprofile
rateprofile 4

# rateprofile
rateprofile 5

# restore original rateprofile selection
rateprofile 0

# save configuration
save

My Quad:

BFF4 FC
Frsky XSR running SBUS
DYS Aria 35a blheli_32 ESC's
AKK FX2 Ultimate vtx
EMAX RS2306 motors

Other stuff:
Taranis QX7 running opentx 2.2.2 
Fatshark Genesis goggle
Tattu R-Line 95c 1300mah

etracer65, 2019-01-24:
Have you tried enabling pit mode from the OSD menus? You may have to check with the manufacturer as some VTX's don't support entering pit mode via telemetry and require using the button (like holding it on power on for example).

As long as you can change channels/power using either the CLI parameters or through the OSD then that means that smartaudio is working. There's nothing extra that needs to be configured from the flight controller to enable pit mode (it's just a command in the protocol). This points towards lack of vtx support.

aurbus, 2019-01-24:
no, have not tried that, will try enabling pit mode through osd :)

ijustwannafly, 2019-01-25:
Which akk vtx are you using?

You may need to use the akk patched version of betaflight.

aurbus, 2019-01-26:
the vtx is almost new (less then 20 flights), and it is the FX2 ultimate

m013, 2019-02-23:
the same with matek 405aio and with this vtx. i  bought  it around 3 moths ago from akk.

i can change channels and power via osd and low power at disarm works, but if i enable pit mode (via switch or osd) - nothing happens until reboot - in that case vtx stops to work lol. i had to press and hold the butt for 10 sec to make it alive again.

no problems with my other tamp-vtx though (panda rc / aomway).

closed 2019-04-01 by stale[bot]

TwoToneEddy, 2019-05-09:
Was this ever solved? I have an atlatl V2 and i can trigger pitmode through the OSD but not with a rx channel.

pierreh, 2019-08-22:
I've got a AKK FX2-Dominator and I experience the same thing. I believe though it behaves the same as the TBS Unify. See https://www.youtube.com/watch?v=dCWngIk8MW4 and https://github.com/betaflight/betaflight/wiki/Unify-Smartaudio#switching-between-freestyle-and-race

suseek, 2020-01-29:
Same here with Atlatl V2 - triggering through RX channel doesn't work. It does work when I first turn it on with a button and then I can turn it off with rx channel, but that's not really efficient.

mmazzonc, 2020-04-13:
you need to reflash your FC with BUTTERFLIGHT configurator  and the activate the LEGACY_SA_SUPPORT in the configutìration panel. This will solve the problem