betaflight/betaflight #7459: FF_Racepit Gyro/Accelerometer Orientation on 4.0.0 Build #1382

brettcollis2212, 2019-01-24:
**Describe the bug**
Flashing the FF_Racepit Target with the 3.5.5 hex shows expected gyro and accelerometer behavior in the setup tab of the GUI,

However flashing 4.0.0 Build #1382 STM32F405 target and then using the config file :- https://github.com/betaflight/betaflight/blob/8f0157da3df36a29efb3cc79772a50b9d0a87875/src/main/target/STM32F405/config/FF_RACEPIT.config results in behaviour not expected in the setup tab. The Quad seems to be 90 degrees rotated in the yaw axis compared to what it should be.

**To Reproduce**
Flash 3.5.5 and it will behave as expected

However flashing 4.0.0 Build #1382 STM32F405 target and then using the config file :- https://github.com/betaflight/betaflight/blob/8f0157da3df36a29efb3cc79772a50b9d0a87875/src/main/target/STM32F405/config/FF_RACEPIT.config results in behaviour not expected in the setup tab. The Quad seems to be 90 degrees rotated in the yaw axis compared to what it should be.

Changing the Gyro Config to any of the values below does not make it behave correctly:

 # set gyro_1_sensor_align = 
###INVALID VALUE###
Allowed values: DEFAULT, CW0, CW90, CW180, CW270, CW0FLIP, CW90FLIP, CW180FLIP, CW270FLIP

However using the below values:-

# master
set align_gyro = CW90FLIP
set align_acc = CW90FLIP

Results in expected behaviour

**Expected behavior**
I expected the flight controller to behave correctly in the GUI in 4.0 as it does in 3.5.5
[image]
[image]

**Flight controller configuration**

4.0.0 Diff for non working

`# version
# Betaflight / STM32F405 (S405) 4.0.0 Jan 23 2019 / 13:47:52 (a4ce8b560) MSP API: 1.41

board_name 
manufacturer_id 

# name

# resources
resource BEEPER 1 B03
resource MOTOR 1 B00
resource MOTOR 2 B01
resource MOTOR 3 B11
resource MOTOR 4 B10
resource LED_STRIP 1 B06
resource SERIAL_TX 1 A09
resource SERIAL_TX 2 A02
resource SERIAL_TX 3 C10
resource SERIAL_TX 4 A00
resource SERIAL_TX 5 C12
resource SERIAL_TX 6 C06
resource SERIAL_RX 1 B07
resource SERIAL_RX 2 A03
resource SERIAL_RX 3 C11
resource SERIAL_RX 4 A01
resource SERIAL_RX 5 D02
resource SERIAL_RX 6 C07
resource INVERTER 3 C15
resource I2C_SCL 3 A08
resource I2C_SDA 3 C09
resource LED 1 B05
resource LED 2 B04
resource SPI_SCK 1 A05
resource SPI_SCK 2 B13
resource SPI_MISO 1 A06
resource SPI_MISO 2 B14
resource SPI_MOSI 1 A07
resource SPI_MOSI 2 B15
resource ESCSERIAL 1 B00
resource CAMERA_CONTROL 1 A10
resource ADC_BATT 1 C02
resource ADC_CURR 1 C01
resource OSD_CS 1 B12
resource GYRO_EXTI 1 C04
resource GYRO_CS 1 A04

# dmaopt

# mixer

# servo

# servo mix

# feature
feature -RX_PARALLEL_PWM
feature RX_SERIAL

# beeper

# beacon

# map

# serial
serial 2 64 115200 57600 0 115200

# led

# color

# mode_color

# aux

# adjrange

# rxrange

# vtx

# rxfail

# master
set dshot_burst = ON
set current_meter = ADC
set battery_meter = ADC
set beeper_inversion = ON
set beeper_od = OFF
set tlm_halfduplex = OFF
set max7456_spi_bus = 2
set pinio_box = 40,255,255,255
set gyro_1_bustype = SPI
set gyro_1_spibus = 1
set gyro_1_sensor_align = CW90FLIP
set i2c3_pullup = ON

# profile
profile 0

# rateprofile
rateprofile 0`

**Setup / Versions**
 - Flight controller [what type is it, where was it bought from];
 - Other components [RX, VTX, brand / model for all of them, firmware version where applicable];
 - how are the different components wired up.

**Additional context**
Add any other context about the problem here.

jflyper, 2019-01-25:
Thanks for filing an issue report.

Okay, this one has revealed another todo item towards generic target goal.

What would be the best way to do this?

- Create a new PG PG_ACC_DEVICE_CONFIG with `accDeviceConfig_s` which only has `align` member, instantiate as single or an array of two as necessary, and use it to set `acc.dev.align` during acc init.

- Duplicate corresponding gyro device alignment; works only if acc and gyro are in the same chip. Do we have a board implementation with separate chips? Probably the resurrected DISCO?

- Others

Thoughts?

mikeller, 2019-01-26:
@brettcollis2212: Thanks for reporting this.

@jflyper:

> Duplicate corresponding gyro device alignment; works only if acc and gyro are in the same chip. Do we have a board implementation with separate chips? Probably the resurrected DISCO?

Yes, STM32F3DISCOVERY and STM32F411DISCOVERY use separate devices for gyro and acc. But alignment isn't the main problem there, as the gyro is SPI and acc is I2C. They currently work by having acc hard coded at compile time, and I do not think it is worthwhile adding the extra complexity to make mixed SPI / I2C gyro / acc combinations work on unified targets just to support these 'exotic' boards. We'll just have to release a unified target configuration for them with gyro support only (should be good enough for most experimenters), and have dedicated targets with gyro and acc.

> Create a new PG PG_ACC_DEVICE_CONFIG with accDeviceConfig_s which only has align member, instantiate as single or an array of two as necessary, and use it to set acc.dev.align during acc init.

Not sure this is worth it - there are other factors requiring gyro and acc to be one shared device (e.g. bus configuration), so separating the alignment will just add a redundant configuration option, increasing the potential for misconfiguration.

Another thing that should be looked at is the relation between `gyro_X_sensor_align` and `align_gyro` / `align_acc`. I suspect the latter is redundant with 4.0, and should probably be removed?

PR #7471 (2019-01-29): Set acc alignment to that of active gyro
Fixes #7459 

Note that an additional PR is required to handle `MSP_SENSOR_ALIGNMENT`.
closed 2019-01-29 by mikeller