betaflight/betaflight #7464: Implement support for FFPV 2.4GHz VTX protocol

harrisongeron, 2019-01-25:
Looking to use radio control to change my FuriousFPV 2.4ghz. 

**Describe the solution you'd like**
Looking to use radio control to change my FuriousFPV 2.4ghz. 

**Describe alternatives you've considered**
iNav recently implemented, but want to use betaflight

DieHertz, 2019-01-26:
It can be done as a custom branch, but given the inherent differences between FuriousFPV protocol and Tramp, it will be a waste of flash space and an invitation for other manufacturers to come up with their own protocols.

mikeller, 2019-01-26:
@DieHertz: Is this the incorrect clone of the tramp protocol? If so, this should be closed, and FuriousFPV should release new firmware that fixes their broken protocol.

DieHertz, 2019-01-26:
Last time I saw it, ffpv 2g4 was heavily inspired by tramp, but with 2g4 bands instead of 5g8

mikeller, 2019-01-26:
@DieHertz: Might be squeezed into the Tramp driver then, if they fix their implementation to make it compatible (I remember talking to a manufacturer about this, I just didn't remember it was FuriousFPV).

@harrisongeron: Do you have a specification of the protocol that FuriousFPV is using, or can you contact them to get one?

jflyper, 2019-01-27:
If the protocol is actually compatible with the original tramp, then we might be able to handle it with user defined vtx table facility.

harrisongeron, 2019-01-29:
I'm not coding savvy, but I can try and act as an intermediary with FuriousFPV if this is something that could be done. Would just need to know what information is needed. I know the protocol was just added to Inav within the past few weeks (and I have heard it's based on the tramp protocol too)

There's a bunch of us on 2.4ghz and would love to use to variable outputs of the 2.4ghz Furious vtx, and don't want to use inav because betaflight is much better....

mikeller, 2019-01-29:
@harrisongeron: Essentially we'd need a specification document for FuriousFPV's protocol for evaluation. And if we decided it was feasible to add it, and we'd do it, then FuriousFPV supplying an engineering sample of their VTX to the developer doing the implementation will help speed up and simplify testing / bugfixing immensely.

DieHertz, 2019-01-29:
The spec is available at iNav GitHub, and I also have it somewhere on my PC 
I have a sample of 2g4 VTX and will look into it.

jflyper, 2019-01-30:
@DieHertz Tramp driver and CMS can easily be converted to use VTX table. I've done it once, but can't find the branch. You might want to check if the protocol is compatible first. If it is, then the VTX table facility will save the support effort.

mikeller, 2019-02-04:
@sniperxnl: Yes please, you can post it here.

sniperxnl, 2019-02-04:
[Furious-24G-REV03.pdf](https://github.com/betaflight/betaflight/files/2827564/Furious-24G-REV03.pdf)

jflyper, 2019-02-04:
It basically look like a tramp protocol... except for the LED control garbage. I think we'll just ignore those.

sniperxnl, 2019-02-04:
I think we are happy if we can control by osd, band, channel and mw

mikeller, 2019-02-04:
@jflyper: Custom VTX tables for the win!

jflyper, 2019-02-05:
@mikeller I wonder how the 2.4G part responds to controls with current table swapped with the 2.4G table 🤔

mikeller, 2019-02-05:
@jflyper: We should try it - if we only had the hardware.

jflyper, 2019-02-06:
#7540 provides a test platform.

The manual states not to use the pit mode, but i didn't see any explicit pit mode control in the manual, so the note was ignored. All LED control garbage are also fully ignored.

harrisongeron, 2019-02-22:
Hey y'all. 

Thanks for the ahrd work on this. We all truly appreciate it. 

Unfortunately, the two of us here cannot seem to get it working. 

Using 4.0.0 1454 and the tables in 7540- the vtx won't change bands/ channels. Oddly, it will reliabily change the power levels correctly on a save-reboot. Am I missing anything other than making sure I'm using the newest 4.0 build, copying and saving the table below into CLI, and then selecting Tramp on the correct Uart? 

If there's anyone in the US I wouldn't mind shipping  a unit for troubleshooting. This feature would be very helpful to us. 

Thanks

harrisongeron, 2019-03-01:
It looks like the Inav people are having the same issue... 

https://github.com/iNavFlight/inav/issues/3285

I would be more than happy to send my vtx to one of you that has the skills/ tools to make this working!

mikeller, 2019-03-01:
@harrisongeron: This is not something that will be happening in 4.0, as the VTX tables feature isn't ready for use for all VTX types yet.
Let's try to make this happen for 4.1. Normally we ask manufacturers to provide us with hardware to implement / test the work we are doing for adding support (without charging for it), I think maybe @MJ666 	can help with this?

MJ666, 2019-03-02:
Mhhh, don't have this FFPV 2.4GHz VTX for testing.

jflyper, 2019-03-02:
Quick look at iNavFlight/inav#3285 suggest that the unit is going unresponsive upon some unknown event. There's not much we can do at user's side. Fortunately(?), the vtxtable facility which was to provide generic driver for tramp protocol has been postponed until 4.1, so FFPV has some time to find out what is going on within the unit's code.

@harrisongeron @MJ666 It is important that FFPV work closely with us to make this work.

MJ666, 2019-03-02:
@jflyper already pinged FFPV to make them aware of this issue.

harrisongeron, 2019-03-02:
Sounds good, I'll wait patiently:) let me know if there's anything I can do

expipiplus1, 2019-07-08:
I have patched the firmware to stop it becoming unresponsive on power up. My patched firmware can be found here: https://github.com/expipiplus1/ffpv-fix

sblaus, 2020-03-31:
Hi, Furius protocol  is not ready on Betaflight setting......have you some news/solution?
Regards

mchaos, 2020-04-27:
Been a while. Where are we with this? I see someone posted firmware for the vtx.. does this make it work? Don't want to flash bad firmware.  Would love to be able to do this from osd...  thanks

expipiplus1, 2020-04-27:
@mchaos I tried quite hard over email to get ffpv to fix the firmware but to no avail. 

FWIW my firmware has been working well for me. Additionally during development of it I didn't have any problems reverting to the official firmware. If you don't want to trust the patched firmware I put on the releases page: I tried to make the instructions clear enough so that it can be easily reproduced. 

If you don't want to flash my patched firmware you could make sure the vtx powers up after the FC some other way, a vtx controlled switch for example.

mchaos, 2020-04-27:
I just want confirmation that your patched fw is good. I'll try it maybe tonight and report back. Right now I have it power up and I added the tables but I get a solid led on the vtx now. Haven't tried seeing if it's transmitting. Was late last night

enlightened26, 2020-04-28:
Bit of a novice here is this as easy as flashing the firmware to the Vtx and setting it up for tramp audio?
Ive recently purchased the vtx and a little annoyed that theyre still selling it as "smartaudio on tue next firmware" almost 2 years later.

expipiplus1, 2020-06-17:
> Bit of a novice here is this as easy as flashing the firmware to the Vtx and setting it up for tramp audio?

IIRC the protocol is almost identical to tramp, I think that the only difference is that obviously the channels have different frequencies. [Flashing the firmware](https://github.com/expipiplus1/ffpv-fix#flashing-the-new-firmware-to-the-board) does involve soldering the ST-Link to the VTX but there are dedicated pads so it isn't too troubling.

> Ive recently purchased the vtx and a little annoyed that theyre still selling it as "smartaudio on tue next firmware" almost 2 years later.

It's very frustrating. I even sent them documentation on my patched firmware to no avail.

closed 2026-09-26 by blckmn