betaflight/betaflight #7474: SPI BMP280 missing after build 1352

fgiudice98, 2019-01-26:
After the build 1352 (included) Betaflight doesn't see the BMP280 (connected by SPI bus)
I noticed that in build 1351 the resource list give BARO_CS A09, after 1352 it gives PREINIT 2 A09
The PREINIT cannot be changed nor freed, but BARO_CS is at A09 (and changing doesn't change anything)
I tried manual setting the baro hardware (SPI, CD pin as above, hardware and so on) but still nothing changes

etracer65, 2019-01-26:
I can replicate this problem on an AIRBOTF4SD target with a SPI-connected BMP280.

```
# status
MCU F40X Clock=168MHz (PLLP-HSE), Vref=3.29V, Core temp=43degC
Stack size: 2048, Stack address: 0x1000fff0
Config size: 2907, Max available config: 16384
GYRO=MPU6000, ACC=MPU6000
System Uptime: 7 seconds
CPU:6%, cycle time: 125, GYRO rate: 8000, RX rate: 33, System rate: 9
Voltage: 0 * 0.01V (0S battery - NOT PRESENT)
I2C Errors: 0
SD card: None inserted
Arming disable flags: ANGLE CLI MSP

# resource
resource BEEPER 1 B04
resource MOTOR 1 B00
resource MOTOR 2 B01
resource MOTOR 3 A03
resource MOTOR 4 A02
resource MOTOR 5 A01
resource MOTOR 6 A08
resource PPM 1 B14
resource PWM 1 B14
resource PWM 2 B15
resource PWM 3 C06
resource PWM 4 C07
resource PWM 5 C08
resource PWM 6 C09
resource LED_STRIP 1 B06
resource SERIAL_TX 1 A09
resource SERIAL_TX 3 B10
resource SERIAL_TX 4 A00
resource SERIAL_TX 6 C06
resource SERIAL_RX 1 A10
resource SERIAL_RX 3 B11
resource SERIAL_RX 4 A01
resource SERIAL_RX 6 C07
resource INVERTER 6 D02
resource LED 1 B05
resource SPI_SCK 1 A05
resource SPI_SCK 3 C10
resource SPI_MISO 1 A06
resource SPI_MISO 3 C11
resource SPI_MOSI 1 A07
resource SPI_MOSI 3 C12
resource ESCSERIAL 1 B14
resource ADC_BATT 1 C02
resource ADC_RSSI 1 A00
resource ADC_CURR 1 C01
resource BARO_CS 1 C13
resource SDCARD_CS 1 B03
resource SDCARD_DETECT 1 C00
resource PINIO 1 C08
resource GYRO_EXTI 1 C04
resource GYRO_CS 1 B13
resource GYRO_CS 2 A04
resource USB_DETECT 1 C05

# resource list
Currently active IO resource assignments:
(reboot to update)
--------------------
A00: FREE
A01: FREE
A02: MOTOR 4
A03: MOTOR 3
A04: PREINIT 2
A05: SPI_SCK 1
A06: SPI_MISO 1
A07: SPI_MOSI 1
A08: FREE
A09: FREE
A10: FREE
A11: USB
A12: USB
A13: FREE
A14: FREE
A15: FREE
B00: MOTOR 1
B01: MOTOR 2
B02: FREE
B03: SDCARD_CS
B04: BEEPER
B05: LED 1
B06: FREE
B07: FREE
B08: FREE
B09: FREE
B10: FREE
B11: FREE
B12: FREE
B13: GYRO_CS 1
B14: FREE
B15: FREE
C00: FREE
C01: FREE
C02: FREE
C03: FREE
C04: GYRO_EXTI
C05: USB_DETECT
C06: FREE
C07: FREE
C08: PINIO 1
C09: FREE
C10: SPI_SCK 3
C11: SPI_MISO 3
C12: SPI_MOSI 3
C13: PREINIT 3
C14: FREE
C15: FREE
D02: INVERTER 6
```

jflyper, 2019-01-27:
Haven't looked deep yet, but by looking at target defs, NOX, AIRBOTF4 and AIRBOTF4SD shares an SPI bus between gyro and BMP280. This might be causing the issue as gyro driver haven't been converted to be shared bus aware yet.

MJ666, 2019-01-27:
For the AFNGF7 taget the BMP280 at the SPI bus is working. Baro, Mag and OSD are sharing the same SPI bus here.

AvaraKaisa, 2019-01-30:
Also Omnibus F4 Pro V2 (OBSD) missing baro after build 1352.

joelucid, 2019-02-09:
Similar problem with rx spi on the noxe board from BG. Had to uncomment spi transactions for it to start working again.

atomiclama, 2019-03-07:
Some more information for this issue: -

Omnibus F4 SD has BMP280 and OSD on a shared SPI. 

On a first power or at least 5 min between power cycles the BMP280 is detected but gives unreliable results, values bounce all over the place, and the OSD flickers. 
With a reset or short time between power cycles the BMP280 is not detected and the OSD works fine.
Disable baro and OSD is fine.

Baro and OSD are not playing nicely with each other, naughty naughty.

mikeller, 2019-03-07:
Tested here with latest `master` with an OMNIBUSF4V6 (has got BMP280 and MAX7456 on SPI3). Baro works, with the usual fluctuation in the 0.1 m range and drift. Not sure if there is anything different with the hardware configuration on the OMNIBUSF4SD, I don't have one to test.

atomiclama, 2019-03-07:
Going to have a bit more of a detailed look later, will first try a master build. As I'm currently a week behind.

atomiclama, 2019-03-07:
Ok just tested build #1492 on my OMNIBUSF4SD and so far so good, the baro is now detected and stable. OSD is stable. Will do some more long duration tests but looks good.

mikeller, 2019-03-07:
> Now what am I going to do in my lunch time ;-(

Eat lunch? :stuck_out_tongue:

mikeller, 2019-03-07:
@fgiudice98: This seems to be fixed. Can you confirm & close please?

fgiudice98, 2019-03-07:
@mikeller I only tested a few moments ago with a board not connected and baro seems working good. I will test more in the next days.
Thanks everyone

closed 2019-03-07 by fgiudice98

atomiclama, 2019-03-08:
> Eat lunch?

Nah lunch time is for interesting stuff ;-)

Anyway lot's of power cycles and a few flights all it well.