betaflight/betaflight #7494: GPS not working in 3.5.5

nerilorenzo, 2019-01-28:
**Describe the bug**
GPS is not working in version 3.5.5

**To Reproduce**
With the same hardware and the last 4.0.0 version the GPS work perfectly.
With versione 3.5.5
Betaflight / OMNIBUSF7V2 (OB72) 3.5.5 Jan 15 2019 / 13:32:46 (8a09989f7) MSP API: 1.40
in the configurator I can see that the gps is reading sats but the position is not acquired even if 6 stas are acquired. in the OSD I don't see the sats=0 info

**Expected behavior**
Sats number in OSD and position acquired

**Flight controller configuration**
```
Entering CLI Mode, type 'exit' to return, or 'help'

# diff

# version
# Betaflight / OMNIBUSF7V2 (OB72) 3.5.5 Jan 15 2019 / 13:32:46 (8a09989f7) MSP API: 1.40

board_name OMNIBUSF7V2
manufacturer_id 

# name
name Blacks

# resources
resource MOTOR 1 E11
resource MOTOR 4 B00

# mixer

# servo

# servo mix

# feature
feature GPS
feature TELEMETRY
feature AIRMODE
feature ESC_SENSOR

# beeper
beeper -GYRO_CALIBRATED
beeper -DISARMING
beeper -GPS_STATUS
beeper -ACC_CALIBRATION
beeper -ACC_CALIBRATION_FAIL
beeper -DISARM_REPEAT
beeper -ARMED
beeper -SYSTEM_INIT
beeper -ON_USB
beeper -BLACKBOX_ERASE
beeper -CAM_CONNECTION_OPEN
beeper -CAM_CONNECTION_CLOSED

# beacon

# map
map TAER1234

# serial
serial 0 32 115200 57600 0 115200
serial 2 2 115200 115200 0 115200
serial 5 2048 115200 57600 0 115200

# led

# color

# mode_color

# aux
aux 0 0 0 900 1200 0 0
aux 1 1 1 1150 1250 0 0
aux 2 1 1 1500 1600 0 0
aux 3 1 1 1850 1950 0 0
aux 4 2 1 1050 1150 0 0
aux 5 2 1 1400 1500 0 0
aux 6 2 1 1750 1850 0 0
aux 7 13 1 1600 2000 0 0
aux 8 26 0 900 1200 0 0
aux 9 35 2 1800 2000 0 0
aux 10 35 2 1525 1625 0 0
aux 11 35 2 1150 1250 0 0

# adjrange

# rxrange

# vtx

# rxfail
rxfail 3 h

# master
set acc_calibration = -141,-126,7
set mag_hardware = NONE
set min_check = 1010
set max_check = 1990
set rssi_channel = 8
set rc_smoothing_type = FILTER
set fpv_mix_degrees = 25
set blackbox_p_ratio = 64
set motor_pwm_protocol = DSHOT1200
set align_board_pitch = 180
set align_board_yaw = 90
set bat_capacity = 1600
set current_meter = ESC
set gps_auto_baud = ON
set osd_warn_visual_beeper = OFF
set osd_cap_alarm = 1700
set osd_vbat_pos = 2081
set osd_rssi_pos = 2145
set osd_anti_gravity_pos = 2547
set osd_vtx_channel_pos = 2529
set osd_crosshairs_pos = 2253
set osd_ah_sbar_pos = 2254
set osd_ah_pos = 2126
set osd_mah_drawn_pos = 2113
set osd_craft_name_pos = 2507
set osd_gps_speed_pos = 2473
set osd_gps_lon_pos = 2409
set osd_gps_lat_pos = 2441
set osd_gps_sats_pos = 2133
set osd_home_dir_pos = 2123
set osd_home_dist_pos = 2125
set osd_altitude_pos = 2537
set osd_esc_tmp_pos = 2521
set osd_core_temp_pos = 2553
set osd_stat_max_alt = ON
set debug_mode = GYRO_SCALED
set vtx_band = 2
set vtx_channel = 2
set vtx_freq = 5752
set vcd_video_system = PAL

# profile
profile 0

set dterm_notch_cutoff = 0
set iterm_relax = RP
set i_pitch = 70
set i_roll = 65
set i_yaw = 70

# rateprofile
rateprofile 0

set roll_expo = 15
set pitch_expo = 15
set yaw_expo = 15
set roll_srate = 80
set pitch_srate = 80
set yaw_srate = 71
set tpa_rate = 0
set tpa_breakpoint = 2000
```

**Setup / Versions**
 - omnibus F7 v2
 - GPS Dual BN-220 GPS GLONASS Antenna Module TTL Level RC Drone Airplane

**Additional context**
hardware and wiring was tested by using firmware 4.0.0 and with this version the GPS behaviour was OK in the configurator and in the OSD output

etracer65, 2019-01-29:
We have no other problem reports of GPS not working on 3.5.5. I just tested using a BN-220 GPS and it worked as expected. It's likely you just have a settings difference. Most BN220 are set up only for UBLOX messaging and I notice in your config above you still have the default of NMEA.

[image]

nerilorenzo, 2019-01-30:
Hi,
with UBLOX it works perfectly.
I opened the ticket because with NMEA and 4.0.0 works, but now I discovered that was not fully working with NMEA and 4.0.0, with UBLOX works faster and better.

Thanks for the support.

etracer65, 2019-01-30:
The problem is not that NMEA doesn't work in Betaflight. It's that your GPS unit is configured to use UBLOX protocol and not NMEA. This is very common for the BN220 units. Basically you were just using the wrong settings for your GPS.

closed 2019-01-30 by etracer65