betaflight/betaflight #7498: Buzzer not working in Betaflight 4.0.0 Jan 29 2019

remizik, 2019-01-29:
**Describe the bug**
Hello,
I can't make the buzzer work with Betaflight 4 :(
Beacon works fine, I even see the LED blinking but the buzzer doesn't make any sound.
I also checked in the MODES and the switch is working fine.
I tried a clean install (erase the chip)

No issue at all with version 3.5 ! works perfectly fine.

**Expected behavior**
I'd like the buzzer to make some noise :)

**Flight controller configuration**
```
# diff

# version
# Betaflight / CLRACINGF4 (CLR4) 4.0.0 Jan 29 2019 / 14:04:03 (436892424) MSP API: 1.41

board_name CLRACINGF4
manufacturer_id 

# name

# resources

# dmaopt

# mixer

# servo

# servo mix

# feature
feature TELEMETRY
feature AIRMODE

# beeper
beeper -GYRO_CALIBRATED
beeper -RX_LOST
beeper -RX_LOST_LANDING
beeper -DISARMING
beeper -ARMING
beeper -ARMING_GPS_FIX
beeper -BAT_LOW
beeper -GPS_STATUS
beeper -ACC_CALIBRATION
beeper -ACC_CALIBRATION_FAIL
beeper -READY_BEEP
beeper -DISARM_REPEAT
beeper -ARMED
beeper -SYSTEM_INIT
beeper -ON_USB
beeper -BLACKBOX_ERASE
beeper -CRASH_FLIP
beeper -CAM_CONNECTION_OPEN
beeper -CAM_CONNECTION_CLOSE

# beacon

# map
map TAER1234

# serial
serial 3 8192 115200 57600 0 115200
serial 5 64 115200 57600 0 115200

# led

# color

# mode_color

# aux
aux 0 0 0 1300 1700 0 0
aux 1 1 4 1375 1625 0 0
aux 2 2 4 1850 2100 0 0
aux 3 13 1 1800 2100 0 0
aux 4 35 2 1300 1700 0 0

# adjrange

# rxrange

# vtx

# rxfail

# master
set min_check = 1005
set max_check = 1995
set rssi_channel = 12
set rc_smoothing_type = INTERPOLATION
set serialrx_provider = CRSF
set dshot_idle_value = 400
set motor_pwm_protocol = DSHOT1200
set failsafe_throttle_low_delay = 50
set bat_capacity = 1500
set vbat_min_cell_voltage = 350
set vbat_warning_cell_voltage = 360
set ibata_scale = 159
set yaw_motors_reversed = ON
set deadband = 3
set yaw_deadband = 3
set pid_process_denom = 2
set osd_warn_batt_not_full = OFF
set osd_warn_core_temp = OFF
set osd_warn_rc_smoothing = OFF
set osd_warn_fail_safe = OFF
set osd_warn_launch_control = OFF
set osd_warn_no_gps_rescue = OFF
set osd_warn_gps_rescue_disabled = OFF
set osd_rssi_pos = 2490
set osd_tim_2_pos = 2508
set osd_current_pos = 416
set osd_mah_drawn_pos = 2520
set osd_warnings_pos = 2473
set osd_avg_cell_voltage_pos = 2497
set vtx_band = 5
set vtx_channel = 7
set vtx_freq = 5880

# profile
profile 0

set p_pitch = 60
set i_pitch = 30
set d_pitch = 70
set p_roll = 50
set i_roll = 35
set d_roll = 55
set i_yaw = 35
set f_yaw = 60

# rateprofile
rateprofile 0

set roll_rc_rate = 140
set pitch_rc_rate = 140
set yaw_rc_rate = 140
set roll_expo = 30
set pitch_expo = 30
set yaw_expo = 30
set roll_srate = 72
set pitch_srate = 72
set yaw_srate = 72
```
**Setup / Versions**
 - Flight controller : Bardwell F4 AiO v2

remizik, 2019-01-29:
hi :)

I'm using that kind of buzzer : 
[image]

Can't find what you ask for in my 3.5 DIFF, so here it is !

````
# diff

# version
# Betaflight / CLRACINGF4 (CLR4) 3.5.3 Nov 14 2018 / 23:17:07 (61d5abf00) MSP API: 1.40

board_name CLRACINGF4
manufacturer_id 

# feature
feature TELEMETRY
feature AIRMODE

# beeper
beeper -GYRO_CALIBRATED
beeper -RX_LOST
beeper -RX_LOST_LANDING
beeper -DISARMING
beeper -ARMING
beeper -ARMING_GPS_FIX
beeper -BAT_LOW
beeper -GPS_STATUS
beeper -ACC_CALIBRATION
beeper -ACC_CALIBRATION_FAIL
beeper -READY_BEEP
beeper -DISARM_REPEAT
beeper -ARMED
beeper -SYSTEM_INIT
beeper -ON_USB
beeper -BLACKBOX_ERASE
beeper -CRASH FLIP
beeper -CAM_CONNECTION_OPEN
beeper -CAM_CONNECTION_CLOSED

# beacon

# map
map TAER1234

# serial
serial 3 8192 115200 57600 0 115200
serial 5 64 115200 57600 0 115200

# led

# color

# mode_color

# aux
aux 0 0 0 1300 1700 0 0
aux 1 1 4 1375 1625 0 0
aux 2 2 4 1850 2100 0 0
aux 3 13 1 1800 2100 0 0
aux 4 35 2 1300 1700 0 0
aux 5 39 3 1300 2100 0 0

# master
set min_check = 1005
set max_check = 1995
set rssi_channel = 12
set serialrx_provider = CRSF
set dshot_idle_value = 400
set motor_pwm_protocol = DSHOT1200
set bat_capacity = 1300
set vbat_min_cell_voltage = 35
set vbat_warning_cell_voltage = 36
set ibata_scale = 159
set yaw_motors_reversed = ON
set deadband = 3
set yaw_deadband = 3
set pid_process_denom = 2
set osd_warn_batt_not_full = OFF
set osd_warn_core_temp = OFF
set osd_vbat_pos = 417
set osd_rssi_pos = 2490
set osd_tim_1_pos = 33
set osd_tim_2_pos = 2508
set osd_crosshairs_pos = 269
set osd_current_pos = 44
set osd_mah_drawn_pos = 2520
set osd_craft_name_pos = 460
set osd_warnings_pos = 2473
set osd_avg_cell_voltage_pos = 2497
set osd_stat_tim_1 = ON
set osd_stat_battery = ON
set osd_stat_min_rssi = OFF
set osd_stat_bbox = OFF
set osd_stat_bb_no = OFF
set vtx_band = 5
set vtx_channel = 7
set vtx_power = 5
set vtx_freq = 5880

# profile
profile 1

set dterm_notch_cutoff = 0
set p_pitch = 70
set d_pitch = 45
set p_roll = 60
set i_roll = 50
set d_roll = 45

# rateprofile
rateprofile 1

set roll_rc_rate = 125
set pitch_rc_rate = 125
set yaw_rc_rate = 120
set roll_expo = 30
set pitch_expo = 30
set yaw_expo = 30
set roll_srate = 72
set pitch_srate = 72
set yaw_srate = 72
````

etracer65, 2019-01-29:
A picture of the beeper doesn't tell us anything. You need to determine if its an active beeper (just needs 5V power) or a passive beeper (needs a frequency sent to it to make the sound - like a speaker).

If it's an active beeper then:
```
set beeper_frequency = 0
```
If it's a passive beeper than you need to replace the 0 with the appropriate frequency for the beeper (provided by the manufacturer).

etracer65, 2019-01-30:
Try `set beeper_frequency = 0` if that doesn't work try `set beeper_frequency = 2000`. This just a settings issue for you. Beeper works fine on BF4.0.

remizik, 2019-01-30:
"set beeper_frequency = 0" fixed the issue, thanks a lot :)

closed 2019-01-30 by remizik