betaflight/betaflight #7505: Build 1352 and later - Dashes in display with DSHOT selected

aowi7280, 2019-02-01:
**Describe the bug**
Build 1352 and later - Dashes in display with any DSHOT protocol selected. I don't need to run motors to see the problem. Brushed, Oneshot, multishot, Proshot all work fine. Change to any DSHOT setting and the problem shows in the display. Definitely a digital issue with dashes across the screen. Not analog noise. Sounds similar to bug 4144, but I am not overclocking.

**To Reproduce**
Flash in build 1352 or later (have tried several up to 1398). No problem. Keep all defaults except change the ESC/Motor protocol to any DSHOT setting. Reboot. Get the noise.

**Expected behavior**
No dashes in OSD/FPV video.

**Flight controller configuration**
 ``` # version
# Betaflight / OMNIBUSF4SD (OBSD) 4.0.0 Jan 13 2019 / 21:08:57 (492d1f323) MSP API: 1.41

board_name OMNIBUSF4SD
manufacturer_id 
mcu_id 002500333136510234373332
signature 

# reset configuration to default settings
defaults nosave

# name

# resources

# dmaopt

# mixer

# servo

# servo mix

# feature

# beeper

# beacon

# map

# serial

# led

# color

# mode_color

# aux

# adjrange

# rxrange

# vtx

# rxfail

# master
set motor_pwm_protocol = DSHOT1200

# profile
profile 0

# profile
profile 1

# profile
profile 2

# restore original profile selection
profile 0

# rateprofile
rateprofile 0

# rateprofile
rateprofile 1

# rateprofile
rateprofile 2

# rateprofile
rateprofile 3

# rateprofile
rateprofile 4

# rateprofile
rateprofile 5

# restore original rateprofile selection
rateprofile 0 ``` 

**Setup / Versions**
 - Flight controller [what type is it, where was it bought from]; RTFQ OmnibusF4SD. No overclocking.
 - Other components [RX, VTX, brand / model for all of them, firmware version where applicable];  ImmersionRC 600mW Vtx. frsky x4r-sb receiver. Holybro micro M8N GPS.
 - how are the different components wired up. 
Batt voltage to external V regulator, 12V to Vtx. Vtx 5V output to camera. Camera to FC. Camera and Vtx have common ground on FC. ESCs each have a cap on the power.

**Additional context**
Add any other context about the problem here.
No problems prior to build 1352. I can make problem go away by either selecting any protocol other than DSHOT or flashing in Build 1351 or earlier. Disassembled everything and checked connections. Cleaned boards. Reassembled and problem still exists in 1352. Problem goes away in 1351. Checked twice. I originally had BLHeli32.3 and been running fine with DSHOT1200. I thought that might be the problem so I flashed in 32.6 but that did not help.

etracer65, 2019-02-01:
"Dashes across the screen" isn't completely helpful. Can you post a DVR video of the problem so we can see what you're actually seeing?

aowi7280, 2019-02-01:
Video is here. You can see them best in the lighter areas of the video. https://youtu.be/4SrrmY5Mg-U

etracer65, 2019-02-02:
@jflyper Might this be related to https://github.com/betaflight/betaflight/pull/6809 ? Seems to only be a problem when DSHOT is enabled though. Unless that flight controller is using DMA for the MAX7456?

mikeller, 2019-02-02:
This looks like the MAX7456 is actually malfunctioning:

[image]

Not sure if this can be induced by activity on the SPI bus, or if it's indicative of an electrical problem.

jflyper, 2019-02-02:
@aowi7280 

What is the exact nature (origin) of your board? Product URL?

aowi7280, 2019-02-02:
It is a "FLIP 32 - F4 - OMNIBUS V2 PRO" purchased 2 years ago from RTFQ. I don't see the exact V2 version any more. But there is a V3 PRO listed https://www.readytoflyquads.com/flip-32-f4-omnibus-v2-pro

hydra, 2019-02-02:
I have seen that one some prototype boards and where customers have shorted out, albeit breifly, the MAX7456 chip, replacing the MAX7456 always fixed it.  On shorted-out boards that I've repaired for customers generally the 3v3 reg needed replacing, the cpu needed replacing and the MAX7456 would then work, but appear similar to your photo above.  I did one about 2-3 weeks ago that looked exactly like it and replacing the MAX7456 fixed it.

mikeller, 2019-02-02:
@aowi7280: Could be a case of a borderline overloaded 3V3 regulator, and Dshot doing something that drives the power consumption up, like trying to drive a pin that should be an input pin, effectively shorting it out.

jflyper, 2019-02-02:
I think it has something to do with the shared SPI code.

@aowi7280 Is baro detected?

aowi7280, 2019-02-02:
I just turned off the baro in Configurator and the dashes stopped (build 1398). And I was wrong when I said it was OK with Proshot100. The dashes appear with that selected too.

hydra, 2019-02-03:
baro, spi and PWM are all unrelated. you have a hardware issue for sure.

butterSnake, 2019-02-05:
i can confirm thanks to @fujin's direction that using ms125 fixes the issue and or turning off my baro also fixes the issues

butterSnake, 2019-02-05:
diff that works with a fixed hex from @fujin 

# diff all

# version
# Betaflight / OMNIBUSF4FW (OBFW) 4.0.0 Feb  5 2019 / 20:43:14 (a193b5318) MSP API: 1.41

board_name OMNIBUSF4FW
manufacturer_id 
mcu_id 00320035574850062038344b
signature 

# reset configuration to default settings
defaults nosave

# name
name raptor 1750 evo

# display_name

# resources
resource MOTOR 1 B05
resource MOTOR 4 B00
resource SERIAL_TX 1 NONE
resource SERIAL_TX 11 A09

# dmaopt

# mixer
mixer CUSTOM
mmix 0  1.000 -0.815  1.000 -1.000
mmix 1  1.000 -0.815 -1.000  1.000
mmix 2  1.000  0.815  1.000  1.000
mmix 3  1.000  0.815 -1.000 -1.000

# servo

# servo mix

# feature
feature SOFTSERIAL
feature TELEMETRY
feature ESC_SENSOR

# beeper
beeper -GYRO_CALIBRATED
beeper -RX_LOST
beeper -RX_LOST_LANDING
beeper -DISARMING
beeper -ARMING
beeper -ARMING_GPS_FIX
beeper -BAT_CRIT_LOW
beeper -BAT_LOW
beeper -GPS_STATUS
beeper -RX_SET
beeper -ACC_CALIBRATION
beeper -ACC_CALIBRATION_FAIL
beeper -READY_BEEP
beeper -DISARM_REPEAT
beeper -ARMED
beeper -SYSTEM_INIT
beeper -ON_USB
beeper -CRASH_FLIP
beeper -CAM_CONNECTION_OPEN
beeper -CAM_CONNECTION_CLOSE

# beacon
beacon RX_SET

# map
map TAER1234

# serial
serial 2 8192 115200 57600 0 115200
serial 30 32 115200 57600 0 115200

# led

# color

# mode_color

# aux
aux 0 0 0 1700 2100 0 0
aux 1 13 2 1325 1725 0 0
aux 2 26 1 1300 1700 0 0
aux 3 35 1 1700 2100 0 0
aux 4 49 2 1700 2100 0 0

# adjrange

# rxrange

# vtx

# rxfail

# master
set gyro_lowpass_type = PT1
set gyro_lowpass_hz = 0
set yaw_spin_threshold = 1500
set gyro_to_use = SECOND
set dyn_notch_range = MEDIUM
set dyn_notch_width_percent = 0
set dyn_notch_q = 200
set dyn_notch_min_hz = 200
set dyn_lpf_gyro_min_hz = 0
set dyn_lpf_gyro_max_hz = 600
set acc_calibration = 105,64,-242
set mag_hardware = NONE
set rssi_channel = 8
set fpv_mix_degrees = 45
set serialrx_provider = SBUS
set blackbox_p_ratio = 64
set min_throttle = 1070
set dshot_bidir = ON
set use_unsynced_pwm = OFF
set motor_pwm_protocol = DSHOT1200
set motor_pwm_rate = 480
set align_board_roll = 180
set align_board_yaw = 90
set vbat_warning_cell_voltage = 340
set current_meter = ESC
set beeper_dshot_beacon_tone = 4
set yaw_motors_reversed = ON
set small_angle = 180
set deadband = 3
set yaw_deadband = 1
set pid_process_denom = 1
set runaway_takeoff_prevention = OFF
set osd_warn_core_temp = OFF
set osd_warn_rc_smoothing = OFF
set osd_warn_fail_safe = OFF
set osd_warn_launch_control = OFF
set osd_warn_no_gps_rescue = OFF
set osd_warn_gps_rescue_disabled = OFF
set osd_rssi_alarm = 50
set osd_cap_alarm = 1050
set osd_alt_alarm = 135
set osd_esc_temp_alarm = 95
set osd_rssi_pos = 248
set osd_g_force_pos = 45
set osd_throttle_pos = 1
set osd_current_pos = 54
set osd_mah_drawn_pos = 2068
set osd_altitude_pos = 11
set osd_avg_cell_voltage_pos = 23
set osd_esc_tmp_pos = 2054
set osd_esc_rpm_pos = 33
set osd_core_temp_pos = 45
set osd_stat_tim_1 = ON
set osd_stat_tim_2 = OFF
set osd_stat_max_spd = OFF
set osd_stat_endbatt = ON
set osd_stat_min_rssi = OFF
set osd_stat_max_alt = ON
set osd_stat_bb_no = OFF
set osd_stat_max_g_force = ON
set osd_stat_max_esc_temp = ON
set osd_stat_max_esc_rpm = ON
set debug_mode = GYRO_SCALED
set cpu_overclock = 216MHZ
set scheduler_optimize_rate = ON
set vtx_band = 5
set vtx_channel = 2
set vtx_power = 2
set vtx_freq = 5695

# profile
profile 0

set dyn_lpf_dterm_min_hz = 200
set dyn_lpf_dterm_max_hz = 400
set dterm_lowpass_hz = 200
set dterm_lowpass2_hz = 0
set anti_gravity_gain = 7000
set pidsum_limit_yaw = 700
set yaw_lowpass_hz = 500
set throttle_boost = 7
set p_pitch = 52
set i_pitch = 80
set d_pitch = 42
set f_pitch = 66
set p_roll = 49
set i_roll = 70
set d_roll = 36
set f_roll = 64
set p_yaw = 15
set i_yaw = 120
set d_min_pitch = 15
set d_min_roll = 13
set d_boost_gain = 25
set launch_trigger_throttle_percent = 30

# profile
profile 1

set dyn_lpf_dterm_min_hz = 200
set dyn_lpf_dterm_max_hz = 400
set dterm_lowpass_hz = 200
set dterm_lowpass2_hz = 0
set anti_gravity_gain = 7000
set pidsum_limit_yaw = 700
set yaw_lowpass_hz = 500
set throttle_boost = 7
set p_pitch = 52
set i_pitch = 80
set d_pitch = 25
set f_pitch = 66
set p_roll = 49
set i_roll = 70
set d_roll = 22
set f_roll = 64
set p_yaw = 15
set i_yaw = 120
set launch_trigger_throttle_percent = 30

# profile
profile 2

# restore original profile selection
profile 0

# rateprofile
rateprofile 0

set thr_mid = 40
set thr_expo = 65
set roll_rc_rate = 75
set pitch_rc_rate = 75
set yaw_rc_rate = 75
set roll_expo = 30
set pitch_expo = 30
set yaw_expo = 30
set roll_srate = 85
set pitch_srate = 85
set yaw_srate = 85
set tpa_rate = 90
set tpa_breakpoint = 1750
set throttle_limit_type = SCALE

# rateprofile
rateprofile 1

set thr_mid = 40
set thr_expo = 65
set roll_rc_rate = 75
set pitch_rc_rate = 75
set yaw_rc_rate = 75
set roll_expo = 30
set pitch_expo = 30
set yaw_expo = 30
set roll_srate = 65
set pitch_srate = 65
set yaw_srate = 65
set tpa_rate = 90
set tpa_breakpoint = 1750
set throttle_limit_type = SCALE
set throttle_limit_percent = 85

# rateprofile
rateprofile 2

set thr_mid = 40
set thr_expo = 65
set roll_rc_rate = 75
set pitch_rc_rate = 75
set yaw_rc_rate = 75
set roll_expo = 30
set pitch_expo = 30
set yaw_expo = 30
set roll_srate = 85
set pitch_srate = 85
set yaw_srate = 85

# rateprofile
rateprofile 3

# rateprofile
rateprofile 4

# rateprofile
rateprofile 5

set tpa_rate = 60
set tpa_breakpoint = 1250

# restore original rateprofile selection
rateprofile 0

butterSnake, 2019-02-05:
diff that needs oneshot or baro off 

# diff all

# version
# Betaflight / OMNIBUSF4FW (OBFW) 4.0.0 Jan 31 2019 / 06:04:31 (a9fcf87d8) MSP API: 1.41

board_name OMNIBUSF4FW
manufacturer_id 
mcu_id 00320035574850062038344b
signature 

# reset configuration to default settings
defaults nosave

# name
name raptor 1750 evo

# resources
resource MOTOR 1 B05
resource MOTOR 4 B00
resource SERIAL_TX 1 NONE
resource SERIAL_TX 11 A09

# dmaopt

# mixer
mixer CUSTOM
mmix 0 1.000 -0.815 1.000 -1.000
mmix 1 1.000 -0.815 -1.000 1.000
mmix 2 1.000 0.815 1.000 1.000
mmix 3 1.000 0.815 -1.000 -1.000

# servo

# servo mix

# feature
feature SOFTSERIAL
feature TELEMETRY
feature AIRMODE
feature ESC_SENSOR

# beeper
beeper -GYRO_CALIBRATED
beeper -RX_LOST
beeper -RX_LOST_LANDING
beeper -DISARMING
beeper -ARMING
beeper -ARMING_GPS_FIX
beeper -BAT_CRIT_LOW
beeper -BAT_LOW
beeper -GPS_STATUS
beeper -RX_SET
beeper -ACC_CALIBRATION
beeper -ACC_CALIBRATION_FAIL
beeper -READY_BEEP
beeper -DISARM_REPEAT
beeper -ARMED
beeper -SYSTEM_INIT
beeper -ON_USB
beeper -CRASH_FLIP
beeper -CAM_CONNECTION_OPEN
beeper -CAM_CONNECTION_CLOSE

# beacon
beacon RX_SET

# map
map TAER1234

# serial
serial 2 8192 115200 57600 0 115200
serial 30 32 115200 57600 0 115200

# led

# color

# mode_color

# aux
aux 0 0 0 1700 2100 0 0
aux 1 13 2 1325 1725 0 0
aux 2 26 1 1300 1700 0 0
aux 3 35 1 1700 2100 0 0
aux 4 49 2 1700 2100 0 0

# adjrange

# rxrange

# vtx

# rxfail

# master
set gyro_lowpass_type = PT1
set gyro_lowpass_hz = 0
set yaw_spin_threshold = 1500
set gyro_to_use = SECOND
set dyn_notch_range = MEDIUM
set dyn_notch_width_percent = 0
set dyn_notch_q = 200
set dyn_notch_min_hz = 200
set dyn_lpf_gyro_min_hz = 0
set dyn_lpf_gyro_max_hz = 600
set acc_calibration = 105,64,-242
set mag_hardware = NONE
set rssi_channel = 8
set fpv_mix_degrees = 45
set serialrx_provider = SBUS
set blackbox_p_ratio = 64
set min_throttle = 1070
set dshot_bidir = ON
set use_unsynced_pwm = OFF
set motor_pwm_protocol = DSHOT1200
set motor_pwm_rate = 480
set align_board_roll = 180
set align_board_yaw = 90
set vbat_warning_cell_voltage = 340
set current_meter = ESC
set beeper_dshot_beacon_tone = 4
set yaw_motors_reversed = ON
set small_angle = 180
set deadband = 3
set yaw_deadband = 1
set pid_process_denom = 1
set runaway_takeoff_prevention = OFF
set osd_warn_core_temp = OFF
set osd_warn_rc_smoothing = OFF
set osd_warn_fail_safe = OFF
set osd_warn_launch_control = OFF
set osd_warn_no_gps_rescue = OFF
set osd_warn_gps_rescue_disabled = OFF
set osd_rssi_alarm = 50
set osd_cap_alarm = 1050
set osd_alt_alarm = 135
set osd_esc_temp_alarm = 95
set osd_rssi_pos = 248
set osd_g_force_pos = 45
set osd_throttle_pos = 1
set osd_current_pos = 54
set osd_mah_drawn_pos = 2068
set osd_altitude_pos = 11
set osd_avg_cell_voltage_pos = 23
set osd_esc_tmp_pos = 2054
set osd_esc_rpm_pos = 33
set osd_core_temp_pos = 45
set osd_stat_tim_1 = ON
set osd_stat_tim_2 = OFF
set osd_stat_max_spd = OFF
set osd_stat_endbatt = ON
set osd_stat_min_rssi = OFF
set osd_stat_max_alt = ON
set osd_stat_bb_no = OFF
set osd_stat_max_g_force = ON
set osd_stat_max_esc_temp = ON
set osd_stat_max_esc_rpm = ON
set debug_mode = GYRO_SCALED
set cpu_overclock = 216MHZ
set scheduler_optimize_rate = ON
set vtx_band = 5
set vtx_channel = 2
set vtx_power = 2
set vtx_freq = 5695

# profile
profile 0

set dyn_lpf_dterm_min_hz = 200
set dyn_lpf_dterm_max_hz = 400
set dterm_lowpass_hz = 200
set dterm_lowpass2_hz = 0
set anti_gravity_gain = 7000
set pidsum_limit_yaw = 700
set yaw_lowpass_hz = 500
set throttle_boost = 7
set p_pitch = 52
set i_pitch = 80
set d_pitch = 25
set f_pitch = 66
set p_roll = 49
set i_roll = 70
set d_roll = 22
set f_roll = 64
set p_yaw = 15
set i_yaw = 120
set dterm_cut_percent = 66
set dterm_cut_gain = 25
set dterm_cut_setpoint_gain = 10
set launch_trigger_throttle_percent = 30

# profile
profile 1

set dyn_lpf_dterm_min_hz = 200
set dyn_lpf_dterm_max_hz = 400
set dterm_lowpass_hz = 200
set dterm_lowpass2_hz = 0
set anti_gravity_gain = 7000
set pidsum_limit_yaw = 700
set yaw_lowpass_hz = 500
set throttle_boost = 7
set p_pitch = 52
set i_pitch = 80
set d_pitch = 25
set f_pitch = 66
set p_roll = 49
set i_roll = 70
set d_roll = 22
set f_roll = 64
set p_yaw = 15
set i_yaw = 120
set dterm_cut_percent = 66
set dterm_cut_gain = 25
set launch_trigger_throttle_percent = 30

# profile
profile 2

# restore original profile selection
profile 0

# rateprofile
rateprofile 0

set thr_mid = 40
set thr_expo = 65
set roll_rc_rate = 75
set pitch_rc_rate = 75
set yaw_rc_rate = 75
set roll_expo = 30
set pitch_expo = 30
set yaw_expo = 30
set roll_srate = 85
set pitch_srate = 85
set yaw_srate = 85
set tpa_rate = 90
set tpa_breakpoint = 1750
set throttle_limit_type = SCALE

# rateprofile
rateprofile 1

set thr_mid = 40
set thr_expo = 65
set roll_rc_rate = 75
set pitch_rc_rate = 75
set yaw_rc_rate = 75
set roll_expo = 30
set pitch_expo = 30
set yaw_expo = 30
set roll_srate = 65
set pitch_srate = 65
set yaw_srate = 65
set tpa_rate = 90
set tpa_breakpoint = 1750
set throttle_limit_type = SCALE
set throttle_limit_percent = 85

# rateprofile
rateprofile 2

set thr_mid = 40
set thr_expo = 65
set roll_rc_rate = 75
set pitch_rc_rate = 75
set yaw_rc_rate = 75
set roll_expo = 30
set pitch_expo = 30
set yaw_expo = 30
set roll_srate = 85
set pitch_srate = 85
set yaw_srate = 85

# rateprofile
rateprofile 3

# rateprofile
rateprofile 4

# rateprofile
rateprofile 5

set tpa_rate = 60
set tpa_breakpoint = 1250

# restore original rateprofile selection
rate profile 0

fujin, 2019-02-05:
I also had @butterSnake test with #6809 reverted and that worked with DSHOT/BARO both ON.

aowi7280, 2019-02-06:
I noticed the baro and OSD are both on the same SPI address. Is that a problem?

# get spi_
mag_spi_device = 0
Allowed range: 0 - 3

baro_spi_device = 3
Allowed range: 0 - 5

sdcard_spi_bus = 2
Allowed range: 0 - 3

max7456_spi_bus = 3
Allowed range: 0 - 3

hydra, 2019-03-08:
oops! mouse slipped.  restored issue state.

closed 2019-03-15 by stale[bot]