betaflight/betaflight #7512: BETA 4.0 #1400 remaping issue

Oriolkun, 2019-02-02:
The last BETA 4.0 #1400 looks having an issue with remaping.
FC: OMNIBUSF4SD. 
When trying to remap MOTOR 4 to B00 and MOTOR 1 to A03, suddenly MOTOR 3 (on A02) stop working like no signal were directed to the ESC of that motor.
I have used copy/paste diff coming from BF 3.4 . Copy/paste diff to BF 4.0 #1400 motor 3 do not respond. After a "defaults" I did a remap manually and motor 3 stop working after remaping M4 and M1.
Then copy/paste diff to BF 3.5.5 and motor works as BF 3.4, everything is ok.

I have other builds with older releases of Beta 4.0 and they have no problem with remaping operations.
Regards

jflyper, 2019-02-02:
Give us full `resource` commands that you are using.

jflyper, 2019-02-02:
Your description is confusing.

> When trying to remap MOTOR 4 to B00 and MOTOR 1 to A03, suddenly MOTOR 3 (on A02) stop working like no signal were directed to the ESC of that motor.

The original mapping is
motor 1 B0
motor 2 B1
motor 3 A3
motor 4 A2
So, if you assign motor 4 to B0 and motor 1 to A3, then you will have
motor 1 A3
motor 2 B1
motor 3 A3
motor 4 B0
No wonder something connected to A2 is not working.

Again, the full set of `resource` commands speaks by itself.

Oriolkun, 2019-02-02:
> Again, the full set of `resource` commands speaks by itself.

The resource not working on 4.0 #1400 is:

`# rescource
Unknown command, try 'help'
# resource
resource BEEPER 1 B04
resource MOTOR 1 A03
resource MOTOR 2 B01
resource MOTOR 3 A02
resource MOTOR 4 B00
resource MOTOR 5 A01
resource MOTOR 6 A08
resource PPM 1 B08
resource PWM 1 B08
resource PWM 2 B09
resource PWM 3 C06
resource PWM 4 C07
resource PWM 5 C08
resource PWM 6 C09
resource SONAR_TRIGGER 1 A01
resource SONAR_ECHO 1 A08
resource LED_STRIP 1 B06
resource SERIAL_TX 1 A09
resource SERIAL_TX 3 B10
resource SERIAL_TX 6 C06
resource SERIAL_RX 1 A10
resource SERIAL_RX 3 B11
resource SERIAL_RX 6 C07
resource INVERTER 3 C09
resource INVERTER 6 C08
resource LED 1 B05
resource SPI_SCK 1 A05
resource SPI_SCK 2 B13
resource SPI_SCK 3 C10
resource SPI_MISO 1 A06
resource SPI_MISO 2 B14
resource SPI_MISO 3 C11
resource SPI_MOSI 1 A07
resource SPI_MOSI 2 B15
resource SPI_MOSI 3 C12
resource ESCSERIAL 1 B08
resource ADC_BATT 1 C02
resource ADC_RSSI 1 A00
resource ADC_CURR 1 C01
resource BARO_CS 1 B03
resource SDCARD_CS 1 B12
resource SDCARD_DETECT 1 B07
resource FLASH_CS 1 B12
resource OSD_CS 1 A15
resource SPI_PREINIT_IPU 1 A04
resource SPI_PREINIT_IPU 2 B12
resource SPI_PREINIT_IPU 3 B03
resource SPI_PREINIT_IPU 4 A15`

But it actually works on 3.4 and 3.5.5

Talking with some friend told me it realised about this issue last past week on a previouse BF4.0 release and it was forced to change the phisical ESC pin to solve that.

jflyper, 2019-02-03:
@Oriolkun Issue reproduced and problematic commit identified.

jflyper, 2019-02-03:
@joelucid 
#7264 seems to cause this one.

Non-contiguous pins of a port group is assumed or something?

kajsilander78, 2019-02-12:
#7366 Same issue? Found that before this :) Anyway, having same issue on kakutef4v2, remapping stops one motor...

joelucid, 2019-02-14:
This issue is like addressed by https://github.com/betaflight/betaflight/pull/7573. I can remap motors on my banggood NOXe without problems after this fix.

joelucid, 2019-02-14:
@Oriolkun, could you try this hex, please?

[betaflight_4.0.0_OMNIBUSF4SD.hex.zip](https://github.com/betaflight/betaflight/files/2864645/betaflight_4.0.0_OMNIBUSF4SD.hex.zip)

Thunkar, 2019-02-14:
@joelucid I had the exact same problem and just tested the hex. It fixes the issue for me, thanks a lot!

closed 2022-05-08 by haslinghuis