betaflight/betaflight #7522: ESC Telemetry in OSD (individual ESCs)

peteoz, 2019-02-03:
**Is your feature request related to a problem? Please describe.**
not related to a problem but current implementation of ESC Telemetry in OSD is not of much use when we only see a single value but not values for all 4 ESCs (assuming quadcopter).
eg. RPM value is the average of all 4 ESCs, this is of no use if I want to analyse in-flight RPM correlation between rear and front motors based on flight characteristics, different props, 3D flying, different CG (battery position, battery size), longer rear arms vs front arms,...

temperature is just a single value not even knowing what ESC it is, just knowing it's the hottest one, but which one ? if for example one ESC readout is 50% more than the other 3 then I want to know about it and investigate, if the variation is only 1% then I don't care, but we can't tell.

**Describe the solution you'd like**
would like to be able to set up my OSD in such a way that I can see all 4 ESCs individual telemetry data for post-flight analysis, with each value to have its own position within the OSD layout (ie. to be able to position each value in different corner within the OSD, each corner to represent corresponding motor/esc)

**Describe alternatives you've considered**
tried setting up debug_mode to esc_sensor_rpm and then enabling "debug" on my OSD, while this works it's not really that practical - first of all I need to disable my preferred debug_mode ("gyro_scaled") and secondly the data in OSD debug option represent itself in a single line, this is hard to read when analysing the RPM data while replaying the DVR footage (eg. trying to visualise that my front motors are 2nd and 4th value from that row)
 
**Additional context**
ESC rpm values are already sent to the FC, I'm only asking to have the data available in OSD.
it's been already something like 2 years (??) since ESC telemetry was introduced to BF and still we can't access the data in OSD the same way it's been done on Kiss for quite some time, while at the same time ESC telemetry data was one of the main selling points when 32bit ESCs were first introduced.

peteoz, 2019-03-05:
any info from any BF developer in the "know" would really help.
is ESC telemetry dead now that BF4 is making it possible to send RPM data over signal wire ?
is the new implementation accurate ?

just trying to determine whether it's worthwhile to run wire from ESC telemetry pads on my next build.

still the original feature request detailed above - to be able to display RPM data for each individual motor in OSD (wether it's through telemetry data or signal wire) - any feedback on that ?

thanks

mikeller, 2019-03-05:
@peteoz:

> is ESC telemetry dead now that BF4 is making it possible to send RPM data over signal wire ?

Depends on what you want - for now bidirectional Dshot only carries RPM information. For any of the other values you can get from an ESC (voltage, current, consumption, temperature) you'll need to run a separate telemetry wire.

peteoz, 2019-03-05:
@mikeller is the new RPM filter meant to work only over signal wire, or is it also going to work through the actual ESC telemetry ?
and if it works with both solutions, which one of the two is more accurate as far as RPM data accuracy ?

thanks

mikeller, 2019-03-05:
@peteoz:

> is the new RPM filter meant to work only over signal wire, or is it also going to work through the actual ESC telemetry ?

It _currently_ only works over the signal wire - using a separate serial connection for it is being investigated, but we don't know yet if it is feasible.

> which one of the two is more accurate as far as RPM data accuracy ?
I would assume that if and when we get it to work over the separate wire, both of them will be equal in terms of accuracy.

closed 2019-04-11 by stale[bot]