betaflight/betaflight #7523: Super_S F4 FC doesn't support uninverted Sbus signal. OmnibusF4SD target.

Durete, 2019-02-03:
Hi folks.
I have 3 Micro/Tinywhoop copters using the FSD Nano V2 RX on Super_S F4 flight controllers, this FC run on OmnibusF4SD target.
This receiver is a little bit special, because the output is uninverted Sbus signal instead normal Sbus signal.
To use this receivers on Super_S F4 FC, I use the DSM/iBus pad. This pad is the same as Sbus pad but jumping the fixed hardware inverter on Sbus pad.
I active UART1 as serial_RX and select Sbus protocol on Betaflight, this works perfect till Betaflight 3.5.5.
I'm trying to install development versions of Betaflight 4 from about one month, and can't make to work this receiver in none of my 3 micro copters. The RX signal is not detected by the FC.
Using an standard Sbus receiver as XM+ attached to Sbus pad works perfect, but using uninverted Sbus signal on DSM/iBus pad the FC can't detect the receiver signal.
Yesterday, I tried to flash this FC with a similar target, OmnibusF4V6. I know this is not the right target, but since most resources are the same and I'm a little bit desperate :) , I decide to give it a chance.
I repeated the same steps with this target, and surprisingly the RX works perfect on this target, same as Betaflight 3.5.5 and older versions.
I'm not sure from what build started this bad behaviour, but at least from about one month old versions.

To reproduce this bug, you can attach an uninverted Sbus signal to DSM/iBus pad (RX1) with a fresh installation with only UART1-Serial_RX and Sbus protocol configured.

This are the resources from OmnibusF4SD on build #1403:
# resource
resource BEEPER 1 B04
resource MOTOR 1 B00
resource MOTOR 2 B01
resource MOTOR 3 A03
resource MOTOR 4 A02
resource MOTOR 5 A01
resource MOTOR 6 A08
resource PPM 1 B08
resource PWM 1 B08
resource PWM 2 B09
resource PWM 3 C06
resource PWM 4 C07
resource PWM 5 C08
resource PWM 6 C09
resource SONAR_TRIGGER 1 A01
resource SONAR_ECHO 1 A08
resource LED_STRIP 1 B06
resource SERIAL_TX 1 A09
resource SERIAL_TX 3 B10
resource SERIAL_TX 6 C06
resource SERIAL_RX 1 A10
resource SERIAL_RX 3 B11
resource SERIAL_RX 6 C07
resource INVERTER 3 C09
resource INVERTER 6 C08
resource LED 1 B05
resource SPI_SCK 1 A05
resource SPI_SCK 2 B13
resource SPI_SCK 3 C10
resource SPI_MISO 1 A06
resource SPI_MISO 2 B14
resource SPI_MISO 3 C11
resource SPI_MOSI 1 A07
resource SPI_MOSI 2 B15
resource SPI_MOSI 3 C12
resource ESCSERIAL 1 B08
resource ADC_BATT 1 C02
resource ADC_RSSI 1 A00
resource ADC_CURR 1 C01
resource BARO_CS 1 B03
resource SDCARD_CS 1 B12
resource SDCARD_DETECT 1 B07
resource FLASH_CS 1 B12
resource OSD_CS 1 A15
resource GYRO_EXTI 1 C04
resource GYRO_CS 1 A04
resource USB_DETECT 1 C05

#

mikeller, 2019-02-03:
@Durete: Can you please supply a `diff`, like you were asked when opening the issue?

jflyper, 2019-02-03:
@Durete 
Genuine Omnibus F4 V3 (OMNIBUSF4SD target), flashed with current master, loaded with your config  `diff` output, works great with a un-inverted SBUS input at UART1_RX without any problem.

I've tested this with Futaba R7008SB receiver with SBUS inverter cable and S-Pad of R-XSR.

I think the problem is hardware oriented.

Durete, 2019-02-03:
@jflyper
I'm not sure if this is hardware related, but for sure the software is affecting this behaviour. I'm running 3 Brushless Tinywhoop using this receiver from about 1 year ago, using older Betafligt builds.
This combination and setup works perfect always using Betafligt 3.5.5 and older versions. I found this issue as soon I started to test BF4 builds.
I will start to test older BF4 builds trying to find from what build this behaviour started.
Thanks!

jflyper, 2019-02-03:
@Durete 

However, since you are saying that you have no trouble with 3.5.x, there must have been a change that caused the issue.

There's only one way to find out. Run bisect between 3.5 and current master. The run is (only) 9 to 10 build and test cycles, and can only be performed by someone with the problematic hardware (means you).

Durete, 2019-02-03:
Ok, after test a bunch of Jenkin builds, I have found latest working build is #1100 
Build #1101 have this erratic behaviour.
So probably problematic commit is https://github.com/betaflight/betaflight/commit/7665a424c7386ed90f7bd042e265a845a41fd63e

This is above my knowledge in programming, I don't know if this is possible.

jflyper, 2019-02-04:
Timer???

Does your flight controller share RX1 pad with something else? PPM?

Durete, 2019-02-04:
@jflyper
The FC shares Sbus/PPM pad. Sbus is RX1 across the hardware inverter.
https://img.staticbg.com/images/oaupload/banggood/images/EC/0A/f9c08e65-1f8b-4301-a503-8f22aca0e066.jpg

I will test as soon I return home from work.
Thanks!

Durete, 2019-02-04:
> Try this
> 
> ```
> resource pinio 1 b8
> set pinio_config = 1,1,1,1
> ```

Just tested with latest Jenkins build ( #1411 ).
This works perfect :)

jflyper, 2019-02-04:
Okay, just as what I thought was happening.

This is caused by inverter circuit driven by PPM (PB8) pin interfering with RX1 input.
Before 7665a42, PB8 was initialized to actively output low level as a part of timer initialization. The inverter circuit seems to have being designed so that it would not drive the output hard when input is low.

After 7665a42, timer initialization is not initializing the PB8 pin anymore, and it remains at input with pull-up which is setup by startup code. Since it is pulled up, inverter sees high level on input and drives output hard (probably sinking current) to low. This must have caused the input from the receiver connected to bare RX1 to be sinked also, rendering the input to flat low.

The solution above force PB8 pin to go high to prevent this from happening.

We might want a wiki entry for this case ...

Durete, 2019-02-04:
> Okay, just as what I thought was happening.
>
> This is caused by inverter circuit driven by PPM (PB8) pin interfering with RX1 input.
> Before [...]

That's sounds like a perfect explanation to me.
I also suspected from something related to signal levels.
Great job ;)

mikeller, 2019-02-05:
@jflyper: How'd you figure that one out? :wink:

Durete, 2019-02-05:
If I'm not wrong, according the Wiki, this is not the only one OmnibusF4SD board sharing Sbus/PPM.
So maybe this could be problematic with other flight controllers?

mikeller, 2019-02-05:
@Durete: Sharing SBUS and PPM on the same (UART RX capable) pin is done by a lot of flight controllers, and works well. What is unusual about this setup is that in this case, the pin is also used to drive an inverter.

jflyper, 2019-02-06:
@mikeller It's not that PPM (PB8) is controlling the inverter. Pull-up at PB8 is back flowing into inverter that cause interference at UART1_RX.

This is my assumption.

[image]

The PPM/SBUS shared pad is known (at least for me) to cause trouble when using "_exotic_" receivers. In the past, an OpenLRS receiver in PPM mode was unhappy with a pull-up at PPM input. The particular board was Omnibus F4, which has an external pull-up resistor at PPM input to avoid the inverter drive the  UART1_RX hard low. It was ironic that this resistor caused the LRS receiver not to work. The solution was to remove the pull-up resistor ...

mikeller, 2019-02-07:
@jflyper: Thanks for the explanation. This seems to be a less than ideal design anyway, as it makes it impossible to use PB8 and PA10 at the same time, and PPM could as well be received on PA10...

closed 2019-03-16 by stale[bot]