betaflight/betaflight #7524: Frsky XM+ RSSI scaling

methisto, 2019-02-03:
Hello!
The scaling of the RSSI value in the OSD dont work correct, so 100% is 100% but 50% in the OSD is 0% RSSI in real, so when i lose connection the RSSI value is about 50% in the OSD! What i am doing wrong? thanks

mikeller, 2019-02-03:
@methisto: Use `rssi_scale` and `rssi_offset` to scale the RSSI to what you want it to be.

methisto, 2019-02-03:
OK when i change the rssi scale it goes from 200 to 100 but what is RSSI offset?

mikeller, 2019-02-03:
`rssi_offset` allows you to adjust where 0 RSSI is reported - leave it alone if all you want is for the top to be 100.

tomstom, 2019-02-04:
Try:
set rssi_offset = -100
set rssi_scale = 200

then you have the scale from 0 to 100 in the OSD, but remark, the Frsky do the failsave at round about 30 in the OSD!

methisto, 2019-02-05:
Thank you will try!
Range is amazing, got 2km today with this reciever on tyro99!

closed 2019-03-14 by stale[bot]