betaflight/betaflight #7525: RunCam compatibility with Arduino

elizh12, 2019-02-04:
**Is your feature request related to a problem? Please describe.**
It would be great since the newer run cams such as the RunCam split s2 has the capability to run through Uart and the Rx/Tx connections for it to have code and libraries that are compatible with Arduino and other microcontrollers instead of simply through flight controllers and beta flight 

**Describe the solution you'd like**
Arduino compatible example code and library. Or even just clearer information on how the CRC8 algorithms work, or for some information on how to implement these things and if the device protocol is  found here https://support.runcam.com/hc/en-us/articles/360014537794-RunCam-Device-Protocol is something that could be implemented.

**Describe alternatives you've considered**
I am attempting to connect RunCam using uart and the RX/TX to a teensy 3.6 microcontroller.
I have been trying to use the command packet structures through the device protocol page found here 
https://support.runcam.com/hc/en-us/articles/360014537794-RunCam-Device-Protocol
I found the CRC8 algorithms through the betaflight firmware in main/src/io/common/crc/c which is hopefully the crc8_dvb_s2
I tried to modify some of the functions found on rcdevice_cam.c or rcdevice.c but the systems of buffer, pointers, and classes are very hard to work with and seperate from the source code.
I have been attempting to use realterm and a serial connector that connects rx/tx through to my usb port and send hex commands through there, calculating the crc8 seperately, to establish that the packet structure is correct. this has not worked, either due to connection or incorrect crc. 
since the packet structure detailed in device protocol is (Header, Command ID, Action ID, crc8) 
I used one of the commands under camera control which has the structure, Header=0xCC, Command ID= 0x01, Action ID=0x01 and the crc8 I calculated being 0xE7. 
this did not do anything. I tried to modify the functions found in rcdevice.h and rcdevice_cam.h but have not had much success.

jflyper, 2019-02-04:
@elizh12 It's not clear what the issue is in terms of runcam support in Betaflight.

ijustwannafly, 2019-02-04:
From my perspective...there is no issue with regard to direct correlation between runcam and betaflight operation.

This is a request (if you will) by someone who is trying to use runcam with arduino based hardware and is hoping the betaflight devs will provide source code to make it possible.

jflyper, 2019-02-04:
@ijustwannafly Thanks for the translation. Yeah, I didn't read the "desired solution" section carefully.
I think this issue will be closed, as it has nothing to do with Betaflight firmware.

@elizh12 I would logic scope working setup to see what's exchanged between an FC and the cam.

@azolyoung You may want to help this guy.

closed 2019-02-04 by jflyper

mikeller, 2019-02-04:
@elizh12: Have a look at the following files:

https://github.com/betaflight/betaflight/blob/master/src/main/io/rcdevice.c
https://github.com/betaflight/betaflight/blob/master/src/main/io/rcdevice_cam.c

This is the source code for the RunCam drivers in Betaflight. It should be possible with some changes, and probably some simplification, to make this build on your target platform.

elizh12, 2019-02-04:
My apologies I am aware that this is not completely relevant to betaflight, however there is just no seperate github repository for the runcam camera that I could bring this up in.

Ozyris, 2019-10-27:
What is the current runcam protocol version and what byte value in packet describes it?

Ozyris, 2019-11-04:
> since the packet structure detailed in device protocol is (Header, Command ID, Action ID, crc8)
> I used one of the commands under camera control which has the structure, Header=0xCC, Command ID= [...]

Packet must be 1byte longer, because Action ID is uint16.