betaflight/betaflight #7531: Generic Target using F405 and Config Files has unexpected behaviour

brettcollis2212, 2019-02-04:
**Describe the bug**
When trying to get a FC that doesn't have a target working under the latest Betaflight builds using the F405 Generic Target and then implementing the Resources using a config file the Battery ADC doesn't work or display any voltage within the configurator

**To Reproduce**
Flash F405 Target and use the relevant config file for resource mapping such as this:- https://github.com/betaflight/betaflight/blob/master/src/main/target/STM32F405/config/FF_RACEPIT.config

Plug in a battery and no Voltage reading is produced,

Flash the Fortini Target and then use the same config file and Voltage will display correctly within the configurator

**Expected behavior**
The Voltage reading and ADC should work with both the targets I believe, However it only works with the Fortini target.

**Flight controller configuration**
N/A
**Setup / Versions**

Also verified by flashing the Fortini with the F405 Target and doing the relevant resource mapping required for it, Voltage ADC still didn't work correctly however flashing with fortini target got it working correctly

**Additional context**
Add any other context about the problem here.

jflyper, 2019-02-05:
@brettcollis2212 

The generic STM32F405 target works with Omnibus F4 V3, configured with "relevant" CLI lines.

I suspect your config does not include `dmaopt`, which is a new addition to configurable-family, for the ADC device used.

These are ADC related lines from Omnibus F4 V3 config.
```
adc_device = 2
dmaopt ADC 2 0 # DMA2 Stream 2 Channel 1
```

For FF_FORTINI, I believe these should be
```
adc_device = 1
dmaopt ADC 1 1 # DMA2 Stream 4 channel 0
```

mikeller, 2019-02-05:
@brettcollis2212: Yes, it will have to be included. Unified targets are very much still work in progress, with support for new CLI settings (like `dmaopt`) being added over time - there are still a few more CLI settings that are needed before unified targets are ready for prime time.
This means that unified target configurations that were defined early on can be out of date, and all target configurations will have to get a once-over once all of the required CLI settings for unified targets are complete.

mikeller, 2019-03-07:
@brettcollis2212: Unified target support is now pretty much complete - use the configuration options listed in a `dump hardware` to set up your target.

Any feedback on unified target support appreciated!

closed 2019-04-13 by stale[bot]