betaflight/betaflight #7533: GPS Home Point First Arm Only

Gaz-B, 2019-02-04:
Could we have an option to only have GPS set home point on first arm only. I use it to find my way back through very thick woods if you crash and need to rearm resetting home is a bit of an issue.

Gaz-B, 2019-02-05:
That reads to Me that it will still reset the home point if rescue isn't being used. Rescue is for those that fly long distance in open space. Flying under the tree canopy rescue is to be avoided I want failsafe to drop me not smash the quad to bits whilst getting stuck in the top of a tree.

McGiverGim, 2019-02-05:
No. If you enable the new mode, it will not reset the home point once fixed, until you unplug the battery. You can have Rescue disabled.

Gaz-B, 2019-02-05:
mikeller 

> @Gaz-B: Can you please confirm that this has been addressed in #7369, and close this?

Yes after a few responses on 7369 it does do what I need now to start playing with 4.0