betaflight/betaflight #7544: ACC and MAG missing after flashing Build 1414 (likely related to Build 1413

gvhlaw, 2019-02-06:
THE BUG - After flashing latest master (1414) and re-connecting (Omnibus F4 and F4SD targets) on the first page of the configurator with the moving model, Accelerometer and Magnetometer are "grayed-out" even though enabled in config.  Moving model does not move (no ACC).  Versions (targets) were verified in CLI.  Reverting to an earlier build produced the desired result - both ACC and MAG show-up as normal.

Flash 1414 to reproduce the bug.

Diff All:

```diff all

# version
# Betaflight / OMNIBUSF4 (OBF4) 4.0.0 Feb  5 2019 / 20:40:57 (c0707dbd1) MSP API: 1.41

board_name OMNIBUSF4
manufacturer_id AIRB
mcu_id 004e002b4748500220373450
signature 

# reset configuration to default settings
defaults nosave

# name
name MD2020

# display_name

# resources

# dmaopt

# mixer

# servo

# servo mix

# feature
feature -RX_SPI
feature RX_SERIAL

# beeper
beeper -ARMING_GPS_FIX
beeper -DISARM_REPEAT
beeper -ARMED
beeper -ON_USB
beeper -BLACKBOX_ERASE
beeper -CRASH_FLIP
beeper -CAM_CONNECTION_OPEN
beeper -CAM_CONNECTION_CLOSE

# beacon

# map

# serial
serial 0 64 115200 57600 0 115200
serial 2 8192 115200 57600 0 115200

# led

# color

# mode_color

# aux
aux 0 0 0 1750 2100 0 0
aux 1 1 1 900 1300 0 0
aux 2 13 2 1275 1600 0 0
aux 3 28 1 1725 2100 0 0
aux 4 35 2 1800 2100 0 0

# adjrange

# rxrange

# vtx

# rxfail
rxfail 5 s 1100

# master
set dyn_lpf_gyro_max_hz = 650
set acc_calibration = 14,-112,37
set mag_hardware = NONE
set baro_hardware = NONE
set min_check = 1030
set max_check = 2000
set rssi_channel = 16
set rc_interp = MANUAL
set rc_smoothing_input_hz = 25
set rc_smoothing_derivative_hz = 25
set serialrx_provider = SBUS
set blackbox_p_ratio = 64
set dshot_idle_value = 600
set motor_pwm_protocol = DSHOT600
set failsafe_delay = 30
set current_meter = NONE
set beeper_dshot_beacon_tone = 3
set yaw_motors_reversed = ON
set small_angle = 180
set deadband = 2
set yaw_deadband = 2
set pid_process_denom = 1
set osd_warn_esc_fail = OFF
set osd_warn_core_temp = OFF
set osd_vbat_pos = 2433
set osd_rssi_pos = 2081
set osd_tim_1_pos = 54
set osd_tim_2_pos = 2423
set osd_flymode_pos = 2444
set osd_throttle_pos = 225
set osd_vtx_channel_pos = 2455
set osd_ah_pos = 200
set osd_current_pos = 385
set osd_mah_drawn_pos = 353
set osd_craft_name_pos = 2401
set osd_gps_speed_pos = 218
set osd_gps_lon_pos = 82
set osd_gps_lat_pos = 65
set osd_gps_sats_pos = 51
set osd_home_dir_pos = 302
set osd_home_dist_pos = 303
set osd_compass_bar_pos = 266
set osd_altitude_pos = 247
set osd_pid_roll_pos = 423
set osd_pid_pitch_pos = 455
set osd_pid_yaw_pos = 487
set osd_debug_pos = 1
set osd_power_pos = 321
set osd_pidrate_profile_pos = 2106
set osd_warnings_pos = 2409
set osd_avg_cell_voltage_pos = 2379
set osd_pit_ang_pos = 257
set osd_rol_ang_pos = 289
set osd_battery_usage_pos = 392
set osd_disarmed_pos = 138
set osd_nheading_pos = 311
set osd_nvario_pos = 279
set osd_esc_tmp_pos = 82
set osd_esc_rpm_pos = 83
set osd_stat_max_fft = ON
set debug_mode = D_MIN
set vtx_band = 2
set vtx_channel = 7
set vtx_power = 4
set vtx_low_power_disarm = ON
set vtx_freq = 5847
set vcd_video_system = NTSC
set gyro_rpm_notch_harmonics = 0

# profile
profile 0

set vbat_pid_gain = ON

# profile
profile 1

set dyn_lpf_dterm_min_hz = 120
set dyn_lpf_dterm_max_hz = 180
set dterm_lowpass2_type = PT1
set dterm_lowpass2_hz = 180
set vbat_pid_gain = ON
set feedforward_transition = 20
set iterm_limit = 500
set throttle_boost = 7
set p_pitch = 38
set i_pitch = 85
set d_pitch = 25
set f_pitch = 96
set p_roll = 35
set i_roll = 75
set d_roll = 22
set f_roll = 85
set p_yaw = 25
set i_yaw = 130
set horizon_tilt_expert_mode = ON

# profile
profile 2

set dterm_lowpass2_hz = 200
set vbat_pid_gain = ON
set anti_gravity_threshold = 350
set anti_gravity_gain = 7000
set yaw_lowpass_hz = 100
set throttle_boost = 7
set p_pitch = 38
set i_pitch = 85
set d_pitch = 33
set f_pitch = 100
set p_roll = 35
set i_roll = 75
set d_roll = 30
set f_roll = 100
set p_yaw = 20
set i_yaw = 130
set horizon_tilt_expert_mode = ON
set d_min_pitch = 16
set d_min_roll = 16
set d_min_advance = 50

# restore original profile selection
profile 2

# rateprofile
rateprofile 0

set roll_rc_rate = 192
set pitch_rc_rate = 192
set yaw_rc_rate = 192
set roll_expo = 20
set pitch_expo = 20
set yaw_expo = 12
set roll_srate = 68
set pitch_srate = 68
set yaw_srate = 59
set tpa_rate = 30

# rateprofile
rateprofile 1

set roll_srate = 82
set pitch_srate = 82
set yaw_srate = 82
set tpa_rate = 15
set tpa_breakpoint = 1050

# rateprofile
rateprofile 2

set roll_rc_rate = 93
set pitch_rc_rate = 93
set yaw_rc_rate = 93
set roll_expo = 20
set pitch_expo = 20
set yaw_expo = 20
set roll_srate = 82
set pitch_srate = 82
set yaw_srate = 82
set throttle_limit_type = SCALE
set throttle_limit_percent = 95

# rateprofile
rateprofile 3

set roll_rc_rate = 77
set pitch_rc_rate = 77
set yaw_rc_rate = 77
set roll_expo = 30
set pitch_expo = 30
set yaw_expo = 30
set throttle_limit_type = SCALE
set throttle_limit_percent = 95

# rateprofile
rateprofile 4

# rateprofile
rateprofile 5

# restore original rateprofile selection
rateprofile 2

# save configuration
save
#```

HGLRC Omnibus F4 (438) and Skystars Omnibus F4SD (unknown)

jflyper, 2019-02-06:
May have been a side effect of #7529. Let me see if #7541 fixes it.

jflyper, 2019-02-06:
Yeah, ci#1413 does the same. Let's see...

jflyper, 2019-02-06:
Yes, #7541 brings ACC back to life. Not sure about the MAG...

etracer65, 2019-02-06:
In the `diff` you have `set mag_hardware = NONE`.

gvhlaw, 2019-02-06:
True - my FC'S don't have Magnetometers - and I apologize for not enabling
it to see if it would work.  If you would like me to try and enable the
mag, just let me know and I will and will amend the Issue I posted.  I
didn't try to enable the Mag after I checked to see if the ACC was enabled
- that was enough to spur me to bring this to your attention for prompt
remediation, and I appreciate the effort.

George

-- 

*Confidentiality Notice*:  This e-mail transmission and any accompanying
document(s), may contain confidential information which is legally
privileged under the attorney-client or attorney work product
privileges. Forwarding or sending attorney-client emails or attachments
whether directly or in an email string to persons other than the attorney
and client engaged in an attorney-client relationship may limit or destroy
the attorney-client privilege.  The information contained herein or any
attachment hereto is intended only for the use by the individual(s) or
entities named as the addressee(s) above.  If you are not the intended
recipient, or the person responsible for delivering it to the intended
recipient, you are hereby notified that any disclosure, copying,
distribution, or use of any of the information contained in this
transmission is strictly *PROHIBITED*.  Any such inadvertent disclosure of
information, which includes any metadata in emails or attachments, shall
not compromise or waive the attorney-client privilege as to this
communication or otherwise.  If you have received this transmission in
error, please notify me by telephone at (209)-956-9940 or reply e-mail and
delete this message and any attachments.  Thank you.

George V. Hartmann, Esq.
THE HARTMANN LAW FIRM
3425 Brookside Road, Suite A
Stockton, CA 95219

209.956.9940 O
209.956.9929 F

Discourage litigation. Persuade your neighbors to compromise whenever you
can. As a peacemaker the lawyer has superior opportunity of being a good
man. There will still be business enough.

Abraham Lincoln
16th president of US (1809 - 1865)

etracer65, 2019-02-06:
The MAG wouldn't show up if there's no actual hardware detected, even if you had the MAG enabled (or set to "AUTO"). So I think we should be clear that there's no actual reported problem with the MAG and this is about the ACC only (which was a problem).

gvhlaw, 2019-02-07:
We are clear, and my bad for talking about the magnetometer - apologies to all.

Szpero, 2019-02-25:
KiwiF4 v2 - missing ACC too since #1413. Last tested build #1454 (no acc).
Am I something missed in config or #1413 code is affected?

etracer65, 2019-02-25:
The problem with KIWIF4V2 is a separate problem. Fixed in https://github.com/betaflight/betaflight/pull/7660

mikeller, 2019-02-28:
@gvhlaw: Is this still a problem?

gvhlaw, 2019-02-28:
No sir - please close the issue.

George

-- 

*Confidentiality Notice*:  This e-mail transmission and any accompanying
document(s), may contain confidential information which is legally
privileged under the attorney-client or attorney work product
privileges. Forwarding or sending attorney-client emails or attachments
whether directly or in an email string to persons other than the attorney
and client engaged in an attorney-client relationship may limit or destroy
the attorney-client privilege.  The information contained herein or any
attachment hereto is intended only for the use by the individual(s) or
entities named as the addressee(s) above.  If you are not the intended
recipient, or the person responsible for delivering it to the intended
recipient, you are hereby notified that any disclosure, copying,
distribution, or use of any of the information contained in this
transmission is strictly *PROHIBITED*.  Any such inadvertent disclosure of
information, which includes any metadata in emails or attachments, shall
not compromise or waive the attorney-client privilege as to this
communication or otherwise.  If you have received this transmission in
error, please notify me by telephone at (209)-956-9940 or reply e-mail and
delete this message and any attachments.  Thank you.

George V. Hartmann, Esq.
THE HARTMANN LAW FIRM
3425 Brookside Road, Suite A
Stockton, CA 95219

209.956.9940 O
209.956.9929 F

Discourage litigation. Persuade your neighbors to compromise whenever you
can. As a peacemaker the lawyer has superior opportunity of being a good
man. There will still be business enough.

Abraham Lincoln
16th president of US (1809 - 1865)

closed 2019-03-01 by mikeller