betaflight/betaflight #7548: Motor not working using Dshot

Durete, 2019-02-07:
**Describe the bug**
Testing BF4, I found Motor 2 not working using Dshot on Beecore F4 (NOX Target).
All motors works perfect using old protocols as Multishot

**To Reproduce**
Flash latest BF4 build on Beecore F4 flight control board, configure Dshot as ESC protocol and test motors.

**Expected behavior**
All motors should work using Dshot protocol

**Flight controller configuration**

```# diff all

# version
# Betaflight / NOX (NOX1) 4.0.0 Feb  7 2019 / 11:34:41 (760300b56) MSP API: 1.41

board_name NOX
manufacturer_id 
mcu_id 002f002e3437511832363736
signature 

# reset configuration to default settings
defaults nosave

# name

# display_name

# resources

# dmaopt

# mixer

# servo

# servo mix

# feature

# beeper

# beacon

# map

# serial

# led

# color

# mode_color

# aux

# adjrange

# rxrange

# vtx

# vtxtable

# rxfail

# master
set motor_pwm_protocol = DSHOT600

# profile
profile 0

# profile
profile 1

# profile
profile 2

# restore original profile selection
profile 0

# rateprofile
rateprofile 0

# rateprofile
rateprofile 1

# rateprofile
rateprofile 2

# rateprofile
rateprofile 3

# rateprofile
rateprofile 4

# rateprofile
rateprofile 5

# restore original rateprofile selection
rateprofile 0

# save configuration
save
```

```
# resource
resource BEEPER 1 C13
resource MOTOR 1 A01
resource MOTOR 2 A07
resource MOTOR 3 B08
resource MOTOR 4 B01
resource PPM 1 B10
resource LED_STRIP 1 A00
resource SERIAL_TX 1 B06
resource SERIAL_TX 2 A02
resource SERIAL_TX 11 A02
resource SERIAL_RX 1 B07
resource SERIAL_RX 2 A03
resource SERIAL_RX 11 A02
resource INVERTER 2 C14
resource LED 1 A04
resource SPI_SCK 1 B03
resource SPI_SCK 2 B13
resource SPI_MISO 1 B04
resource SPI_MISO 2 B14
resource SPI_MOSI 1 B05
resource SPI_MOSI 2 B15
resource ESCSERIAL 1 B10
resource ADC_BATT 1 A05
resource BARO_CS 1 A09
resource FLASH_CS 1 A15
resource OSD_CS 1 A10
resource GYRO_EXTI 1 A08
resource GYRO_CS 1 B12
```

The FC is a Beecore F4 (https://www.banggood.com/BeeCore-Betaflight-F4-Flight-Controller-OSD-Integrated-4-In-1-10A-Brushless-ESC-1-2S-for-RC-Drone-p-1409682.html) using NOX target.

Using Betaflight 3.5.5 all motors works using Dshot protocol.

mikeller, 2019-02-07:
Looks like the change to the NOX target in #7264 broke this.

mikeller, 2019-02-07:
@Durete: Can you test this and report back please:

[betaflight_4.0.0_NOX.zip](https://github.com/betaflight/betaflight/files/2842538/betaflight_4.0.0_NOX.zip)

Durete, 2019-02-07:
> @Durete: Can you test this and report back please:
> 
> [betaflight_4.0.0_NOX.zip](https://github.com/betaflight/betaflight/files/2842538/betaflight_4.0.0_NOX.zip)

Tested.
Is working fine :)

Didn't fly the micro copter because is an unfinished build, but at least the four motors are working fine using Dshot.

ctzsnooze, 2019-02-07:
I had the same issue, it happened with `set dshot_burst = OFF` and `set dshot_bidir = OFF`.
Reverting the #7264 target changes fixed it for me.

dabit20, 2019-02-08:
DSHOT motor issues here too, but only when remapping motor resources.
FC Matek F405, BF4.0 #1422. Default motor order:
resource MOTOR 1 C06
resource MOTOR 2 C07
resource MOTOR 3 C08
resource MOTOR 4 C09

This works with DSHOT600.

When remapping the motors to:

resource MOTOR 1 C07
resource MOTOR 2 C09
resource MOTOR 3 C06
resource MOTOR 4 C08

motor 4 stops working, no matter the DSHOT speed (tried 300/600/1200).
Multishot does work though.

Setting `dshot_burst = OFF` and `dshot_bidir = OFF` returns all motors to functional with DSHOT600.

Durete, 2019-02-08:
Hi guys.
Unfortunately I can't continue testing this issue.
This FC is crap, I tested hover with patched version from @mikeller, and in about 20 second the Mosfets were unsolder and fall down from the board! What a crap product.
Sorry.

mikeller, 2019-02-09:
@Durete: Ouch, sorry to hear. No idea how any change to the flight controller firmware could cause the ESC to self-destruct in this way. Looks like the Beecore F4 is a very bad copy of the NOX then. :-(

dabit20, 2019-02-09:
Unless the DShot frames sent bij the FC are so damaged that the motor is continuously trying to start and stop I see no reason why firmware can overheat the ESC's to the point that FET's solder themselves loose. And even then; usually you will cook and burn a motor winding first which usually results in an ESC bursting in flames. 

Regarding the 'motor not working' issue: DShot works with BF 4.0 build #1424 on the Matek F405 with swapped motor numbers only after setting 'dshot_burst = off'.

Durete, 2019-02-09:
I'm sorry if I have not explained myself well. I never tried to say that the firmware was responsible.
I'm 100% sure the problem is the FC. It's a new FC and probably have some quality issues because there are various failure reports on banggood.

joelucid, 2019-02-14:
I've fixed an issue with dshot burst which should address the majority of the problems in this issue. I could not reproduce problems without dshot burst on the NOX target. I used the banggood NOXe for the tests.

joelucid, 2019-02-14:
Durete, could you try this hex, please?

[betaflight_4.0.0_NOX.hex.zip](https://github.com/betaflight/betaflight/files/2864634/betaflight_4.0.0_NOX.hex.zip)

Durete, 2019-02-14:
Hi @joelucid.
My Beecore F4 using Nox target was totally destroyed, so I can't test myself.
But I talked with a friend who I know have the same FC assembled and he test your hex.
He replicated the same issue reported by me till version from february 7, but He test your hex and older versions from february 11 till today, and all motors are working.
So I guess any commit from february 7 to february 11 solved this issue.
He confirmed me is working fine with latest Jenkins build and with your hex.

joelucid, 2019-02-14:
Thank you, @Durete!

closed 2019-03-23 by stale[bot]