betaflight/betaflight #7552: Tramp VTX control not working

dabit20, 2019-02-08:
Betaflight 4.0, build #1422.

VTX control through IRC Tramp protocol does not work. OSD gives weird values such as 'Band 1' and a '-' for the VTX power. Band, channel and power level cannot be changed through OSD of through the Taranis LUA script.

With older builds such as BF 4.0 #1328 VTX control works fine.

To reproduce: Flash build #1422, chip erase, load defaults, setup RX, connect IRC tramp to spare FC UART, setup IRC Tramp peripheral on that UART, and try to control the VTX.

**Flight controller configuration**
```
# diff all

# version
# Betaflight / MATEKF405 (MKF4) 4.0.0 Feb  8 2019 / 08:21:08 (2c739346c) MSP API: 1.41

board_name MATEKF405
manufacturer_id 
mcu_id 002f001d3436510737323339
signature 

# reset configuration to default settings
defaults nosave

# name

# display_name

# resources

# dmaopt

# mixer

# servo

# servo mix

# feature
feature GPS
feature LED_STRIP
feature ESC_SENSOR

# beeper

# beacon

# map

# serial
serial 0 1024 115200 57600 0 115200
serial 1 64 115200 57600 115200 115200
serial 2 16384 115200 57600 0 115200
serial 3 8192 115200 57600 0 115200
serial 4 2 115200 115200 0 115200

# led

# color

# mode_color

# aux

# adjrange

# rxrange

# vtx

# vtxtable

# rxfail

# master
set vtx_band = 5
set vtx_power = 3
set vtx_freq = 0
set vcd_video_system = NTSC
set rcdevice_init_dev_attempts = 4

# profile
profile 0

# profile
profile 1

# profile
profile 2

# restore original profile selection
profile 0

# rateprofile
rateprofile 0

# rateprofile
rateprofile 1

# rateprofile
rateprofile 2

# rateprofile
rateprofile 3

# rateprofile
rateprofile 4

# rateprofile
rateprofile 5

# restore original rateprofile selection
rateprofile 0

# save configuration
save
```
**Setup / Versions**
 - Flight controller Matek F405
 - VTX ImmersionRC Tramp

jflyper, 2019-02-08:
Welcome to the _*Beta*_flight 😉 

Please take a look at #7540 and enter your own CLI `vtxtable` commands.

dabit20, 2019-02-08:
Oh, I totally accept that things do not work from time to time when using Beta-firmware.
I just thought I would mention it in case it helps you guys. If this is expected behaviour for now, feel free to close the issue. 

I will (re)try a recent build with the vtxtable commands, and see how it goes. 

I suppose this also allows me to create my own 'bands' with custom frequencies such as the IMD-free frequency sets? Cool!

mikeller, 2019-02-08:
@dabit20:

> I suppose this also allows me to create my own 'bands' with custom frequencies such as the IMD-free frequency sets? Cool!

That's the whole purpose of this change - we cannot and don't want to be responsible for the dozens if not 100s of possible band sets that users are required to use around the globe due to legal restrictions and event / location rules.

Leaving this open for now so others getting into the same situation are more likely to find it. Please report back on how you are going with the new setup.

dabit20, 2019-02-08:
I will check and report back.

Just curious: I understand the desire to handoff all responsibility; often it is required to create a device in such a way that it is not trivial to break the law. Could become a problem for resellers in the EU for example when users can do things like setting VTX power to >25mW from the OSD.

Which makes me wonder: are you going to implement a default band set (such a mentioned in #7540) that is loaded when a 'defaults' command is issued, or are you going to always leave table setup to the user?

mikeller, 2019-02-08:
> Just curious: I understand the desire to handoff all responsibility; often it is required to create a device in such a way that it is not trivial to break the law. Could become a problem for resellers in the EU for example when users can do things like setting VTX power to >25mW from the OSD.

Yes, that is definitely the case. With this change, Betaflight takes a clear stance that manufacturers / resellers will have to step up to the plate here, and take on the responsibility to abide by the local laws of wherever they are supplying their products to, and only supply products that are limited to whatever is legal in the jurisdiction they are supplying.
The main motivation for Betaflight to go this way is that it is impossible for us to take on this responsibility - if we were to supply firmware with hard coded frequency / power limits in accordance with local laws, we'd have to supply different versions of the firmware for dozens if not 100s of different jurisdictions, completely blowing out how much effort we, as a volunteer organisation, can put into just this one aspect of our product. Another factor is that, even if we were to release firmware versions that were enforcing radio regulations in different jurisdictions, since our distribution is electronic, there is not much we can do to prevent users from downloading / using firmware that was built for a different jurisdiction, allowing them to break the local laws.

> Which makes me wonder: are you going to implement a default band set (such a mentioned in #7540) that is loaded when a 'defaults' command is issued, or are you going to always leave table setup to the user?

I think my wordy answer above pretty much implies the answer to this one: There will be no pre-defined defaults in the firmware.
We will be looking into defining a file format for 'VTX band plans', and adding the capability to configurator for users to easily load such a band plan, and write it to the flight controller, and we'll probably set up a wiki for users to share and maintain band plans for different jurisdictions and localised band plans for air fields and events - again, as a volunteer organisation we simply do not have the resources to take on the responsibility to stay on top of law changes through 100+ jurisdictions and provide 'official' band plans that are updated accordingly.

dabit20, 2019-02-09:
I think this is a good change and I completely agree with the reasoning behind it.

Regarding the original 'issue': after flashing BF 4.0 build #1424 and adding the required vtxtable entries (I used the set in #7540), Tramp control through the OSD works. I did not check the Lua script on the Taranis yet, but I expect no issues there.

mikeller, 2019-02-09:
Thanks for testing and reporting back, @dabit20!

jflyper, 2019-02-09:
@dabit20 Thanks for testing. You are the first one with the hardware to report back!

dabit20, 2019-02-10:
Right on time. After clipping a branch yesterday the quad turned itself into a fire and smoke bomb. It was not hard to find; a large plume of smoke marked the exact crash location. Hard to tell what caused it; the fire seems to have originated at the Tramp HV (which was directly powered by the lipo voltage), but with everything covered in sooth it is hard to tell. ESC is fairly OK, damage to that part is collateral. 

It is very unlikely that this is related to BF4.0 code though.

0crap, 2019-02-10:
Thx for opening this issue, I'm in the same boat.

OSD and LUA don't work and my rig is now stuck at the latest setting I used.
I looked at #7540 but don't understand how to fix this issue with that info.
I tried "set vtxtable band 1 BOSCAM_A A 5865 5845 5825 5805 5785 5765 5745 5725" on the CLI but that seems not the correct way.

Any help on how to fix this myself would be much appreciated. 👍 

[image]

As a side note, in my country it's allowed to own anything that is able to transmit with whatever power.
It's from day one up to the end user to only use such equipment in a way that it is legal.
For example I have a HAM licence which gives me the possibility to use more output power on the 5.8GHz band compared to non HAM licensed users.

jflyper, 2019-02-10:
@0crap `vtxtable` is a command by itself.

0crap, 2019-02-10:
@jflyper That is embarrassing. :-) Thanks for the heads up.
I pasted below without any "set" and it works again. I'm happy. 👍  

vtxtable bands 5
vtxtable channels 8
vtxtable band 1 BOSCAM_A A 5865 5845 5825 5805 5785 5765 5745 5725
vtxtable band 2 BOSCAM_B B 5733 5752 5771 5790 5809 5828 5847 5866
vtxtable band 3 BOSCAM_E E 5705 5685 5665 5645 5885 5905 5925 5945
vtxtable band 4 FATSHARK F 5740 5760 5780 5800 5820 5840 5860 5880
vtxtable band 5 RACEBAND R 5658 5695 5732 5769 5806 5843 5880 5917
vtxtable powerlevels 5
vtxtable powervalues 25 100 200 400 600
vtxtable powerlabels 25 100 200 400 600

0crap, 2019-02-10:
My Matek FCHUB VTX has only three power levels. 25, 200 and 500mW.

Guess I can now do:
vtxtable powerlevels 3
vtxtable powervalues 25 200 500
vtxtable powerlabels 25 200 500

jflyper, 2019-02-10:
Exactly ;)

jflyper, 2019-02-10:
> Edit: The above works partially and that is because Matek needs to update the FCHUB VTX firmware.
In the past they changed something and that is why I need to select 600 in BF to get the 500mW setting on the VTX.

`vtxtable powerlabels` defines string literals to be displayed and
`vtxtable powerlevels` defines control values sent to VTX.

In your case, why not fiddle with these to match the actual power the VTX is outputting and the value displayed match, starting with

```
vtxtable powerlabels 25 200 500
vtxtable powerlevels 25 200 600
```

0crap, 2019-02-10:
I'll give that a try, thx!

ghostface, 2019-02-10:
thank you that explains why it stopped working :)

jflyper, 2019-02-10:
@0crap Tramp protocol use direct values for RF output level request, so it should somewhere in the CMS code. I will look into it.

0crap, 2019-02-10:
Great 👍

jflyper, 2019-02-10:
@0crap What is your FC?

jflyper, 2019-02-10:
@0crap Here's the updated firmware to test.
[betaflight_4.0.0_MATEKF405-trampfix0.hex.zip](https://github.com/betaflight/betaflight/files/2849244/betaflight_4.0.0_MATEKF405-trampfix0.hex.zip)

mikeller, 2019-02-10:
@jflyper: Good point, looks like VTX tables will also solve the long standing issue of non-IRC VTX using the Tramp protocol with non-IRC power levels not working as they should. :+1:

0crap, 2019-02-11:
@jflyper Fix confirmed! 💯 

I tried for example this:

vtxtable powerlevels 3
vtxtable powervalues  25 200 600
vtxtable powerlabels LOW MID HIG

The VTX now perfectly switches between all settings, also the 500mW now works! 👍 

The only thing I can think of is that, on top of the OSD screen, the value 600 is shown.
This is obviously the "powervalues" that is send to the VTX.
In my case this is effectively the 500 mW setting.
So if I used as powerlabel 25 200 500 it still shows 600 on top of the screen.
In other words, it maybe is an idea to follow the powerlabels on top of the OSD screen as well?

Anyway, great job!

[image]

0crap, 2019-02-12:
I send Matek an e-mail to ask if they will support the "vtxtable powervalues 500" setting.
If I now use 500 it's looping trying to enable the 25mW setting. :/

Maybe they will do it to make it perfect for BF 4.0 👍

PR #7569 (2019-02-14): [VTXTABLE, TRAMP] Fix power index handling
Fixes #7552 

Matek variant still have a problem. Have to find out how they interpret and respond to 600mW request.
closed 2019-02-14 by mikeller