betaflight/betaflight #7555: support lm1881 for Osd

jack113113, 2019-02-09:
Hello betaflight group
any idea to replace Osd max7456 ? idk why a long time some my Fc burn regular 3v3 or burn chip max7456. and i worry lost video when fly longrange with Osd . 
i searching and see infor about Lm1881  and seem good Osd .small cheap 
what do you think about it ?

mikeller, 2019-02-09:
@jack113113: LM1881 is not an overlay generator, it is just a sync extractor. To use this for OSD generation, the actual overlay generation would have to be done by the MCU - this is definitely possible for faster MCUs (F7, possibly F4), but it would most likely preclude other features like 32k gyro sampling.

jack113113, 2019-02-09:
thank for answer 
So ATM if i OFF OSD or use fc no OSD , will reduce latency processing ? i thinking about external OSD .

jflyper, 2019-02-09:
@jack113113 
Some additional info.

- The original MAX7456 runs on 5V, not 3V3.
- Majority of boards with BFOSD use AT7456E or AB7456 which runs considerably cooler than the original MAX7456, because these run on 3V3.
- I suspect your board is using the original MAX7456. Check out the surface marking of the chip.
- And the 3V3 LDO is burning because overall current draw is large.

jflyper, 2019-02-09:
@jack113113 
> So ATM if i OFF OSD or use fc no OSD , will reduce latency processing ?

Very little saving.

What @mikeller mentioned is, if a main F4/F7 chip was used to generate OSD overlay with a help from LM1881/LMH1980 sync separator, then considerable amount of CPU cycles would be consumed to generate the overlay.

> i thinking about external OSD

If you have a spare UART, that's an option. However, pay attention to what chip the external OSD use. Older El cheapo ones use MAX7456 which just runs as hot as your current chip.

jack113113, 2019-02-09:
Thanks @jflyper 

seem hard to trust about 7456 when i have so much trouble with it . But unlucky on market OSD use it about 99%😥
anyhow wait new something in future .....

p/s i click wrong close topic .Sorry

closed 2019-02-09 by jflyper

jflyper, 2019-02-09:
Airbot MicroOSD V2.3 or later, or AiroMAX OSD are MWOSD capable external OSDs that use AT7456E/AB7456 chip.

mikeller, 2019-02-09:
@jack113113: I can only second @jflyper in this - buying flight controllers from quality brands instead of cheaply made clones will will help you avoid problems stemming from sub-standard hardware.