betaflight/betaflight #7557: prearm and arm on the same channel

tloki, 2019-02-09:
I do have a feature request I would find useful on Transmitters that have a small number or channels (6 or even less). This also may only be an issue isolated to my case, or just my ignorance...

I would like to use Prearm + ARM switch on the same AUX channel, in a way that utilizes mixer (arm switch on AUX channel and "prearm switch" as a mixer setting).

Let's say that my ARM (two-way switch) does have two positions encoded as:

1400 (ARMED)
1600 (DISARMED)

and my "prearm" mixer switch does the following (when engaged) to my arm values:

1350 (ARMED)
1550 (DISARMED)

Now I could hypothetically set my arm range to (in modes tab):
1300-1450 for arming
and prearm range to:
1330-1575

And I even could do it, but my quadcopter is acting very weird and I couldn't say it really does work:

This is what configurator reports (apart from 17)

when both switches are to OFF position (disarm + non-prearm):
10 (no prearm) - I suppose that one is ok, although not sure why it does not report 20 (arm switch not on)

when I push just the prearm switch (arm switch stil on DISARM position):
no flags apart from 17...

when I press first the pre-arm switch, and then arm switch (now both on):
20 (ARM_SWITCH flag - what?!)

after last situation, if I just put back prearm switch:
20 (again...)

... But all of that is very unstable, sometimes in all four combinations configurator does report only a 17...
-----------------------
Apart from the weird situation, while not connected to configurator, I have experienced the following:

The quadcopter does ARM no matter the prearm switch EXCEPT:
if I do arm it the standard way (prearm, then arm) but leave prearm on for whole flight (until disarm), it wont arm again until I toggle the prearm (I suppose it is meant to be that way, but this is the only thing that works in my situation).

Encoded values are not totally accurate, they are imaginary just to depict the situation, I am posting true values and situations in screenshots of modes tav.
---------------------
My gear:

Betaflight Configurator 10.4.0 @ Ubuntu 18.04 (non-chrome app)
CC3D F1 @ betaflight 3.5.5 (custom built with spektrum telemetry)
Spektrum 4649T receiver (on a single UART pad with telemetry)
DX6 Gen2 EU old revision (DSMX @ 11ms)
Current and Voltage ADC enabled (the latter over voltage divider I soldered on ESC)
Using MULTISHOT for ESC
0.5kHz loop frequency, 9-11% CPU utilization

additional info:

tasks:
`tasks
00 - (         SYSTEM)      9       5       2    0.5%    0.5%         4
01 - (         SYSTEM)    994       6       2    1.0%    0.6%       358
02 - (       GYRO/PID)    497     426     362   21.6%   18.4%     49072
03 - (            ACC)    994      75      51    7.9%    5.5%     13967
04 - (       ATTITUDE)    497     296     240   15.2%   12.4%     32964
05 - (             RX)     92     162     123    1.9%    1.6%      3485
06 - (         SERIAL)     99   33758       3  334.7%    0.5%      1088
08 - (BATTERY_VOLTAGE)     49      18      13    0.5%    0.5%       186
09 - (BATTERY_CURRENT)     49      23      18    0.6%    0.5%       240
10 - ( BATTERY_ALERTS)      5       8       4    0.5%    0.5%         4
11 - (      TELEMETRY)    249     220       1    5.9%    0.5%      5279
RX Check Function                  18      12                       338
Total (excluding SERIAL)                        55.6%   41.0%`

features:
`Enabled: RX_SERIAL TELEMETRY AIRMODE ANTI_GRAVITY `

diff:
`# Betaflight / CC3D (CC3D) 3.5.5 Jan 27 2019 / 10:45:33 (8892167) MSP API: 1.40

name Loki
feature -RX_PPM
feature -DYNAMIC_FILTER
feature RX_SERIAL
feature TELEMETRY
feature AIRMODE
map TAER1234
serial 2 64 115200 57600 0 115200
aux 0 0 0 1200 1550 0 0
aux 1 1 1 975 1050 0 0
aux 2 27 0 1975 2025 0 0
aux 3 36 0 1300 1750 0 0
aux 4 47 1 1300 1700 0 0
rxfail 4 s 2000
set gyro_sync_denom = 16
set min_check = 1015
set max_check = 1985
set serialrx_provider = SRXL
set rx_min_usec = 985
set rx_max_usec = 2015
set serialrx_halfduplex = ON
set blackbox_device = NONE
set min_throttle = 1035
set motor_pwm_protocol = MULTISHOT
set vbat_min_cell_voltage = 32
set current_meter = ADC
set battery_meter = ADC
set vbat_scale = 103
set vbat_multiplier = 2
profile 0

rateprofile 1`
[image]
[image]
[image]
[image]

tloki, 2019-02-09:
Well, I did achieve what I wanted by having multiple ranges for prearm switch (so it wouldn't register as "on" when arm switch is on) - and now it does work ok (but I do not understand why it wouldn't work the way I posted it...). Arming flags are still confusing to me. Leaving this issue opened so someone smarter than me could see whether there is indeed some issue in here or everything works by design...
[image]

jflyper, 2019-02-09:
I don't fully understand what you are trying to do, but prearm must be active before arm.

```
aux 0 0 0 1475 1725 0 0
aux 1 36 0 1300 1600 0 0
```

etracer65, 2019-02-09:
Stop testing in the configurator. If you have any other arming disabled reason (like 17 - MSP) then you'll also get the arming switch reason as well because it's telling you that the arming switch must be moved back to disarm before you can try arming again (once all the other reasons are resolved).

jflyper, 2019-02-09:
@etracer65 
I'm testing mode configs in the configurator ... and with MSP RX...

Reason 17 can be suppressed by swearing to motor tab that you know what you are doing. Anything else that I shouldn't use the configurator for modes testing?

etracer65, 2019-02-09:
@jflyper - I was replying to the original poster @tloki

tloki, 2019-02-09:
> Stop testing in the configurator. If you have any other arming disabled reason (like 17 - MSP) then you'll also get the arming switch reason as well because it's telling you that the arming switch [...]

Got that, but I get very weird actions from quadcopter when configured like in the first post - and my question is, why would configuration from post 2 act differently than configuration from post 1? Assume four combinations encodings of arm/prearm switches are the same as on the post no. 1 (bc they are...). 

> I don't fully understand what you are trying to do, but PREARM must be active when ARM is active to actually arm.
> 
> ```
> aux 0 0 0 1475 1725 0 0
> aux 1 36 0 1300 1600 0 0
> ```

I got that it must be active - and if you take a look in the 4 photos I posted, it is active IN ANY scenario EXCEPT when both my ARM and PREARM switches are off. - I thought that the quadcopter won't arm if prearm switch was not active prior to arming - but in my case it techincally activates at the same time as arming - so there may be my logic flaw...

Should I make a video demonstrating a problem (at least the thing I think is the problem)? English is not my native language, so there may be a language barrier trying to explain it like this... Closing this as it might not be of relevance...

closed 2019-02-09 by tloki

mikeller, 2019-02-09:
[image]