betaflight/betaflight #7562: Allow User custom names for PID profiles and display on OSD

rhys100, 2019-02-10:
Following on from https://github.com/betaflight/betaflight/pull/7516 + https://github.com/betaflight/betaflight/pull/7482

Currently there is an osd feature called show PID/RATE profile and 1,1 would be displayed as 1:1

With a few quads and different setups, this can become confusing.

It would be more user friendly to enable a custom name on each pid profile eg. 6s Test 1, 6S test 2, 4s setup and then that output enabled on OSD.

It could then join with https://github.com/betaflight/betaflight/issues/7481#event-2125181110 and say 6s profile has been loaded by voltage input, "6S test 1" PID profile now active for example.

rhys100, 2019-03-12:
@mikeller ;)

mikeller, 2019-03-13:
@rhys100: You wanted to say something?

closed 2019-03-14 by mikeller