betaflight/betaflight #7568: eachine us65 Frsky

DonaldFila, 2019-02-12:
buon giorno a tutti, ho un problema con il mio eachine us65 con ricevente frsky, bindato con D8 il quadricottero funziona, ma quando lo connetto a betaflight con radiocomando acceso Taranis qx7 continua a darmi un loop di "telemetria disponibile" "telemetria assente" e nella sezione ricevente, i comandi vanno a caso, messo così non è possibile configurare i parametri 1000 1500 2000. grazie buana giornata. 
[image]

0crap, 2019-02-12:
@DonaldFila Pretty ignorant to ask a question in Italian.
If you value the time of the devs make sure to write in English.
It doesn't have to be perfect, use Google translate if you have to.

Asizon, 2019-02-12:
Update your quads to the last 4.0 build,use Frsky_X with D16,and disable Betaflight telemetry(dont forget),for the moment doesnt suport Lua's.

DonaldFila, 2019-02-13:
> Aggiorna i quad all'ultima build 4.0, usa Frsky_X con D16 e disabilita la telemetria di Betaflight (non dimenticare), per il momento non suport di Lua, usando Lua il tuo quad può morire nell'aria.

grazie mille molto gentile, la versione è già la 4.0 fatto tutto quello che hai detto non cambia nulla. magari problema hardware. grazie ciao

DonaldFila, 2019-02-13:
> @DonaldFila Abbastanza ignorante per fare una domanda in italiano. 
> Se apprezzi il tempo degli sviluppatori, assicurati di scrivere in inglese. 
> Non deve essere perfetto, usa Google translate se devi.

La domanda di una persona, fatta in qualsiasi lingua, non penso sia stupida, tante volte però stupida è la risposta. il tempo per rompere le scatole e non dare una soluzione l'ha trovato . che le piaccia o meno continui ad usare google traslate buona giornata.

mikeller, 2019-02-13:
@DonaldFila: The language used in here is English, for the benefit of others seeking help being able to read and participate in existing threads. Please listen to what others are telling you.
Removing offending posts.

closed 2019-02-13 by mikeller