betaflight/betaflight #7572: IRC TRAMP TELEMETRY NOT WORKING

Barux, 2019-02-13:
**Tramp telemetry not working on latest Development beta 4.0. On LUA it says "saved" but actually doesnt change the channel.**

nikybiasion, 2019-02-13:
It's not an issue, please don't spam

Asizon, 2019-02-13:
You need to use "vtxtable" like this #7552

Barux, 2019-02-13:
Now it's working. Many thanks.

So from now to official release of Beta 4.0 we should copy and paste the vtxtable to have the whole channels/powers working?

vtxtable bands 5
vtxtable channels 8
vtxtable band 1 BOSCAM_A A 5865 5845 5825 5805 5785 5765 5745 5725
vtxtable band 2 BOSCAM_B B 5733 5752 5771 5790 5809 5828 5847 5866
vtxtable band 3 BOSCAM_E E 5705 5685 5665 5645 5885 5905 5925 5945
vtxtable band 4 FATSHARK F 5740 5760 5780 5800 5820 5840 5860 5880
vtxtable band 5 RACEBAND R 5658 5695 5732 5769 5806 5843 5880 5917
vtxtable powerlevels 5
vtxtable powervalues 25 100 200 400 600
vtxtable powerlabels 25 100 200 400 600

jflyper, 2019-02-13:
@Barux 
> So from now to official release of Beta 4.0 we should copy and paste the vtxtable to have the whole channels/powers working?

Never.

You should enter `vtxtable` commands based on respective regulations of area you are flying.

jflyper, 2019-02-13:
Duplicate of #7552. Closing and locking.

closed 2019-02-13 by jflyper