betaflight/betaflight #7575: F3 board freezes

0crap, 2019-02-14:
Maybe not a valid used case, if so please close this issue.

I baked myself a hex file from the latest master, with below settings, to fit into an OMNIBUS F3 FC.

#define USE_LED_STRIP
#define USE_LED_STRIP_STATUS_MODE
#define USE_SMART_FEEDFORWARD
#define USE_BOARD_INFO
#define USE_EXTENDED_CMS_MENUS
#define USE_ITERM_RELAX
#define USE_RC_SMOOTHING_FILTER
#define USE_TPA_MODE
#define USE_THROTTLE_BOOST
#define USE_DYN_LPF
#define USE_D_MIN
//#define USE_ACRO_TRAINER
//#define USE_BLACKBOX
#define USE_RESOURCE_MGMT
#define USE_RUNAWAY_TAKEOFF     // Runaway Takeoff Prevention (anti-taz)
//#define USE_SERVOS
#define USE_TELEMETRY
#define USE_TELEMETRY_FRSKY_HUB
#define USE_TELEMETRY_SMARTPORT

This works fine and the quad flies perfect,
However if I go into the BF OSD and select MAIN->FEATURES->VTX TR the FC freezes,
There is no VTX connected to the FC, it was just me going thru all the menus and got stuck.
Only way out is unplug the LiPo.

Because there is some work done on the Tramp VTX code, I would like to point this out just in case.
I think a complete freeze of the FC should never happen.

[diff all](https://pastebin.com/raw/yfDDjQqz)

jflyper, 2019-02-15:
It must be un-conditionalized `USE_VTX_TABLE` code that F3 is executing.
I will look into this.

jflyper, 2019-02-15:
I haven't had a chance to look into this yet ... This doesn't happen with 3.5.x, right?

0crap, 2019-02-15:
That is correct, flashed 3.5.6 # 112 and no OSD freeze happens.

jflyper, 2019-02-18:
@0crap I think I've nailed it.

Can you try this?

[betaflight_4.0.0_OMNIBUS-tramp_hardfault_fix_0_69f6088.hex.zip](https://github.com/betaflight/betaflight/files/2875631/betaflight_4.0.0_OMNIBUS-tramp_hardfault_fix_0_69f6088.hex.zip)

0crap, 2019-02-18:
Flashed it and no more freezes!
What I see is that it now will not let me go into the VTX TR menu.
Which in my case is good because I don't have any VTX connected to the FC.

Maybe it's needed that someone, with an F3, can test the VTX TR functions that actually has a VTX connected to the FC?

jflyper, 2019-02-18:
Should turn out to be okay, as the fix was relatively simple; it just goes back if device isn't configured.

0crap, 2019-02-18:
For consistency sake you might also do that for the "VTX SA" menu option?

PR #7605 (2019-02-19): [CMS, TRAMP] One level back if Tramp is not configured
Fixes #7575

Entering VTX_TR menu when tramp device is not initialized or configured, `vtxDevice` returns `NULL` and causes hard fault at some point in subsequent menu processing.

This PR fixes it by jumping back to where `trampCmsOnEnter` was called (features menu).

A drawback of this handling is that the transition is not visible, and the screen stays at features menu.
This might be confusing for users trying to make Tramp work. Before VTX TABLE change, VTX_TR menu screen presented default strings for non-existent/non-online device.

T [... cut]
closed 2019-02-19 by mikeller