betaflight/betaflight #7579: Amperage and Battery Voltage not working

TheIsotopes, 2019-02-15:
**Describe the bug**
since `build #1435` the Amperage and Battery Voltage not working anymore 

**To Reproduce**
first signs in `build #1434` ... on `build #1435` and higher it gets worst
you can see it in bbf file (see the screenshots at the bottom)

also you can reproduce this in bfc on motors tab:
- plug in battery, the voltage and amperage are drawing
- enable motor control and use the master slider
- motors are turning but voltage and amperage stays at last seeing values after battery pluged in
### I could only find this on my F7 board on my F4 board everything is fine

**Flight controller configuration**
```
# Betaflight / MATEKF722SE (MF7S) 4.0.0 Feb 15 2019 / 10:03:42 (b9375d697) MSP API: 1.41

# name
name Mark1

# feature
feature -TELEMETRY
feature -OSD
feature LED_STRIP

# beeper
beeper -BAT_CRIT_LOW
beeper -BAT_LOW

# map
map TAER1234

# led
led 0 9,2::AW:0
led 1 9,3::AW:0
led 2 9,4::AW:0
led 3 9,5::AW:0
led 4 6,5::AW:0
led 5 6,4::AW:0
led 6 6,3::AW:0
led 7 6,2::AW:0

# mode_color
mode_color 6 0 3
mode_color 6 1 6

# aux
aux 0 0 0 1600 2000 0 0
aux 1 13 1 1000 1200 0 0
aux 2 26 0 1800 2000 0 0
aux 3 31 0 1200 1400 0 0
aux 4 36 1 1800 2000 0 0

# adjrange
adjrange 0 0 2 1400 2000 30 2 0 0

# master
set acc_hardware = NONE
set mag_hardware = NONE
set baro_hardware = NONE
set serialrx_provider = SUMD
set blackbox_device = SPIFLASH
set motor_pwm_protocol = DSHOT600
set pid_process_denom = 1
set ledstrip_beacon_period_ms = 350
set ledstrip_beacon_percent = 15

# profile
set feedforward_transition = 50
set iterm_relax_type = GYRO

# rateprofile
set roll_rc_rate = 135
set pitch_rc_rate = 135
set yaw_rc_rate = 120
set roll_expo = 15
set pitch_expo = 15
set roll_srate = 75
set pitch_srate = 75
```
**Setup / Versions**
 [Matek F722-Mini](http://www.mateksys.com/?portfolio=f722-mini) mounted on [FCHUB-A5](http://www.mateksys.com/?portfolio=fchub-a5)

**Additional context**
`build # 1433` no signs yet ... all ok
[image]

`build # 1434` it's starting to look ... the steps are getting wider
[image]

`build # 1435` from here it is worst
[image]

klutvott123, 2019-02-15:
@TheIsotopes Same here. This issue has been coming and going with different builds for some time now. https://github.com/betaflight/betaflight/issues/7359 describes similar behaviour. Seems to only affect F7 boards(matekf722 std in my case). My ff fortini f4 works perfectly with all builds. 

I had the issue with build 1434 too, but managed to get it working again. Be sure to do a full chip erase when flashing the firmware or ``defaults`` in cli. 
The problem seems to appear after pasting the diff, so i started pasting only parts of my diff, and testing between each change. First i did RX settings just to get up and running, before i started pasting from the top of my diff, saving and testing in between. 

What i found in my case was that 
```
feature -TELEMETRY
```
and/or
```
set acc_hardware = NONE
set mag_hardware = NONE
set baro_hardware = NONE
```
made it stop working. Unstable voltage/current readings, and the values would stop updating when i turned off my leds through AUX4(previously i experienced that the updates stopped when arming).

Then i compiled the configurator from the latest master, did ``defaults`` in cli and pasted my diff. 
Works perfectly now. Stable readings and no more freezing. 

Can you try setting the settings mentioned above back to default and see if that does anything?

TheIsotopes, 2019-02-15:
@klutvott123 yep ... you're right
- `set acc_hardware = NONE` calls this issue 
- after default `set acc_hardware = AUTO` -> Amperage and Battery Voltage is working again

klutvott123, 2019-02-15:
@TheIsotopes Thanks for testing and confirming. Did you notice any strange motor behaviour? Oscillations or twitches? I suspect this might be affecting other things too.

TheIsotopes, 2019-02-16:
@fujin @mikeller is there went wrong after cleaning up from #7567

klutvott123, 2019-02-16:
I doubt that it's related. First time i experienced this issue was on build 1333(probably goes further back too). Everytime it pops up, i can't see anything obvious in the commits that triggered the build that would make it behave like this. This makes me think that this goes deeper, and the voltage/current behaviour we're seeing is just one of the symptoms. Perhaps related  to https://github.com/betaflight/betaflight/issues/7413, where ram usage is causing issues on F7 boards? No obvious issues found in the triggering commits there either.

TheIsotopes, 2019-02-18:
I have now tested the bf 3.5.6 ... there i could not determine the issue
@klutvott123 can you confirm this

klutvott123, 2019-02-18:
@TheIsotopes I can confirm. I have not seen this issue with any of the 3.5 versions.

TheIsotopes, 2019-02-18:
with the current `build #1445` it seems to work again ... no idea why?

klutvott123, 2019-02-18:
I tried 1445 but it still behaved the same for me. BUT, I just flashed build 1443 from the configurator. Same issue. Then i compiled it locally, flashed, pasted diff(including the settings mentioned in my post further up) and now it works. My local repository is up to date with 1443 so it should be the same.

klutvott123, 2019-02-18:
Tried the same with 1445. 
Flashed it from configurator. Issue still there.
Compiled locally. Works.

klutvott123, 2019-02-20:
Just a little follow up to my above posts. I should have mentioned that i compile in cygwin. Tried the same with build 1448. Docker/jenkins builds both have the problem. When compiling the same code in cygwin, the output is a little different but it works. Maybe someone can make sense of this.

Docker build 1448:
```
Memory region         Used Size  Region Size  %age Used
        ITCM_RAM:       14168 B        16 KB     86.47%
      ITCM_FLASH:          0 GB        16 KB      0.00%
ITCM_FLASH_CONFIG:          0 GB        16 KB      0.00%
     ITCM_FLASH1:          0 GB       480 KB      0.00%
      AXIM_FLASH:       14904 B        16 KB     90.97%
AXIM_FLASH_CONFIG:          0 GB        16 KB      0.00%
     AXIM_FLASH1:      379024 B       480 KB     77.11%
        DTCM_RAM:       17988 B        64 KB     27.45%
           SRAM1:       50904 B       176 KB     28.24%
           SRAM2:          0 GB        16 KB      0.00%
       MEMORY_B1:          0 GB         0 GB      -nan%
   text    data     bss     dec     hex filename
 381280   12648   56244  450172   6de7c ./obj/main/betaflight_MATEKF722.elf
```
Jenkins build 1448:
```
Memory region         Used Size  Region Size  %age Used
        ITCM_RAM:       14168 B        16 KB     86.47%
      ITCM_FLASH:          0 GB        16 KB      0.00%
ITCM_FLASH_CONFIG:          0 GB        16 KB      0.00%
     ITCM_FLASH1:          0 GB       480 KB      0.00%
      AXIM_FLASH:       14904 B        16 KB     90.97%
AXIM_FLASH_CONFIG:          0 GB        16 KB      0.00%
     AXIM_FLASH1:      379032 B       480 KB     77.11%
        DTCM_RAM:       17988 B        64 KB     27.45%
           SRAM1:       50904 B       176 KB     28.24%
           SRAM2:          0 GB        16 KB      0.00%
       MEMORY_B1:          0 GB         0 GB      -nan%
   text	   data	    bss	    dec	    hex	filename
 381288	  12648	  56244	 450180	  6de84	./obj/main/betaflight_MATEKF722.elf
```

Cygwin build 1448: 
```
Memory region         Used Size  Region Size  %age Used
        ITCM_RAM:       14176 B        16 KB     86.52%
      ITCM_FLASH:          0 GB        16 KB      0.00%
ITCM_FLASH_CONFIG:          0 GB        16 KB      0.00%
     ITCM_FLASH1:          0 GB       480 KB      0.00%
      AXIM_FLASH:       14910 B        16 KB     91.00%
AXIM_FLASH_CONFIG:          0 GB        16 KB      0.00%
     AXIM_FLASH1:      378816 B       480 KB     77.07%
        DTCM_RAM:       17988 B        64 KB     27.45%
           SRAM1:       50912 B       176 KB     28.25%
           SRAM2:          0 GB        16 KB      0.00%
       MEMORY_B1:          0 GB         0 GB     -1.#J%
   text    data     bss     dec     hex filename
 381074   12652   56244  449970   6ddb2 ./obj/main/betaflight_MATEKF722.elf
```

TheIsotopes, 2019-02-20:
hmm, it's strange ... I've tested `jenkins build #1450` and it makes problems
now i compiled last master like `build #1450` on bash on windows and cygwin and both have no problems.
just like you @klutvott123

klutvott123, 2019-02-20:
@TheIsotopes Thanks for testing and confirming this. I tried it with 1450 too, and got the same result. At least we have a clue now.

TheIsotopes, 2019-02-20:
here again the size comparison of the three files ... I had previously forgotten ... As we can see, the size of the jenkins build differs significantly from the other two builds ... as if something is missing in the jenkins build

Jenkins build 1450
```
Memory region         Used Size  Region Size  %age Used
        ITCM_RAM:       14152 B        16 KB     86.38%
      ITCM_FLASH:          0 GB        16 KB      0.00%
ITCM_FLASH_CONFIG:          0 GB        16 KB      0.00%
     ITCM_FLASH1:          0 GB       480 KB      0.00%
      AXIM_FLASH:       15028 B        16 KB     91.72%
AXIM_FLASH_CONFIG:          0 GB        16 KB      0.00%
     AXIM_FLASH1:      386496 B       480 KB     78.63%
        DTCM_RAM:       17988 B        64 KB     27.45%
           SRAM1:       64216 B       176 KB     35.63%
           SRAM2:          0 GB        16 KB      0.00%
       MEMORY_B1:          0 GB         0 GB      -nan%
   text	   data	    bss	    dec	    hex	filename

 388772	  12752	  69452	 470976	  72fc0	./obj/main/betaflight_MATEKF722SE.elf
```
Cygwin build 1450
```
Memory region         Used Size  Region Size  %age Used
        ITCM_RAM:       14464 B        16 KB     88.28%
      ITCM_FLASH:          0 GB        16 KB      0.00%
ITCM_FLASH_CONFIG:          0 GB        16 KB      0.00%
     ITCM_FLASH1:          0 GB       480 KB      0.00%
      AXIM_FLASH:       15024 B        16 KB     91.70%
AXIM_FLASH_CONFIG:          0 GB        16 KB      0.00%
     AXIM_FLASH1:      387168 B       480 KB     78.77%
        DTCM_RAM:       17988 B        64 KB     27.45%
           SRAM1:       64252 B       176 KB     35.65%
           SRAM2:          0 GB        16 KB      0.00%
       MEMORY_B1:          0 GB         0 GB     -1.#J%
   text    data     bss     dec     hex filename
 389444   12748   69488  471680   73280 ./obj/main/betaflight_MATEKF722SE.elf
```
bash (ubuntu on windows) build 1450
```
Memory region         Used Size  Region Size  %age Used
        ITCM_RAM:       14472 B        16 KB     88.33%
      ITCM_FLASH:          0 GB        16 KB      0.00%
ITCM_FLASH_CONFIG:          0 GB        16 KB      0.00%
     ITCM_FLASH1:          0 GB       480 KB      0.00%
      AXIM_FLASH:       15024 B        16 KB     91.70%
AXIM_FLASH_CONFIG:          0 GB        16 KB      0.00%
     AXIM_FLASH1:      387200 B       480 KB     78.78%
        DTCM_RAM:       17988 B        64 KB     27.45%
           SRAM1:       64244 B       176 KB     35.65%
           SRAM2:          0 GB        16 KB      0.00%
       MEMORY_B1:          0 GB         0 GB      -nan%
   text    data     bss     dec     hex filename
 389476   12748   69480  471704   73298 ./obj/main/betaflight_MATEKF722SE.elf
```

TheIsotopes, 2019-02-20:
the same size comparison I have now made with a f4 board ... as we can see the sizes differ minor ... where the builds from jenkins and bash are even the same size

Jenkins build 1450
```
Memory region         Used Size  Region Size  %age Used
           FLASH:       14632 B        16 KB     89.31%
    FLASH_CONFIG:          0 GB        16 KB      0.00%
          FLASH1:      382232 B       992 KB     37.63%
   SYSTEM_MEMORY:          0 GB        29 KB      0.00%
             RAM:       80444 B       128 KB     61.37%
             CCM:       10012 B        64 KB     15.28%
     BACKUP_SRAM:          0 GB         4 KB      0.00%
       MEMORY_B1:          0 GB         0 GB      -nan%
   text	   data	    bss	    dec	    hex	filename
 384400	  12464	  77600	 474464	  73d60	./obj/main/betaflight_MATEKF405.elf
```
Cygwin build 1450
```
Memory region         Used Size  Region Size  %age Used
           FLASH:       14632 B        16 KB     89.31%
    FLASH_CONFIG:          0 GB        16 KB      0.00%
          FLASH1:      382224 B       992 KB     37.63%
   SYSTEM_MEMORY:          0 GB        29 KB      0.00%
             RAM:       80444 B       128 KB     61.37%
             CCM:       10012 B        64 KB     15.28%
     BACKUP_SRAM:          0 GB         4 KB      0.00%
       MEMORY_B1:          0 GB         0 GB     -1.#J%
   text    data     bss     dec     hex filename
 384392   12464   77600  474456   73d58 ./obj/main/betaflight_MATEKF405.elf
```
bash build 1450
```
Memory region         Used Size  Region Size  %age Used
           FLASH:       14632 B        16 KB     89.31%
    FLASH_CONFIG:          0 GB        16 KB      0.00%
          FLASH1:      382232 B       992 KB     37.63%
   SYSTEM_MEMORY:          0 GB        29 KB      0.00%
             RAM:       80444 B       128 KB     61.37%
             CCM:       10012 B        64 KB     15.28%
     BACKUP_SRAM:          0 GB         4 KB      0.00%
       MEMORY_B1:          0 GB         0 GB      -nan%
   text    data     bss     dec     hex filename
 384400   12464   77600  474464   73d60 ./obj/main/betaflight_MATEKF405.elf
```

klutvott123, 2019-02-20:
@TheIsotopes Yeah i can confirm. My findings are the same with AIKONF4 and FF_FORTINIF4. No problems with those. I tried compiling 3.5.6 for the MATEKF722, and both docker and cygwin build are practically the same. Something happened between 3.5 and 4.0.0

mikeller, 2019-02-20:
Just a little comment re firmware size comparisons: There will always be a slight (a couple of bytes flash) variation in size between different builds, even when built on the same sytem. The reason for this is that we add a build timestamp as a string (shown in CLI `version`).

klutvott123, 2019-02-20:
@mikeller That makes sense, but in the comparisons between builds above the difference is more than just a couple of bytes. 6/700 Bytes in @TheIsotopes post. The thing is that the jenkins/docker builds are not working as expected while the cygwin builds of the same code are working perfectly. I have done this test with quite a few of the latest builds and the results are consistent. 
Do you have any idea about what might be going on here?

mikeller, 2019-02-20:
@klutvott123:

> but in the comparisons between builds above the difference is more than just a couple of bytes.

I saw that. Just letting you know that the small differences are expected to keep you from chasing up a red herring.

I don't have an idea of what is going on there, except that you should double check to make sure you are using exactly the same version of the ARM SDK on both sides.

And thank you for looking into this, keep us posted.

klutvott123, 2019-02-20:
@mikeller I understand. Thanks 🙂 

Using cygwin i installed the sdk in the tools folder by ``make arm_sdk_install``. ``make arm_sdk_version`` tells me that the version is 7.3.1. This https://travis-ci.com/betaflight/betaflight/jobs/179120135 also shows that it was built with 7.3.1. I assume when compiling with docker it will use the sdk found in the tools folder? I find docker a little confusing. 

Can't promise anything, but at least we can gather some clues. Since there are seemingly working builds inbetween those that don't work, and the behaviour can vary depending on which settings the user changes, this one might be hard to track down.

mikeller, 2019-02-20:
@klutvott123: Docker is a bit different, as in it uses the SDK that is pre-installed in the container. This is to make docker usable to users on platforms that do not have the tooling installed to run `make arm-sdk-install`. So docker should be excluded from these investigations.

klutvott123, 2019-02-20:
Thanks for explaining. Then i would assume it's using 7.3.1 too since it's not complaining. How can i make it tell me which sdk version is installed in the container?

mikeller, 2019-02-21:
Not necessarily. The docker build command explicitly the location for the SDK to be used with `ARM_SDK_DIR`, disabling the version check - this way it is possible to build branches at different levels without having to change the toolkit - the idea of the image is to give users the simplest possible way to build.

Check https://github.com/betaflight/docker-betaflight-build/blob/master/Dockerfile to see what the docker build does behind the scenes.

klutvott123, 2019-02-22:
I ran a bash shell in the docker container and checked the SDK version. It's 7.3.1. 
I'm currently testing this on linux. The output i get is exactly the same as the jenkins build. Tried with several different SDK versions and all builds have the same issue. Maybe there's some difference between the linux and windows SDKs. When compiling using cygwin or ubuntu on windows(@TheIsotopes I assume it's the windows store app?), the SDK used would be the windows version. 

Then i tried something else. Still on linux. I commented this out:
https://github.com/betaflight/betaflight/blob/06d9f54f7767b38341434bc50d08ec5013f9f989/src/main/target/common_pre.h#L76

This puts ``FAST_CODE`` in flash instead. Produces a working build. Voltage and current readings are super stable and doesn't freeze anymore.

TheIsotopes, 2019-02-22:
@klutvott123 only cygwin uses the windows sdk
ubuntu on windows uses linux sdk ... both have the same sdk version 7.3.1

klutvott123, 2019-02-22:
@TheIsotopes I see. Thanks for clarifying. I just tried it in ubuntu on windows and it gave me the exact same non working build.

mikeller, 2019-02-23:
@klutvott123:

> Then i tried something else. Still on linux. I commented this out:
> #define USE_ITCM_RAM 
> This puts FAST_CODE in flash instead. Produces a working build.

Good find! Looks like we've stuck something into ITCM RAM that should not be (not sure if ADC access works from ITCM RAM, or if it's something else that's tripping us up.

klutvott123, 2019-02-23:
@mikeller Maybe. Doesn't look like it's much difference between master and maintenance branch when it comes to FAST_CODE. There's a few new pwm functions in master, but they were added after i first encountered this problem in build 1333. Good news is that I finally managed to produce a build with the same issue in cygwin. Build 1458 has the issue even worse when compiled using cygwin. Again i tried commenting out ``#define USE_ITCM_RAM`` and it magically starts working again.

klutvott123, 2019-02-24:
Looks like this goes pretty far back. Reproduced on 1150, 1149, 1140, 1060 and 950. Just trying every 50th build or so and a few in between. 

**Unrelated**
Looks like #5322 is causing the voltage to read 0.2V higher on F7 boards. On build 678 it was fine. On 679 and up it's not.

mikeller, 2019-02-25:
@klutvott123: Have you found a good revision? If so, `git bisect` (https://git-scm.com/docs/git-bisect) will help you find the culprit changeset in log2 tests.

klutvott123, 2019-02-25:
@mikeller I find a lot of seemingly working builds between the non working ones. I see it in build 790, not in 795 and it's back in 800. Right now it looks like build 780(#5729) are showing symptoms of it. Voltage and current updates freezes for short periods when i flip a switch to turn of the leds through aux4, which is consistent with the behaviour observed so far. Happens in 781, 782, 783, 784 and 785 too. I tried reverting the changes in #5729 in master, but didn't make any difference. I have been trying different builds in steps of 5 and 10 from build 600 to 780 and haven't been able to reproduce the issue. I'm starting to suspect #5674 which is build 774. That's a lot of commits. Too much optimization maybe? But i can't say for sure.

klutvott123, 2019-02-25:
Ok. Might be on to something. I discovered that in #5674 the D-cache was enabled for F7 mcus. I tried disabling it in build 1458 and it started working again. By placing the adc DMA buffer in DTCM RAM instead, I got it working with the D-cache enabled. DTCM RAM is uncached. 
[image]

 ``FAST_RAM_ZERO_INIT`` in front of this one:
https://github.com/betaflight/betaflight/blob/b776bf3a9e9a8aabcf8c04e990d88757f17e231b/src/main/drivers/adc.c#L42
It makes sense since this has already been done other places in the code. This one must have been missed.
@mikeller Any thoughts on this?

mikeller, 2019-02-25:
Not sure if DTCM RAM is the right place for this to go, but using cached memory for DMA definitely won't work. @jflyper, @DieHertz thoughts?

hydra, 2019-02-27:
a couple of potential solutions:
1) invalidate/clear ADC cache memory after reading from it.
2) use the MPU to configure a region so it's not cached, configure linker script to place these structures in that region.  region boundaries and sizes are dictated by MPU.  memory location must be DMA accessible.

it sounds like in certain circumstances the cache doesn't contain the adc data, so the CPU access the memory directly, this this can be caused by ordering of instructions and varying workload of the CPU, which is why it may seem somewhat random when it works and when it doesn't and what causes it to stop working.

klutvott123, 2019-02-27:
@hydra Thanks for your input. I have been reading a little about this now, and what you're suggesting makes sense. The same is suggested in https://www.st.com/resource/en/application_note/dm00272913.pdf.
One of the reasons why i suggested placing the adc dma buffer in DTCM RAM is that it seems to be the approach used for other dma buffers. For example the led strip also used to have similar problems which was fixed by #6131.

ledvinap, 2019-02-27:
As for ADC, it would be probably best to reimplement it to use ISR instead of DMA. ADC read frequency is quite low, wasting DMA on it is overkill anyway...

mikeller, 2019-02-27:
Agreed with @ledvinap that moving to ISR for ADC is probably the best solution long term. However, considering that we are only days away from the feature freeze for the next release, I think that for the 4.0 release it would be better if we considered a solution that fixed the problem at hand with less risk of side effects.

@hydra: Do you have any pointers to how to specifically invalidate regions of memory? I couldn't find anything.

hydra, 2019-02-27:
check the cortex m programmers manual for rhe cpu in question. see section on caching.

also in the cubemx packages there are examples of data cache cleaning/invalidation.

not at my desk right now so no links, sorry.

klutvott123, 2019-02-27:
There is this one:
https://github.com/betaflight/betaflight/blob/2f1f07d186cd4023a4b7486e6fcf9f6237e5b847/src/main/drivers/dma.h#L46
which has been commented out a long time ago. Takes the address and size of the DMA buffer, cleans and invalidates. Don't know if it's the best way. The application note i linked to above states: 

> Always better to use non-cacheable regions for DMA buffers. The software can use the MPU to set up a
non-cacheable memory block to use as a shared memory between the CPU and DMA.

mikeller, 2019-02-27:
@klutvott123: Good find!

Reading the blame for this I remembered what went down way back when: We experimentally enabled the Dcache on F7, ran into problems, and then re-enabled it at a later point after fixing the problems. According to https://github.com/betaflight/betaflight/commit/1102df5d68ed48a6968658aa102e8c54786f9704, cache invalidation for DMA was removed when we disabled the Dcache, but never re-enabled (presumably because we re-enabled Dcache after we'd moved more or less all DMA access into FAST_RAM by that time).

Can I get you to test if enabling this again will solve the ADC problem?

hydra, 2019-02-27:
Here you go:

F722 programming manual - https://www.st.com/content/ccc/resource/technical/document/programming_manual/group0/78/47/33/dd/30/37/4c/66/DM00237416/files/DM00237416.pdf/jcr:content/translations/en.DM00237416.pdf

See:
* 4.6 Memory protection unit
* 4.8.4         Cortex®-M7 cache maintenance operations using CMSIS

CubeF7 code with examples:

https://www.st.com/content/st_com/en/extended-query.html?querycriteria=productId=LN1897$$associatedTo=LN1903

klutvott123, 2019-02-27:
@mikeller Looks to me like the code was incomplete, and that is what caused problems. From what i can tell, the adc cache was cleaned/invalidated only once in ``adcInit()``. I can't see that this:

> If the software is using cacheable memory regions for the DMA source/or destination buffers. The software
must trigger a cache clean before starting a DMA operation to ensure that all the data are committed to the
subsystem memory. After the DMA transfer complete, when reading the data from the peripheral, the
software must perform a cache invalidate before reading the DMA updated memory region.

was ever done. I just uncommented the clean/invalidate macro in the places it's used for adc, and it doesn't do anything.

mikeller, 2019-02-27:
@klutvott123: Looks like we'll have to add additional `HAL_CLEANINVALIDATECACHE()` calls after the ADC output has been read.

hydra, 2019-02-28:
@mikeller I thought as much.  from the BB logs posted above you say see when the ADC data stayed in the cache.  the one with the long steps is a telling indication.

klutvott123, 2019-02-28:
@mikeller Looks like it, yes. AN4839 says that we only have to invalidate the cache before reading from the DMA buffer. Or after like suggested by @hydra.

Looks like this is the function used to read from ``adcValues``:
https://github.com/betaflight/betaflight/blob/b776bf3a9e9a8aabcf8c04e990d88757f17e231b/src/main/drivers/adc.c#L85
I defined a new macro:
```
#define HAL_INVALIDATECACHE(addr, size) (SCB_InvalidateDCache_by_Addr((uint32_t*)((uint32_t)addr & ~0x1f), ((uint32_t)(addr + size + 0x1f) & ~0x1f) - ((uint32_t)addr & ~0x1f)))
```
and called it at the top of ``adcGetChannel()`` before anything is read from ``adcValues``, and that seems to work too. 

One thing that caught my eye is that the size argument passed to ``HAL_CLEANINVALIDATECACHE``  in the past was the number of configured channels. But 
https://github.com/betaflight/betaflight/blob/2f1f07d186cd4023a4b7486e6fcf9f6237e5b847/lib/main/CMSIS/Core/Include/core_cm7.h#L2502-L2525
states that it wants the size of the memory block in bytes. So ``sizeof(adcValues)``?
This function also says that the cache line is fixed at 32 bytes. From what i understand this means that we should use at least 32 bytes or multiples of 32 bytes for ``adcValues`` to avoid cleaning and invalidating something else too. What i have been reading about 32-byte alignment also makes sense when looking at this function. Is alignment what's going on in ``HAL_INVALIDATECACHE``(and is it correct?)?

``FAST_RAM`` solution just becomes more and more attractive the more i dig into this 🙂

mikeller, 2019-02-28:
@klutvott123:

> @mikeller Looks like it, yes. AN4839 says that we only have to invalidate the cache before reading from the DMA buffer. Or after like suggested by @hydra.

Before or after should not matter - _in between_ reads is what is needed, to prevent cached values from being read if the DMA has updated the memory underneath.

> I defined a new macro:
#define HAL_INVALIDATECACHE(addr, size) (SCB_InvalidateDCache_by_Addr((uint32_t*)((uint32_t)addr & ~0x1f), ((uint32_t)(addr + size + 0x1f) & ~0x1f) - ((uint32_t)addr & ~0x1f)))
and called it at the top of adcGetChannel() before anything is read from adcValues, and that seems to work too.

Looks like a perfectly workable and suitable solution to me.

> states that it wants the size of the memory block in bytes. So sizeof(adcValues)?
This function also says that the cache line is fixed at 32 bytes. From what i understand this means that we should use at least 32 bytes or multiples of 32 bytes for adcValues to avoid cleaning and invalidating something else too. What i have been reading about 32-byte alignment also makes sense when looking at this function. Is alignment what's going on in HAL_CLEANINVALIDATECACHE(and is it correct?)?

That is correct. Which means that `SCB_InvalidateDCache_by_Addr()` takes care of the 32 byte alignment by itself, and all we have to do is to supply the number of bytes we want the cache invalidated for. The cache will then be cleared for the next higher multiple of 32 bytes (which is the reason why us supplying the wrong number to start with did not matter).

> FAST_RAM solution just becomes more and more attractive the more i dig into this.

I disagree. Using ￼FAST_RAM is using the wrong tool, in a way that is not obvious, and relying on a side effect to fix this. Making matters worse, `FAST_RAM` is a limited resource, and we should not use it for things that do not actually benefit from what it was designed for.
We should go for the working solution that you tested above. Do you want to open a pull request for it?

DieHertz, 2019-02-28:
DTCM-RAM is a limited resource as much as ordinary SRAM is. There's plenty of it, and one of the purposes of DTCM on F7 is hosting DMA buffers.

klutvott123, 2019-03-01:
@mikeller Thanks for replying to all that, and point taken on the ``FAST_RAM`` use, even if i still like that solution for it's simplicity and consistency with the rest of the code. I can make a PR for the cache maintenance solution above as soon as I'm back at my big computer. 🙂 

@DieHertz True. 64KB on the F722. Currently using 24.5% on MATEKF722.

I'm just glad we can now have this fixed in time for BF 4 release.

mikeller, 2019-03-01:
@DieHertz: True. I somehow thought DTCM RAM was the same 16 kB that ITCM RAM is.

@klutvott123: Considering there is less contention for DTCM RAM than I thought, I am fine with either way. All in all we should look into rewriting this to stop using DMA after 4.0, so we can free up that DMA channel - so all this needs to do is tide us over for 4.0.

klutvott123, 2019-03-01:
@mikeller Sounds good. I'll just go ahead and put it in DTCM RAM then. Will be less code to clean up when the time comes to move this over to using interrupts too.

PR #7687 (2019-03-02): Place ADC DMA buffer in DTCM RAM
Places adcValues in non cached DTCM RAM to avoid problems when D-cache is enabled for F7 mcus.
Should fix #7579.
closed 2019-03-02 by mikeller