betaflight/betaflight #7582: RPM Filtering gyro harmonic notch 0,0,0 value issue

fujin, 2019-02-16:
**Describe the bug**
When Bidirectional DShot is enabled, DShot Burst is off, and RPM filtering configured per defaults, the filtered gyro outputs 0,0,0. My current theory is that the previous code's (prior to patch, then hotfix) 1000Hz `deactivateFreq` notch-parking capability allowed the filters to behave as expected when zero `0` or `4096`, or that we have an error in cutoff calculation, or some division by zero occurs.

**To Reproduce**
1) Flash build [1433+](https://ci.betaflight.tech/job/Betaflight/1433/changes)
2) Configure 4/4 bidirectional dshot RPM filtering defaults:
```
defaults nosave
set gyro_sync_denom = 2
set dshot_burst = OFF
set dshot_bidir = ON
set motor_pwm_protocol = DSHOT600
set pid_process_denom = 1
set scheduler_optimize_rate = ON
save
```

**Expected behavior**
Gyro filtering works as per builds prior to 1433.

**Flight controller configuration**
See 'To Reproduce'. No additional configuration required. Presents on non-compatible hardware, etc.

**Setup / Versions**
 - Flight controller: AIKONF4, pirofliprc.
 - ESC: aikon32pinf0 4in1 blheli32.6.3 test firmware
 - JST-SH-8P connectors between FC/ESC.

**Additional context**
I have published branch `fujin` with revert of #7484 #7578 for temporary testing (to support my training regime)

If `gyro_rpm_notch_harmonics = 0`, normal behavior is observed (i.e. when the RPM notch filtering is disabled)

Hydroculture, 2019-02-16:
Hi Fujin!

Im also running into problems with the latest jenkins builds, running rpm filtering.
I have OBf4 @ hglrc xjb f440, latest BLHELI Testcode xx.4 installed, running dshot 600

Since Im forced to run the optimized sheduler to have DSHOT rpm telemetry, 
i have to plug the lipo in/out several times to get all ESCS to initialize at the same time.
If they initialize correctly:
     - it may be to get no throttle response at all, motors are spinning peacefully, but no throttle response.
     - it may fly perfect

Going back to earlier version (the one Mikeller made me to test your SA2.1 fix, maybe you remember)
everything is fine, including DSHOT RPM filtering.

Curious if this sounds similar to your experience described above

best 
Holger

fujin, 2019-02-16:
@Hydroculture That does sound similar yes. You can check with my simple reproduction case by observing the Expert mode 'Sensors tab' with the supplied debug configuration. 

We're honing in on the problem now. It seems one of the test cases experiences the problem with 8kHz mode on MPU-6000, but one other test case experiences it with 4kHz mode on ICM-20602, and on 8kHz is fine, suggesting a potential interaction with #7567 (but not confirmed yet.)

Hydroculture, 2019-02-16:
Just opened your branch, looking forward, full of hope and joy to test outside.
But no Hex for my OBf4 and me. So sad. So hard to bear. 
Hope you brain-boosted-people can get it fixed.

best
Holger

Hydroculture, 2019-02-16:
Jenkins #1434 and #1433 (before 32khz removal) give me same errors.

joelucid, 2019-02-16:
The reason for this behavior is that our notch filters become dysfunctional if fed too high of a frequency. This frequency used to be limited to 1000 hz in the rpm filter. When eliminating the parking at 1000 hz the upper limit at 1000hz was also removed which now creates the issue. I'll have a fix momentarily.

Hydroculture, 2019-02-18:
Hm, im still experiencing troubles, using dshot_bidir in latest jenkins builds.
Going back to earlier version (the one Mikeller made me to test your SA2.1 fix, maybe you remember still)
everything is fine, including DSHOT RPM filtering.
Just wanted to say, seems to me still related to the above issue AJ pointed out.
Thank you so much for your work.
best
Holger

closed 2019-03-15 by joelucid